visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #1: chat

visualapproach, 2020-07-30:


tomganleylee, 2020-08-12:
just came across this, looks great project, I've got a 6pin version as well, 

when I get a chance ill try and add MQQT support to it. 

also design a single pcb for it, (happy to send one to you)

visualapproach, 2020-08-12:
@tomganleylee Thank you! Looking forward to improvements :-) I just uploaded a newer revision that you can use instead. /Thomas

faboaic, 2020-08-15:
Thanks so much for your project.
I have a lay-z-spa Miami and at least the display looks the same.
I never had it open until now and will check.

MQTT support would be perfect feature to integrate the lay-z-spa into house automation with node red.
I have 20 kWp photovoltaics and automatically pushing excess PV power into the lay-z-spa instead of regulating it away and leaving it lying on the roof would be great. 

I have all the parts except the level shifter. 
So it will take a little time until I have the project up and running.
But thanks so much  :-)

Cheers,
Fabian

tomganleylee, 2020-08-15:
Visualapproach has done a great job, I have built the circuit and
programmed the esp, going to test it tommorw.

Also putting a PCB together for it. To make it look nice :). ( I will
upload the files )

Hopefully start mqtt next week ( I have 5p energy at night

877dev, 2020-08-15:
@visualapproach really excellent work, thanks for sharing. I have all the required parts and will build and test within the next week. Also as an mqtt user any improvements will be much appreciated.

When I read the code I realise I was very far from getting this working myself :)

Is the nodemcu powered from the 5V line of the 6pin connector?

visualapproach, 2020-08-15:
@877dev Yes, it is powered via the 6 pin cable (5 V to Vin). I hesitated at first since I didn’t know if it would support the extra load. But it did :). At least in my case. First I used an Arduino nano + esp-01 and a dc-dc converter for the 3.3V side and a trafo for the Arduino. But it was really tempting to get rid of all the extra stuff so I tried. Also the step to an esp-12e meant that I only had to program one device which is neat.

carverhaines, 2020-08-16:
MQTT would be awesome, then I can integrate into HomeAssistant!

byles, 2020-08-16:
Thanks @visualapproach , I only got as far as analysing and decoding, other real life commitments get in the way :( can't wait to try this next weekend! I'd contribute mqtt and pcb design but for sure somebody else will beat me to it.

If you're in the UK you can find the connectors on Amazon: https://www.amazon.co.uk/sourcing-map-Plastic-Terminal-Connector/dp/B07GT51DZ2

If you're not, at least it gives you a description of what to look for.

visualapproach, 2020-08-16:
@byles Thanks, good find on the connector!

byles, 2020-08-16:
I only used two, if you have any trouble sourcing them I'll happily send some your way!

hhovde, 2020-08-16:
Hi!
Nice work @visualapproach. 
I stumbled up on your solution via another project; https://github.com/ximon/Hot-tub-remote
Both of which I found during research for my own design. 
I started tinkering with my spa last year, but are under heavy time constraints atm. and where looking for a turn-key solution, but mine is a 4-pin version.

@byles - I believe the connectors are called "JST SM Connector" 
I ordered some of those for my pump last year (4 pin version) and they fitted nicely.

/H

carverhaines, 2020-08-16:
@visualapproach, Can you please upload a picture of the back of your device? I want to see how you laid it out. Thanks!

877dev, 2020-08-16:
Here’s the cables I got, confirmed working on Vegas 6pin model:
https://www.ebay.co.uk/itm/323733997560

Some description of how the code works would be most interesting, it’s way way beyond what I could create but I like to at least be able to read the code 😂

byles, 2020-08-16:
Damnmit! I spent ages crimping the tiny little buggers... think I'll get them from eBay too! Cheers @877dev

visualapproach, 2020-08-16:
> Some description of how the code works would be most interesting, it’s way way beyond what I could create but I like to at least be able to read the code 😂

Yes I will try to put something together.

visualapproach, 2020-08-17:
> @visualapproach, Can you please upload a picture of the back of your device? I want to see how you laid it out. Thanks!

I'd rather not. I would have to disassemble the pump and it would add no extra value. It's just a bunch of wires soldered according to the schematics.

visualapproach, 2020-08-17:
> Some description of how the code works would be most interesting, it’s way way beyond what I could create but I like to at least be able to read the code 😂

Wish partly granted. There is an excel file among the files now. "github-bestwayhackdocs". I'll make more comments in the code when I have time to do so.

visualapproach, 2020-08-17:
Regarding MQTT. I am not using this myself yet, so if you guys will try it out it would probably get done faster. Hint: Every time the program decides that something has changed, it calls "savelog()". So I think that is were you want to start tinkering and send stuff to the broker. On the receiver side, I suggest looking at the websocket event.

Have fun

faboaic, 2020-08-19:
I made some progress... 
But these damned connectors don't fit.  👎 

Regarding warranty I don't want to crimp the cable.

So I am stuck now  :-( 

[image]
[image]

visualapproach, 2020-08-19:
@faboaic is it the 6 pin Female on your build that don’t fit? I sanded/cut off half a mm on both sides to make it fit.

faboaic, 2020-08-19:
Yes, exactly the female one.
I will try once again this evening.
I doubt that the pins will fit since they are larger than the standard connectors.

byles, 2020-08-19:
@faboaic check the links above to find the SM connectors, they're only a couple of £/$/whatever :) :)

@visualapproach i've got this working! I'm gonna have fun \o/

faboaic, 2020-08-19:
@byles yes, I've already ordered 20 pairs off Amazon, should arrive on Friday.
What model of lay-z-spa do you have ?
Some people in the German Facebook group are asking which models are supported.
Maybe we could create a list of supported spas (model name, model number, build year )

byles, 2020-08-19:
It's only kind of working at the moment... Keeps 'locking up', and webserver stops responding. It's late and windy out at the moment so will carry on playing tomorrow.

Model is UK #54123 (Miami liner). Not sure where I find the year.

carverhaines, 2020-08-20:
I have built mine according to the diagram, but it is not working, the spa beeps once when I plug it in, so I know it's seeing it, but the control panel does not turn, and the web interface is displaying but no controls work.
I am using 2x 4 channel LLC, so I'm hoping that is not messing it up. I attached ground to both and 3V to both

visualapproach, 2020-08-20:
> I have built mine according to the diagram, but it is not working, the spa beeps once when I plug it in, so I know it's seeing it, but the control panel does not turn, and the web interface is [...]

One beep at startup seems odd. It should play a melody. Have you perhaps connected the sound pin from the cio?

carverhaines, 2020-08-20:
I think it has to with #3 , I am going to fix and then I'll let you all know.

carverhaines, 2020-08-20:
> I think it has to with #3 , I am going to fix and then I'll let you all know.

So I've fixed that, now there are 3V going to the LV pin on both boards, Ground going to GND on both boards. It is still not working. I do believe I hear a melody (very faint, much quieter than usual beep) on plugin. Control panel does not light up, web app appears but does not do anything.
Of note, I do not have anything plugged into the HV side of the LTCs per the original schematic. Does that need connection to 5V from the pump?
Images here: https://photos.app.goo.gl/y8YX8JQacFwiC9XJ7

carverhaines, 2020-08-21:
I've double and triple checked my wiring, this is what I have:

[image]

visualapproach, 2020-08-21:
@carverhaines debugging; test signal throughput with a multimeter. Easiest to write a program that toggles all pins every x seconds and measure wether it shows on all pins at the expected voltage (3.3 and 5)

877dev, 2020-08-23:
Man I cannot wait to build this myself, and tinker with adding mqtt and Blynk iOS support. I'm so busy with home renovations and now I am relaxing with a beer. Maybe when I sober up :)  Great project!!

visualapproach, 2020-08-23:
Thanks @877dev , you got your priorities right ;-)

877dev, 2020-08-24:
Is there any reason the pin numbers cannot be changed in code?

`const int CS_DSP_PIN = D3;
const int CLK_DSP_PIN = D4;
const int DATA_DSP_PIN = D5;
const int AUDIO_OUT_PIN = D6;

const int CS_CIO_PIN = D1;
const int CLK_CIO_PIN = D2;
const int DATA_CIO_PIN = D7;
`

I think I read you are bit banging so not using and hardware SPI? 
As it would make the actual wiring much easier if every pin lined up..

I know different ESP8266 pins do things on booting, maybe it would break something.

visualapproach, 2020-08-24:
@877dev welcome back! I wondered when that question should come up. The reason was both the restrictions at boot but mostly because some pins is better as input than output and vice versa. So I decided early on this layout and after I got it working it was already soldered and I didn’t bother to change it. It is probably ok to switch some pins but I recommend that you doublecheck data lines to be on a bidirectional pin. And the other ones according to their input/output mode.

visualapproach, 2020-08-24:
https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/

carverhaines, 2020-08-24:
> Is there any reason the pin numbers cannot be changed in code?
> ...
> I think I read you are bit banging so not using and hardware SPI?
> As it would make the actual wiring much easier if every pin lined up with the respective connector pin.

I strongly agree with this, would make things significantly simpler!

877dev, 2020-08-24:
Thanks, i've been looking at it a while. Seems like this might work, what do you think?

As I've not read the code properly maybe something there could cause an issue, if CIO CS is pulled low at boot, or if DISP AUDIO is pull high at boot.

[image]
[image]

visualapproach, 2020-08-24:
Every pin is pulled high from 3.3->LLC I think so that audio might be trouble. I don’t believe anything will break so you could try it out. Maybe bootfailure will occur.

877dev, 2020-08-24:
Ok I will have a think, why mess with something the already works! 👍

carverhaines, 2020-08-25:
> @carverhaines debugging; test signal throughput with a multimeter. Easiest to write a program that toggles all pins every x seconds and measure wether it shows on all pins at the expected voltage (3.3 and 5)
@visualapproach, it turns out one of the LTCs were bad. Never tested it before building. I have rebuilt w/ working ones (luckily I ordered pack of 10 : ). Now the display powers on and displays the correct temp and state, but I cannot use any of the buttons. Additionally, the web interface does not boot up or load (however it does when unplugged from spa and ESP is powered via micro usb.)
[image]

byles, 2020-08-25:
> > Niice.
> > I have it working, kind of.. It seems to lock up after a minute, not sure what happens if/when Wi-Fi disconnects? Guessing something is happening and blocking. Not enough hours in the day! :(
> 
> @byles You are not talking about the display beeing locked (as it should be after a couple of minutes) right? Check if the ESP goes into AP mode (look for wifi ap "auto AP" or whatever I called it)

Hey @visualapproach. Nah, not locked display. Just seemingly stops responding. I'll do some sensible poking around when I get time :)

visualapproach, 2020-08-25:
@byles @carverhaines Remote debugging is hard. Latest software in MASTER is confirmed to work, so I must conclude hardware fault. If you own a logic analyzer it would be your best friend. Or an oscilloscope. Or worst case - arduino with serial plotter as a rude kind of analyzer, or at least a LED or something to see if any pins are dead or alive. Don't forget a current limiting resistor though. (HINT: the built in led might tell you something about one of the lines)
Good advice for any bug report is to be as specific as possible - software version*, settings, hardware, what happens just before or after your hangup etc. *mqtt test version without any customization is guaranteed to hang, waiting for connection.
Does the program really hang, or is it waiting in AP mode?
If the Serial monitor doesn't show ANYTHING, even after reboot, I'd say boot failure, because it writes a start message almost immidiately.

visualapproach, 2020-08-25:
I haven't looked in to it concerning the 8266, but I've had some bad smoke experiences with arduino with simultaneous use of USB power and 5V.

carverhaines, 2020-08-26:
Ok, I created a little script that would turn all the D-pins used in this project on high (and the board LED would be on high as well) for 5 seconds, then pins and LED off for 5 seconds: https://pastebin.com/F4fF2xrz

On the LV side, when on HIGH, D-pins would show some voltage >.1V, and on LOW would show almost zero.
On the HV side (the CIO & DSP pins), they would show close to 5V when the D pins were LOW (led off), and lower (~0.6V) when the D pins were HIGH (led on)

Is this expected for it to reverse the binary like this, or are my LLCs not working as expected?

visualapproach, 2020-08-26:
@carverhaines good work, now we are getting somewhere 👍 ! No that’s not the expected behavior. Maybe check some of the other converters if they are consistent and behave as expected out of circuit. For instance there should be a diode you can measure between LV and hv. Or if it is the other way around. I might look at it when I have more time, but google can probably help you meanwhile.

877dev, 2020-08-26:
@carverhaines I did the same thing myself today if you want to try it?
https://pastebin.com/FvpkBHMR

On the HV sides, I get around 0.6V when low, and around 4.6V when high.

carverhaines, 2020-08-26:
@877dev , I'll try your script too to see if I get the same result

visualapproach, 2020-08-26:
One more thing- to be sure the pins are on when LED is lit (it may be reversed) you should for example make the on time different from off time.

carverhaines, 2020-08-26:
@visualapproach , you are right, it is reversed, LED is on when pins are LOW, and I double checked, the LV and HV are high/low at the same time. 🤦‍♂️

carverhaines, 2020-08-26:
Ok, so I've had multiple problems which has made it hard to diagnose, 1st bad I2C LCC, next 2 bad jumper cables.
Now I have made sure every resistor, cable, and LCC works. I verified PIN BY PIN that the correct Digital pins goes to the correct pins on the CIO/DSP, and that when each D pin is high, it will measure ~3v on only that pin on the LV side, and ~5v only on that equivalent CIO/DSP pin on the HV side. So everything seems to be built correctly and all parts functioning. 

Now having these problems, using v1.26

1) With jacuzzi already on, temp set to 104, current temp 102, I unplug the display, plug the ESP into the display first, and then into the CIO. I hear a melody, the display shows what it did before unplugging, and the webapp shows the current temp (but not set temp), however none of the buttons on the display work, and buttons on the web-app don't do anything. (see left image phone screenshot)

2) With jacuzzi UNPLUGGED, i plug in the ESP, replug the jacuzzi into 110v, I hear the melody, and the display powers on with every light lit, and random characters on the display (see 4 right images). Pushing display does nothing, web app loads but does not see current state of anything. 

[image]

visualapproach, 2020-08-27:
Maybe I should post a video on how the start sequence is supposed to look like, so people can compare exactly what differs? I will upload in a day or two.

visualapproach, 2020-08-27:
[](https://youtu.be/rdZLVbrmNxs)

visualapproach, 2020-08-27:
> Now I have made sure every resistor, cable, and LCC works. I verified PIN BY PIN that the correct Digital pins goes to the correct pins on the CIO/DSP, and that when each D pin is high, it will [...]

Hm, maybe the rebranding does matter. Give me a while and I will make and upload a debugging version for you.

visualapproach, 2020-08-27:
Hey @carverhaines , I uploaded a testversion just for you. Download and follow the instructions in the code remarks. (Top lines)

877dev, 2020-08-27:
Hmm first test, I heard the song, but the display is totally blank. 
Time to check my wiring again ;)

877dev, 2020-08-31:
I compared hackdocs excel file to one of my traces, strangely it checks out :-)

Amazing to see how it all works under the head, really well done @visualapproach for getting this working. If I thought the hardware was difficult, then the code is on another level!

In my free time I will be reading the code and learning from it, it has applications for another project of mine.

[image]

visualapproach, 2020-08-31:
@877dev Thanks. Nice graph :-)
I reused a lot of knowledge I got on an earlier project (Volvo-melbus). It has almost identical "one data wire SPI" connection. I also learned that life is much easier with a logic analyzer instead of using an arduino for probing. When I eventually understood how the protocol worked and where the important bits were and what they meant, I tested it with an arduino. Then I went on to the 8266 for the first time, and it was a lot to learn there too. It crashed countless times and I almost gave up, but eventually it worked. Feel free to ask if you have any specific questions about the code.
I'm currently working on a new version. Nothing fancy, but hopefully with some more bells and whistles, like a config page. HTML is not my strongest talent though but I might get something out in a week or two. Lot of real life stuff that needs attention also...

Naesstrom, 2020-09-02:
Love it, found this and ordering the parts now to build it for our Paris spa!
PiHut have 6-pin JST Cables to, just ordered a pair there: https://thepihut.com/products/6-pin-jst-sm-plug-receptacle-cable-set

Will the Adafruit levelshifter work, the labels doesn't look like yours: https://www.adafruit.com/product/395

visualapproach, 2020-09-02:
> Love it, found this and ordering the parts now to build it for our Paris spa!
> PiHut have 6-pin JST Cables to, just ordered a pair there: [...]

Hej och tack @Naesstrom ,
That levelshifter may work but isn’t it extremely expensive? I looked at it briefly and it has enough lines but I don’t really know anything about it to be honest.

877dev, 2020-09-02:
I tried that level converter today as it happens, and although the display works it gives out distorted audio. So I would probably stick to the one in the readme.

[49 more replies]

closed 2020-12-09 by visualapproach