visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #3: schematic incorrect

byles, 2020-08-19:
Hey do's,

D2 and 3v are drawn wrong.

D2 --> LV5
3v --> LV2

Should be the other way round?

:)

carverhaines, 2020-08-20:
Let me know if this turns out to be true, because I'm having trouble getting mine working based on the schematic. I connected D2 to LV5 which is the LV pin input on the LLC which I thought was strange, but I did it anyways

visualapproach, 2020-08-20:
There was an error. I removed the picture and will update later. Out of town for the moment. 
However, the important thing is that 3.3V goes to LV and GND to GND. 5V to HV and GND to GND. The other pins go to any pin on the LLC as long as it’s on the correct side (LV/HV) and connects the right pin on the esp to the pump. Be aware though: I have had LLCs that were marked wrong (LV and hv switched)

byles, 2020-08-20:
I happened to end up with the same LLCs you have in your photo (or at least look identical), and communication seems flakey... I best look more closely!