visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #10: HWchecker v.01 results

877dev, 2020-08-27:
Hi @visualapproach 

Hope you don't mind me posting my results, I am getting no melody with the HWchecker version, can access the web interface, but no button on the physical pump work and no lights:

[rawdata.zip](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/5138339/rawdata.zip)

visualapproach, 2020-08-27:
I don't mind. You followed the instructions ;-)
Communications between CIO (pump) and ESP is working fine. Your problem is (likely) between ESP and DSP.
There should not be any sound in this version. Only logging, but touch buttons should work. (Not tested though)

carverhaines, 2020-08-27:
@visualapproach Here are my results: https://pastebin.com/raw/BRH4Bypx
 Display was just random gibberish during this process

visualapproach, 2020-08-28:
@carverhaines Same here, com from cio is working. I can make another test program.

visualapproach, 2020-08-28:
In loop() replace
UpdateDSP(DSP_BRT_IN);
With ...(0xF1);
This should force max brightness no matter what the pump says. It won’t solve the gibberish but at least it should lit up something

visualapproach, 2020-08-28:
Hey guys, I uploaded v0.2 of the HWchecker. It should give some more info on the display. The display is supposed to blink all LEDs on and off every half second or so. Also it saves the button presses.

877dev, 2020-08-28:
> Hey guys, I uploaded v0.2 of the HWchecker. It should give some more info on the display. The display is supposed to blink all LEDs on and off every half second or so. Also it saves the button presses.

Great thank you. I’ll try it very soon. Same as before, just plug in and leave for a few minutes?

visualapproach, 2020-08-28:
> Great thank you. I’ll try it very soon. Same as before, just plug in and leave for a few minutes?

/*
upload this firmware and let it run for a minute or two. **Press a couple of touch buttons.**
Then browse to the ip/rawdata.csv - for example 192.168.4.123/rawdata.csv
copy and paste it into a new issue on github.
*/

closed 2020-08-31 by visualapproach