visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #11: HWchecker 0.2 results

877dev, 2020-08-28:
Hi @visualapproach 

Here's my results, again no lights at all on the display, but when removing the device the lights come immeditely back on with the current temperature showing.

Testing on DSP side HV1-4 I have 20KOhms between them, so I don't think there's a short.

[rawdata.zip](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/5142973/rawdata.zip)

Thank you

visualapproach, 2020-08-28:
u have a multimeter with diode check?

877dev, 2020-08-28:
Yes it's a Flue 115 meter

visualapproach, 2020-08-28:
Good. Disconnect device from pump and display. Put fluke in diode mode. Put red lead on an LVx and black on the corresponding HVx. Meter should show a value, like 500. Switch red and black. It should give no reading (or OC). Do that for all display lines

877dev, 2020-08-28:
[image]

Display & CIO sides give same readings/

Red to black:
LV1 to HV1 0.552V
LV2 to HV2 0.558V
LV3 to HV3 0.541V
LV4 to HV4 0.547V

Black to red:
LV1 to HV1 OL
LV2 to HV2 OL
LV3 to HV3 OL
LV4 to HV4 OL

visualapproach, 2020-08-28:
>I have a cheap logic analyzer, is it worth hooking i up to my display data/clock/cs outputs?
Currently sat on my desk, it's raining outside here in the UK typically :)

YES, absolutely!

visualapproach, 2020-08-28:
Same here btw. And now it's beer o'clock, but I'll check in here during the evening

877dev, 2020-08-28:
[877_traces.zip](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/5143207/877_traces.zip)
[image]

Had a quick look, the file was saved using Sigrok if you use it?
No connection to CIO, analyzer connected to DISP connections:
D0 - DSP_DATA
D1 - DSP_CLK
D2 - DSP_CS
D3 - AUDIO

You can see some activity when booting up, sound signal shows. But after that, nothing.
When pressing buttons on the web interface, they show in serial monitor, but no pulses on the trace.

877dev, 2020-08-28:
> Same here btw. And now it's beer o'clock, but I'll check in here during the evening

Yes some time off would be nice, I have food on the way ;)
Enjoy your beer(s) :)

visualapproach, 2020-08-28:
Thx, just to doublecheck, is LV 3.3V and GND connected?

877dev, 2020-08-28:
Yes LV is 3.3V and all grounds connected.

visualapproach, 2020-08-28:
Maybe analyze directly on the esp pins (LV side)

877dev, 2020-08-28:
Yes ok, I will try that next 👍

visualapproach, 2020-08-28:
Ok I checked the .sr file and it seems OK! Are you kidding me, or is it really not showing anything the first 5 seconds (the data shows activity in 5 secs). Abandon the LV analysis and analyze the lines between CIO and DSP without the ESP device. Lets see if there is anything weird going on.

877dev, 2020-08-28:
I'm definitely not kidding I'm afraid. This is what I'm doing:

Turning power off to the pump.
Connect the ESP device inline.
Turning power on to pump.
The startup melody plays.
No lights on display, not even power light.
Nothing shows on LCD at all.

> Abandon the LV analysis and analyze the lines between CIO and DSP without the ESP device. Lets see if there is anything weird going on.

You mean on the pump itself, no ESP connected inline? I have already done that, I can add the traces.

visualapproach, 2020-08-28:
OK I like a challenge, but this is annoying. Data is getting through, except for button requests. That would indicate SW is the culprit rather than HW. But why does it work for other people then? I need too sleep on this. Did you sample the checker version or the full v1.26?

877dev, 2020-08-28:
I'm using full 1.26 and that's the trace I posted above. Yes very annoying!
Strange that @carverhaines is getting the same results as me, maybe some difference in protocol?

It's strange how the message data is getting through, but not being written to the screen.

I can do more trace files over the weekend if you need, for now maybe a good nights sleep will help.

Thanks so far for the help!!

visualapproach, 2020-08-28:
You are welcome. Regarding carver Haines I think it’s because he hasn’t scared the pixies enough. I mean 110v ... pfft ;) (No offense ch)
If you could sample hwchecker 0.2 for comparison it might help.
Good night

carverhaines, 2020-08-29:
Just got home from work. Tried this new checker. No lights on the display, gave it about 30 seconds incase it was booting up, then pushed buttons in this order w/ 1 second pause in between: PWR, Lock, Timer, Bubbles, C/F, Heat, circulate, down, up.

My data looks about the same as @877dev , so I don't think we can blame my wimpy 110V power. 
https://pastebin.com/raw/kDrQagFn

@visualapproach:
> But why does it work for other people then?
Does it, have you confirmed this with anyone?

Totally shooting in the dark, but is it possible each pump/display have some sort of encryption key that encodes & decodes the signals to each other, so that people can't just go buy a replacement display for example and fix their own pump, would have to buy a new one or get manufacturer to fix instead?

carverhaines, 2020-08-29:
Second example, plugging in ESP after pump already on and running:
https://pastebin.com/raw/mgjrbxm8
Get some lights on and gibberish on LCD

visualapproach, 2020-08-29:
@carverhaines 
> Does it? Have you confirmed this with anyone?

Yes. And I have tested it on two pumps myself.

I don't think it's encrypted because the data that is working is consistent with the protocol.

Let's see @877dev 's analysis file first. 877- take notes of which analyzer channel is on what cable pin and corresponding ESP PIN please.
I'm thinking like this: If there is proper signals reaching the display without it working, it can be because of following reasons.
1. Pins connected to wrong line.
2. Timing requirement is different on this particular device (not likely because my tests indicates a broad tolerance in data rate)
3. Device is broken (in which case it wouldn't work without the ESP)
4. Severe electromagnetic noise. (You don't live in a transformer station or radio broadcast center I suppose)
There may be more reasons but that's what I could think of.

877dev, 2020-08-29:
I was going to say I am sure @visualapproach would have checked it works on his own pump! :)

I'm pretty sure it's not encrypted as well, as I was able to check the data bits against the datasheet and they matched.

Here is the trace from the HWchecker 0.2, not connected to pump.
None of the button in the web interface do anything, and no message shows up in serial monitor when pressed (maybe intended with this hwchecker version).

[image]
[877_trace_hwchecker0.2.zip](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/5145189/877_trace_hwchecker0.2.zip)

visualapproach, 2020-08-29:
> I was going to say I am sure @visualapproach would have checked it works on his own pump! :) I also have two pumps, a result of Bestway's warranty procedure!

hehe ... thanks!

> None of the button in the web interface do anything, and no message shows up in serial monitor when pressed (maybe intended with this hwchecker version).

Correct, it's minimalistic. Webserver is only running to deliver the data file.

> As the analyzer is picking up the data signal correctly from the ESP, we are happy there is no issues with the wrong pins being used/hardware issues on that end?

No and no :-). It seems you used too low sample rate. Can you redo it with at least 100 KHz? And by the second "no" I'm thinking that either data is good. In that case the most probable cause is wrong line to wrong pin. (From my point of view. I bet you have checked over and over again.) In the other case, if data is corrupt, it would still most likely be caused by hardware, like a bad cable or similar. 
To make me stop nagging about the cables beeing wrong:
Write down 
ESP Pin# - LLC LV# - LLC HV# --*--Connector to DSP Pin #     DATA CHANNEL#
  stick your analyzer at the asterisk *
It's not that I don't trust you. It's from my own experience. I need to do this to myself when things keep fightning me :-).

> I can do some real traces on the pump soon after I finished my chores :D

Super!

visualapproach, 2020-08-29:
[image]

877dev, 2020-08-29:
Here's a new hwchecker 0.2 analyzer file with 250kHz sample this time:

[hwchecker_0.2_250kHz.zip](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/5145312/hwchecker_0.2_250kHz.zip)

One thing I realised and hope is not causing the issue, as you keep asking about wiring. I did not exactly follow your pinouts, mine is as below. But according to the info it should be fine. 
I did change the code to match my new pinouts...

[image]
[image]

visualapproach, 2020-08-29:
I can't see any obvious errors with that layout. To rule out one more thing - while I check out the file, you could verify that the connectors is not connected backwards. (reconnect the original cables without ESP. Mark pin 1 (5V) on both cables. Reconnect to your device and check that it is correct) I know it sounds stupid but sh*t happens and also @faboaic admitted that's what happened to him (due to my poor description).

visualapproach, 2020-08-29:
@877dev I'm sorry but unless you put the connector backwards it beats me! Data file checks out perfectly fine and yet, the display doesn't answer.

877dev, 2020-08-29:
[image]
[image]

877dev, 2020-08-29:
> @877dev I'm sorry but unless you put the connector backwards it beats me! Data file checks out perfectly fine and yet, the display doesn't answer.

I'm going blind double checking but it seems ok, I will have a cuppa then head down the garden to plug in for real and check the 5V/GND connections. Then after that I will try and get some analyzer reading from the display end with my janky test cable ;)

I kinds hope it is something stupid I am doing!!

Test cable to connect analyzer in series:
[image]
[image]
[image]

877dev, 2020-08-29:
Hmm double and triple checked my wiring, even got my spare pump head on the desk and it's the same (melody plays, but no display at all).

I notice my Wemos is giving out 4.6V, not the full 5V. Could this be the issue? Although the pump provides the 5V in normal running, so not so sure..

877dev, 2020-08-29:
SUCCESS!!!!!!!!!!!!!!!!!!!!!!!!!
Ok so I know I am kinda just talking to myself now :)

So I quickly built a breadboard with a NodeMCU and level converter on just the display side, and it worked first time - so happy! 

So maybe something with the Wemos is does not like, not enough current, resistors making it worse, etc...

Thanks you @visualapproach for your patience and help!

[image]

visualapproach, 2020-08-29:
High five! Thank you for sharing the good news!
Maybe @carverhaines can solve his problem too now. Good work

faboaic, 2020-08-29:
> 
> So maybe something with the Wemos is does not like, not enough current, resistors making it worse, etc...
> 

Glad it works for you, now.

My wemos d1 mini works perfectly.
I used 820 ohm resistors, because I did not find 1k at that moment.

877dev, 2020-08-29:
> High five! Thank you for sharing the good news!
> Maybe @carverhaines can solve his problem too now. Good work

Yes hopefully, thanks for all the help!

> > So maybe something with the Wemos is does not like, not enough current, resistors making it worse, etc...
> 
> Glad it works for you, now.
> 
> My wemos d1 mini works perfectly.
> I used 820 ohm resistors, because I did not find 1k at that moment.

Thank you, I wonder if removing the resistors would fix the issue. Or maybe it was changing the pins that it did not like, I can try that as well.

But for now at least I know it works!  :)

877dev, 2020-08-30:
Update- removed the resistors on my original board and that works too! So it must’ve been the resistors limiting the current so much that the display could not register the signal. I guess. 

I will try with maybe 500ohm resistors when I have more time.

visualapproach, 2020-08-30:
Okay, very weird! But good that you discovered that. The best explanation I can come up with is this:
When digital pin on ESP is LOW the 1K resistor forms a voltage divider with the 10K pullup resistor on the LV side. This would give some 0.5-0.6 V as LOW signal at the display. The logic analyzer interprets that as a 0, but the display may require 0.4V to count it as 0. Long cables acts as resistors also which may pronounce the issue.

carverhaines, 2020-08-30:
@visualapproach, @877dev, 
YES! I took out the resistors and volia! It works!

Is it safe to run like this without the resistors?

visualapproach, 2020-08-31:
@carverhaines Excellent! Thanks for sharing your success.
Is it safe to omit the resistors? I updated the readme with some info but the short answer is yes. No guarantees given, but I don't have them in my circuit and I sleep at night.

877dev, 2020-08-31:
@carverhaines great news! 

I tried adding 1K resistors with my nodemcu breadboard setup, and that worked perfectly. So it seems something specific to the Wemos board in my case. There are many Wemos clones out there, so there might be differences with results.

carverhaines, 2020-08-31:
I have a LoLin nodemcu v1.0, looks the same as yours, not using a wemos.
I'm going to let it be for now :)

closed 2020-08-31 by visualapproach