visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #13: Our cozy places

visualapproach, 2020-08-31:
You are welcome to post pictures of your SPA here for us to be inspired of. Here is my place:

[image]

877dev, 2020-09-01:
Very nice! My spa is down the garden while we have house renovations. I will post when it’s back in place.

eddcaton, 2020-10-07:
[image]
[image]

visualapproach, 2020-10-07:
Awesome! Me like the star heaven! Thanks for showing it to us!

eddcaton, 2020-10-07:
Thanks :) I have to say drilling almost 1000 holes by hand with a 2mm drill bit was not the highlight of my week!

closed 2020-12-09 by visualapproach