visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #14: Failed to get time from server, retry

877dev, 2020-09-03:
Am I the only one getting this error a lot of the time? Sometimes it connects first time, other times it takes forever.

I used to get this a lot on other projects until I swapped to using the EZtime library:
https://github.com/ropg/ezTime

EZtime always connects first time for me.

Or I could just imagine I am back in 1st Jan 1970 in the Hot Tub Time Machine :)

[image]

877dev, 2020-09-03:
Update, this may be a fix:
https://github.com/mcxiaoke/ESPDateTime/issues/7#issuecomment-686408433

`void setupDateTime() {
  // setup this after wifi connected
  // you can use custom timeZone,server and timeout
  // DateTime.setTimeZone(-4);
  //   DateTime.setServer("asia.pool.ntp.org");
  //   DateTime.begin(15 * 1000);
 
  DateTime.setServer("pool.ntp.org"); // ***add this line***
  
  DateTime.setTimeZone(TZ);            
  DateTime.begin();
  if (!DateTime.isTimeValid()) {
    Serial.println("Failed to get time from server.");
  }
}`

visualapproach, 2020-09-04:
Good to know!

closed 2020-09-04 by 877dev

engbryan, 2021-01-17:
The issue persists for me, even using the cloudflare ntp. Does anyone have a light on it?