visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #15: feature request: switch/timer on either D0 or D8

creativtransfer, 2020-09-03:
I have an idea you and probably others might like: 

We are using an ozone generator such as this one [https://de.aliexpress.com/item/4000891202092.html](http://s.click.aliexpress.com/deep_link.htm?dl_target_url=https%3A%2F%2Fde.aliexpress.com%2Fitem%2F4000891202092.html&aff_short_key=2VrnmiM) in addition to chlorine tabs to keep the water clean. Until now, I just put the pipe in the water and hope that I don't forget to switch it off after approx. 30 minutes. 

Since there is enough space in the unit, I was thinking about installing that generator inside of the case, add a non-return check valve and connect it somehow to the pipe after the heater element. I would then install a relais module to switch it with the ESP8266. It would be a nice upgrade if there was a feature to turn it on/off and have a timer for e.g. 30 minutes...

What do you think?

Thank you and keep up the great work!

877dev, 2020-09-04:
My thoughts are that it could be done quite easily in code, but connecting the water pipes would be quite difficult, and prone to leaking. Also having a 230V relay floating around inside is not great. 

Better and simpler would be a smart plug connected to the ozone generator to stop it after 30mins, I like the Sonoff S20’s as they can be flashed and integrate with MQTT/node red. Although they can just be used out the box with their app too.

visualapproach, 2020-09-04:
> What do you think?

Yes, I've been thinking of adding auxiliary pins for extra equipment, like LEDs and stuff. Things to consider before doing so:
1. max current of the ESP pin is something like 10 mA. Probably not enough to drive a relay without adding extra components.
2. max current from the pump's 5V is unknown. It might not be enough to drive a relay coil without releasing the magic smoke.
3. If using a coil relay anyway - be sure to have a fly-wheel diode!
4. Considering the above; maybe solid state relays, or FETs would be the way to go, depending on what kind of voltage/current you are controlling.
5. Pins can't be pulled up/down at boot and may trigger high/low at boot making aux equipment behave strange.

> Thank you and keep up the great work!

You are welcome :hugs:

visualapproach, 2020-10-26:
Function added (but not tested) in v2.0.0. Examples.
Closing issue

closed 2020-10-26 by visualapproach