visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #18: Code discussion

877dev, 2020-09-07:
I'm taking some time to understand the code, in order to learn more and help with mqtt integration. Hope this is ok to ask a few (lots of) questions? :)

(Questions in brackets), please correct anything you have the energy to address :)

1) startup code:
Setup pins
Play intro melody and display welcome message
Connect to WiFi and display IP
Connect to NTP time server
Start OTA server
Start SPIFFS server
Start websocket server to listen for events
Start http server for web interface

2) loop code:
Set a variable called counter which persists through each call of the loop (why static?)
Call handleData() which handles button presses (i think?)
****
(Not sure how the following counter works, why 300?)
Call these functions frequently to keep up to date - webSocket / https / OTA
Increment counter by one, check if > 300, if so and NTP is set then call sendWSmessage(), which sends latest json packet to web page via web socket (correct??)
Decrement counter by 300 (???)
****

3) webSocketEvent() - called when "virtual" button pressed on web interface:
When button is pressed a websocket message is received by the server.
This is handled by function webSocketEvent()  - much the same as mqtt callback.
Payload is filtered by "uint32_t cmd = strtol((const char *) &payload[0], NULL, 16);"  -(converts string to long integer base 16 i.e. hex ???)
Switch case used to compare received hex, to check which virtual button was pressed.
Variable "virtualBTN" is updated with pressed button name, and button command is set as !current state
sendWSmessage() is called to update the web interface with the new button state

4) handleData() - called frequently by loop, deals with virtual or real button press logic (correct???)
Call releaseVirtualButtons() to check "virtualBTN" value and reset to "NOBTN", like a latch to stop spamming(?)
Call filterButtons() to check real or virtual button press, check for special AP combo.
Call schedule() to turn filter on or off
Some sort of magic code to send the value to display, either from CIO first or spoofed.
Update web interface via websocket.
Save changes to SPIFFS.

5) releaseVirtualButtons()
Compares the commanded value to the actual state, if the same then set "virtualBTN" to "NOBTN" i.e. stop button press.

6)other functions:
ISR/Data setting/sending, I have no clue, too complicated for me. I'm just gonnna assume you know what you're doing here :)

visualapproach, 2020-09-07:
> I'm taking some time to understand the code, in order to learn more and help with mqtt integration. Hope this is ok to ask a few (lots of) questions? :)

Of course! I'll comment below.

> (Questions in brackets), please correct anything you have the energy to address :)
> 
> 1. startup code:
>    Setup pins
>    Play intro melody and display welcome message
>    Connect to WiFi and display IP
>    Connect to NTP time server
>    Start OTA server
>    Start SPIFFS server

SPIFFS deprecated, LittleFS used in upcoming version. You have to install tool for uploading via Arduino ide. Same as for SPIFFS.

>    Start websocket server to listen for events
>    Start http server for web interface
> 2. loop code:
>    Set a variable called counter which persists through each call of the loop (why static?)

Because it should be persistent but not global, since it's a common variable name.

>    Call handleData() which handles button presses (i think?)

Yes and most of the communication with DSP and CIO (indirectly)

> (Not sure how the following counter works, why 300?)
> Call these functions frequently to keep up to date - webSocket / https / OTA
> Increment counter by one, check if > 300, if so and NTP is set then [...]

Forget it, it will be replaced by a real timer in next version. It was just  a stupid way of doing things at a certain interval. I think the loop executed 200 times per second, so that meant around 1.5 seconds.

> 
> 1. webSocketEvent() - called when "virtual" button pressed on web interface:
>    When button is pressed a websocket message is received by the server.
>    This is handled by function webSocketEvent()  - much the same as mqtt callback.
>    Payload is filtered by "uint32_t cmd = strtol((const char *) &payload[0], NULL, 16);"  -(converts string to long integer base 16 i.e. hex ???)

Yes, good! Now wtf use HEX you say. Lets say... because of reasons... it seems stupid but I had an idea of discriminating between 1-character length and 2-characters length messages, and still fit more than 10 commands in one char. But I later abandon that (the sliders are gone in next version)

>    Switch case used to compare received hex, to check which virtual button was pressed.
>    Variable "virtualBTN" is updated with pressed button name, and button command is set as !current state

Yes, it is a toggle function, just as on the physical panel buttons.

>    sendWSmessage() is called to update the web interface with the new button state
> 2. handleData() - called frequently by loop, deals with virtual or real button press logic (correct???)

And updates the states and values in a bunch of variables, like temperature etc.

>    Call releaseVirtualButtons() to check "virtualBTN" value and reset to "NOBTN", like a latch to stop spamming(?)

Yes, since you must press the button "a while". I don't want to time the button presses so I did it this way. Except for UP/DOWN.

>    Call filterButtons() to check real or virtual button press, check for special AP combo.

Real buttons win over virtual ones. I expanded this in next version to not allow heater to be on. Btw, I don't like the name filterButtons(). It has nothing to do with the pump filter :-)

>    Call schedule() to turn filter on or off
>    Some sort of magic code to send the value to display, either from CIO first or spoofed.
>    Update web interface via websocket.
>    Save changes to SPIFFS.
> 3. releaseVirtualButtons()
>    Compares the commanded value to the actual state, if the same then set "virtualBTN" to "NOBTN" i.e. stop button press.
> 4. other functions:
>    ISR/Data setting/sending, I have no clue, too complicated for me. I'm just gonnna assume you know what you're doing here :)

Yes I hope so :-).

877dev, 2020-09-07:
Hey thanks a lot for your comments, I am even more impressed having gone through it. I am not a software person, and just got into coding around 18 months ago as a hobby. It must be your profession I would say?

Look forward to seeing the next version :)

877dev, 2020-09-07:
> Thank you for thinking that! But I'm also just a hobbyist, be it that I started some 30+ years ago 😳 when I bricked my daddys work handheld programmable calculator.
> And I'm stubborn as a mule when I want. If you don't succeed first time, have a look at my dev folder:

There is hope for me yet then :) Ha I also broke so many things growing up too, well they worked but did not fit back together!

When you find a project interesting you just keep going :)
[image]

visualapproach, 2020-09-07:
>There is hope for me yet then

You bet!

1.27 uploaded btw 👍

877dev, 2020-09-07:
Awesome 👍

closed 2020-09-16 by visualapproach