visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #19: v1.27 Failed to open config file (load)

877dev, 2020-09-08:
Morning,

Had a quick try of this, but getting an error. I flashed sketch then uploaded sketch data. Maybe something I am doing?

`10:43:35.490 -> IP address:	192.168.1.133
10:43:35.537 -> 
10:43:39.241 -> *WM: freeing allocated params!
10:43:39.241 -> OTA ready
10:43:39.241 -> 
10:43:39.428 -> LittleFS started. Contents:
10:43:39.428 -> 
10:43:39.428 -> WebSocket server started.
10:43:39.428 -> HTTP server started.
10:43:39.428 -> Failed to open config file (load)
10:43:39.428 -> Failed to open appdata file
10:43:39.756 -> 2020-09-08 09:43:40
`

visualapproach, 2020-09-08:
Nothing wrong. You have to save config first. Then restart and you shouldn’t get that error. Appdata will be saved after 24h

877dev, 2020-09-08:
I'm getting 404 File not found error on web page too.

visualapproach, 2020-09-08:
Webpage should work anyway I think.

faboaic, 2020-09-08:
Maybe the transition from SPIFFS to LittleFS killed all files and you did not upload the files to LittleFS after the update?

877dev, 2020-09-08:
I used this to upload, same a SPIFFs. Maybe wrong?
[image]

877dev, 2020-09-08:
Yup I needed to install this:
https://github.com/earlephilhower/arduino-esp8266littlefs-plugin#:~:text=Go%20to%20sketch%20directory%20(choose,in%20the%20file%20system%20there.&text=Select%20Tools%20%3E%20ESP8266%20LittleFS%20Data,into%20ESP8266%20flash%20file%20system.

Working now thank you :)

visualapproach, 2020-09-08:
Yes that’s your problem. You need to download littlefs tool. Then you will have a new option “upload littlefs data” or similar. Sorry, I think I wrote about it on readme but I could’ve been clearer.

877dev, 2020-09-08:
Readme's are for wimps! :)

visualapproach, 2020-09-08:
Ok great 🎉

visualapproach, 2020-09-09:
patch 1 uploaded. Hope it's more stable. Had a couple of reboots, probably caused by calling yield() from other functions than loop()

closed 2020-10-26 by visualapproach