visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #24: PCB?

Naesstrom, 2020-09-11:
I was working on a pcb mount for this but I'm not sure about one thing. Shouldn't Pin6 on the CIO be connected to something to?

You can see the PCB here: https://easyeda.com/Naesstrom/lay-z-spa_remote
[image]

visualapproach, 2020-09-11:
Hi @Naesstrom , Pin 6 is just outgoing sound. Which is not presently used in code. We are making our own sound.

Naesstrom, 2020-09-11:
Sweet thanks, then I'll order some PCB's
Or are there any changes coming up before I get them etched? I saw some discussion about adding extra pins for various things

visualapproach, 2020-09-11:
No immediate changes in progress regarding extra pins, but why not make a couple of traces from D0, D8 and GND to through holes somewhere on the PCB.

Naesstrom, 2020-09-11:
Great idea!
[image]

carverhaines, 2020-09-11:
Make sure the dsp and cio have enough room around them to plug in, the plastic around the plugs takes a bit of room.

carverhaines, 2020-09-11:
Sorry, just realized you may be planning on using a cable, I am using these currently soldered into my pcb so I can plug in directly.

[image]

Naesstrom, 2020-09-11:
> Sorry, just realized you may be planning on using a cable, I am using these currently soldered into my pcb so I can plug in directly.
> 
> [image]

Ohh, do they fit straight into the JST connectors?

I bought a [set of male/female connectors](https://thepihut.com/products/6-pin-jst-sm-plug-receptacle-cable-set) with cables this time since I couldn't find JST-SM connectors as PCB mounted?
http://www.jst-mfg.com/product/pdf/eng/eSM.pdf

carverhaines, 2020-09-11:
Yes, it does, that's how I have it setup

Naesstrom, 2020-09-11:
Oh thats nice, I moved them over a bit now, do you think your kind of connectors would fit the design now?
[image]

carverhaines, 2020-09-11:
It won't lock into place like the connectors, but is a tight enough fit I'm not worried

carverhaines, 2020-09-11:
Should work. Any chance you can make a second design that uses two 4 I/O LVCs for those of us that can't get ahold of that specific LVC (just needs 2 3v inputs, 2 HV inputs, and 4 gnd) 
[image]

Naesstrom, 2020-09-11:
Absolutely, does this look right?
[image]

You can check the schematics at https://easyeda.com/Naesstrom/lay-z-spa_remote to, and edit and download gerbers for production!

877dev, 2020-09-11:
Very nice @Naesstrom , thanks for sharing. I might adapt them for Wemos D1 mini.

One thing, the resistors are showing as 10K where they should be around 0.5-1K according to readme.

carverhaines, 2020-09-11:
@Naesstrom , that looks perfect! I can't wait to print one.
@visualapproach , these 2 schematics should be included in the repo for anyone wanting to print their PCBs

carverhaines, 2020-09-11:
Just put in my order to PCBway!

Naesstrom, 2020-09-12:
> Very nice @Naesstrom , thanks for sharing. I might adapt them for Wemos D1 mini.
> 
> One thing, the resistors are showing as 10K where they should be around 0.5-1K according to readme.

I started doing it for a d1 actually but became unsure if I could feed it with 12V as I guess the VIN/HV side is getting here!
Bloody hell, my bad with the resistors. Changing them now 👍 

> Just put in my order to PCBway!
Nice, I'll order mine this weekend to!

visualapproach, 2020-09-12:
I put the link in readme, thank you.

closed 2020-09-16 by visualapproach