visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #25: Updates

visualapproach, 2020-09-12:
I'll post here when I update the code.

visualapproach, 2020-09-12:
2020-09-12: updated webpage and changed some code to get rid of some automode bugs I discovered. Time will tell if it works. I could wait to post my updates until after testing some time, but then I would be the only one testing. Another idea is to have a "release" and "beta" branch. That sounds like more work to me, but what do you guys think?

visualapproach, 2020-09-12:
2020-09-12 14:44 UTC - More small updates.

877dev, 2020-09-14:
Am I correct, there is no way to upload the LittleFS data OTA?

visualapproach, 2020-09-14:
yes you can. Just select ip-port same way as with sketch OTA

visualapproach, 2020-09-14:
I'll look at your mqtt code today I hope. I'm also struggling with a bug in automode. Should be trivial, but this non blocking code spiced with interrupts makes it really hard sometimes. Thing aren't necessarily happening in the order one might think...

877dev, 2020-09-14:
No problems, I'm just on a roll today!

The problem was having enabled the OTA password, for some reason LittleFS does not like that (I can't see how to enter password for that).

I just flashed the v003 mqtt code to my live system, will let you know how it goes.

visualapproach, 2020-09-14:
Great, I was just skimming through your code. If it works it is really nice. Just thinking we might get memory issues with a tiny stack of just 4KB. To minimize ram usage, I put all literal strings in F(), like Serial.print(F("blablabla")); Also I'm worried that setting MQTT buffer to 2048 will cause memory issues. It did when I set JSON doc to that size. We could optimize the payload easily by reducing human readability. (Use abbreviations, and/or storing booleans in a uintXX_t)
Just my initial thoughts. And wouldn't it be nice to put the credentials on AP-mode portal instead! I'm not sure how to do that yet, so it'll have to wait until the rest is working.

877dev, 2020-09-14:
> To minimize ram usage, I put all literal strings in F(), like Serial.print(F("blablabla"))
Yes sounds good, I use the debug prints as of course serial prints are not good when connecting OTA.

No worries for now, I'll keep having a play! :)

visualapproach, 2020-09-14:
> Am I correct, there is no way to upload the LittleFS data OTA?

Also, for one or two files, use /upload.html 
That way you don't get a reboot

877dev, 2020-09-14:
That's much better I forgot about that, thanks.

visualapproach, 2020-09-14:
Upload this (unzipped) to ESP and get a fake serial output by 
webSocket.broadcastTXT("hi I'm debugging");

[debug.zip](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/5219547/debug.zip)

877dev, 2020-09-14:
Cool! But I just get this:
[image]

visualapproach, 2020-09-14:
[debug.zip](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/5219641/debug.zip)
New debug file... the old expected a JSON. This is general.

visualapproach, 2020-09-14:
[image]
This is manual mode. Otherwise yes, but you have to check all boxes. In your pic the heater is only on between 00:00 and 01:00

877dev, 2020-09-14:
Hmm same with that one, I am going to http://192.168.1.135/debug.html
correct?

> [image]
> This is manual mode. Otherwise yes, but you have to check all boxes. In your pic the heater is only on between 00:00 and 01:00

Ah so auto mode is always on, that's fine.

visualapproach, 2020-09-14:
No, when automode is checked, it follows the config settings. AutoMode OFF = it never changes anything automatically.

877dev, 2020-09-14:
Ok I understand now :)
It's very hard to describe something that everyone understands the same way!

visualapproach, 2020-09-14:
Haha, I better change the labels...

visualapproach, 2020-09-14:
[image]

visualapproach, 2020-09-14:
> Hmm same with that one, I am going to http://192.168.1.135/debug.html
> correct?
> 

Yes, then right click / inspect, to show "console". Press the tab "console" and anything sent over websockets will be displayed here.

877dev, 2020-09-14:
Nice one. looks easier to understand for people like me :)

Got console working now, that's really useful!

visualapproach, 2020-09-15:
patch 2 uploaded.

877dev, 2020-09-15:
> patch 2 uploaded.

Nice! Can I ask what was changed? It would help me knowing as I can easily merge it into my MQTT test code. 

Also I’m part way through making my own MQTT interface with Blynk:

[image]

visualapproach, 2020-09-15:
How DARE YOU turn off mr Frankenstein?!

visualapproach, 2020-09-15:
And what does holiday mode do?

visualapproach, 2020-09-15:
> > patch 2 uploaded.
> 
> Nice! Can I ask what was changed? It would help me knowing as I can easily merge it into my MQTT test code.

Main thing is I moved releasedButtons() further up in validateButtons()
I might have done other things but you have to check file dates or compare the files in github.
But does it really matter? Your code is more or less stand alone with exception for a few lines. Should be easy to inject in this version, right?
Good work if I didn’t say so

877dev, 2020-09-15:
> And what does holiday mode do?

Oh those tabs across the top are for different things, not the hot tub. Holiday mode is for switching on lights around the house at certain times while we are away. 

Basically that one iOS app can control all my automations.

visualapproach, 2020-09-16:
@877dev (and others)
Please try my latest MQTT branch. It is v1.27p2 with 877dev's MQTT-code massaged into the project.
Don't forget to upload new data files to LittleFS.
OTA password is "esp8266" by default. (was off before)
Enter config.html and click mqtt and enter your credentials. Save.
Activate mqtt on main page.

877dev, 2020-09-16:
Yay this sounds great 👍

visualapproach, 2020-09-21:
v1.28 up in the MQTT branch.
 * v1.28 
 * MQTT version
 * Will pull to MASTER branch after feedback from you
 * Enter credentials in TAB "User_parameters.h"
 * Also comment or uncomment the defines you want there.
 * Removed "use mqtt" checkbox from main web page since you decide whether or not to use it from "user_parameters"

dave2802, 2020-09-22:
Is there anywhere we can change the frequency of the messages it sends to the mqtt server?
I believe it currently sends a message every 1 second, personally I think every 30 seconds to 1 minute would be fine.

great work on the mqtt version so far 👍

visualapproach, 2020-09-22:
@dave2802 Hi, thanks. I’m on my phone now but in the startup tab, you can set a second timer to 30 instead of 1.4. It will also slow the web sockets.  Can’t remember if it affect anything else but I will look at it in a couple of days.

dave2802, 2020-09-22:
@visualapproach....perfect thanks for the info & fast response.

visualapproach, 2020-09-29:
Pulled v1.28 to master branch. It has been working a week on my system now. I might have overwritten the previous v1.27 with an MQTT test version in the PULL request... Well, I think 1.28 is better anyways :-)
If someone for some weird reason wants it back, you have to tell me.

visualapproach, 2020-10-18:
BWC_v2.0.0-beta branch is up. A lot of time and effort went into it. I think it will be much easier for other people to modify the code now and the biggest change is a command queue and all communication is made with JSON strings. All the bitbanging and other distracting code is hidden from view in a library now. I tried to implement all your wishes, like not having to unlock device (except when using the real display) and the commands are not TOGGLES anymore. Just state the desired state. 
MQTT users - Some changes in communication protocols were made, so you have to make new MQTT/node red nodes. I'll try to upload some templates later.

staylo, 2020-10-22:
It's clear a lot of work went into the new version, thanks very much for your efforts!

closed 2021-06-14 by visualapproach