visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #26: v.1227 p2 possible bugs

877dev, 2020-09-16:
Just testing for a while, seems like there may be a couple of issues, or may be I am using it incorrectly.

1) After reboot "auto mode" selection is not remembered, defaults to ON. So if the ESP reboots for some reason, the heater will likely go off.

2) When pressing up or down:
"Actual temp" changes instead of "Target"
"Actual temp" flashes "0" a couple of times.
"Actual temp" shows correct value again 
"Target temp" gets updated to new value

Thanks

visualapproach, 2020-09-16:
@

> Just testing for a while, seems like there may be a couple of issues, or may be I am using it incorrectly.
> 
> 1. After reboot "auto mode" selection is not remembered, defaults to ON. So if the ESP reboots for some reason, the heater will likely go off.

I'll sort it out :-). Meanwhile, check all boxes in allowed heater hours if you want it on all the time even after reboot.

> 2. When pressing up or down:
>    "Actual temp" changes instead of "Target"
>    "Actual temp" flashes "0" a couple of times.
>    "Actual temp" shows correct value again
>    "Target temp" gets updated to new value
> 
> Thanks

Yes. Although it might be possible to change this behavior, I'll instead explain why it does that:
The display is just a "dumb" display. The actual temperature or target temp only exist within the CIO (pump MCU). So for the ESP to get the values it "reads" what is beeing displayed and translate the combinations of LEDs ON/OFF to digits.  And then transmit it to the web page or whatever. When you press UP/DOWN the display starts flashing and the ESP waits for a blank screen and then capture whatever number is displayed next time, and assumes that it is the target temp. So the numbers on the web page is mirroring what your real display is actually doing, Thanks for testing and sharing. I'll think about it. I still need to be able to send error messages like E02 to web page. Maybe just mirror the real display as is... We'll see if I get motivated some day

visualapproach, 2020-09-16:
1. Done

visualapproach, 2020-09-16:
2. Done. (yeah, I was just whining ;-)) If you are happy - close this issue @877dev

877dev, 2020-09-16:
Cool I’ll check it out as soon as I’m home 👍

877dev, 2020-09-16:
Darn the family are using the hot tub so I am waiting to flash the latest version, looks nice!

Thanks for the explanation above regarding the changing temp values, makes sense now.

Question:
If the ESP is reading only from DISP, how is the web interface showing the different "Actual" and Target" values? I gather there is no way to read from the CIO directly, I seem to recall from your excel sheet that the CIO interrogates the DISP for it's values actually.

visualapproach, 2020-09-16:
Can’t you just throw in a fat spider or something ? I mean, for science man!

Answer: the CIO keeps asking for what buttons are pressed. Or touched rather. And every now and then it sends a stream of ones and zeroes to tell the display which LEDs to light and at what brightness. The actual and target temp is not there directly but can be derived. It’s done in getchar() by converting certain LED combinations to a character and then to a value. If we do this after up/down has been pressed we know that the target temp will blink. So we wait for DSP to go blank once, then next time it shows a value we store it in “set_temp_val”. Any other situations we store it in “cur_tmp_val”.

visualapproach, 2020-09-16:
And at startup the ESP emulates a DOWN command to fool the the CIO to reveal it’s secrets..

877dev, 2020-09-16:
> Can’t you just throw in a fat spider or something ? I mean, for science man!

It worked! :)
Makes sense thanks. Now onto testing, I OTA'd LittleFS then the sketch, all seemed ok.

The webpage is quite slow going to the config page, and to the mqtt page. When I enter settings and press save, I get a nice little "Saved" confirmation box. It takes around 20-30seconds to return to the main page.

(Not sure, should I reboot the ESP after saving a new config? I feel I should know this by now!(.

It returns to the main page, but it's super slow, and loads blank: 
[image]

When I return to the the config and mqtt pages, they are back to default/blank settings.
So something is not saving correctly, or it's me.

Out for 15mins, check back soon!

visualapproach, 2020-09-16:
Try new code now and if it is still slow wait for main page to load. Flip mqtt checkbox and see.

877dev, 2020-09-16:
"event" commented out in loop, now getting a "redefinition of 'Ticker tickerminute" compile error.

visualapproach, 2020-09-16:
I did that from my phone. New try uploaded now. It compiles fine.

877dev, 2020-09-16:
Yup sending now..

visualapproach, 2020-09-16:
Tried, but as soon as MQTT is on its super slow... Even tried 1024 buffer. Wonder if MQTT library is slow or takes up too much mem. I'm out for today. Cheers

877dev, 2020-09-16:
Don't hate me, but I preferred it before when the mqtt was configured via the sketch.  After all, if you are using mqtt you know how to change a few lines of code. And it would reduce the overheads.

My ideal scenario would be a "define" statement to choose either mqtt, websockets or both.
As I personally will only likely use mqtt in the long run, and I think many other mqtt users may do the same. Most people in that world have their own front end (Blynk, Home Assistant etc..).

It was a good idea to configure via web interface (Tasmota does that too), but it seems to kill it as you said.

I am very aware it's your baby, so I'm more than happy to configure code to my own usage. But this latest version makes it a lot harder to unpick the mqtt code from the rest.

Hope this is not too disappointing, the project is great and I appreciate your work :)
Enjoy your evening.

visualapproach, 2020-09-17:
Don't worry, it's just a matter of taste. I thought it would be nice to offer a one-in-all solution that worked out of the box for different use cases. But you got a point in that mqtt-people probably won't use websockets anyway and know how to change things. If someone else build upon and improves the code I'm just happy. My most important job was to crack the communication and share it for others to enjoy. The rest is just dressing up the pig.
But anyhow, I'd like it to WORK, and I really don't think the web interface for mqtt is what is killing it. I might be wrong, but I suspect it's memory or the mqtt.loop/connect is taking to long. 
2 questions:
1. Did you get it to play nicely in your version? Did the webpages load quickly then?
2. Did it connect to MQTT in my version or was it complaining over no connection? (Because your broker wasn't listening or other reason)

877dev, 2020-09-17:
Cool, yes the communication is still baffling to me! Dressing up the pig - lol.

I can't imagine the issue is the mqtt library, the chap who wrote it also created Node Red, and it is a well used and respected library. But you never know if there is some issue.

1) Yes in my version it works great, same as without the mqtt code when using the web interface.

2) In your newest code, it's nice and responsive **until** saving mqtt config. 

Here's what happens after saving mqtt config:
---------------------------------------------------

Then it takes around 30-60 seconds to reload the main page, and then maybe 1-3minutes to populate the button properly. 

If I then tick "MQTT enabled" button, the ESP never publishes to the broker, meaning it did not make a connection.

I then rebooted the ESP, it was very slow to reboot, the page loads with no values loaded.
When it loads, the power/filter/heater are all off. Target temp is 0.

[image]

Clicking on config, it loads with timezone 2 and some kWh value, then after 10seconds reverts back to timezone 0 and 1kWh.

Then click on mqtt config, slow to load and the page has no values.

I'm not sure but it seems the values stored in LittleFS are going wrong, is your timezone 2? As it might be loading some old values...

closed 2020-09-19 by 877dev

visualapproach, 2020-09-19:
It's just the default placeholder values in the .html file. It's showing until you get real data wich you don't in this case.