visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #27: not declared in this scope

eddcaton, 2020-09-29:
Hey, 

Love the project and I ma trying to upload to my NodeMcu but I am getting a lot of not declared in this scope.

ESP12E_Bestway_controller_v1.27p2:28: error: 'startup' was not declared in this scope
startup();

ESP12E_Bestway_controller_v1.27p2:151: error: 'schedule' was not declared in this scope
schedule(); //check automated tasks

ESP12E_Bestway_controller_v1.27p2:158: error: 'sendWSmessage' was not declared in this scope
sendWSmessage(); //to webclients

ESP12E_Bestway_controller_v1.27p2:159: error: 'savelog' was not declared in this scope
savelog();       //saves status variables to LittleFS

ESP12E_Bestway_controller_v1.27p2:160: error: 'saveappdata' was not declared in this scope
saveappdata();   //saves uptime, heating time etc to LittleFS

ESP12E_Bestway_controller_v1.27p2:323: error: 'enterAPmode' was not declared in this scope
enterAPmode();

Please could you advise?

visualapproach, 2020-09-29:
Hey! Thanks. 
I sometimes get that when I hit upload/verify button before preceeding attempt is finished. A restart of the IDE might help.
Or did you alter anything in the program? Like changing the names of the tabs? That could potentially lead to things getting out of scope since the IDE puts everything in alphabetic order. Hence "a_globals".
Hope it will work for you.

eddcaton, 2020-09-30:
Hey, 

Still no success.

I haven't changed any of the code, any other suggestions?

Edd

visualapproach, 2020-09-30:
Hi, Arduino IDE 1.8.13 and latest (this morning) BWC_v1.28 files compiles fine here. WIthout knowing what the problem is, try this:
- Use IDE v1.8.13
- Download project again and try (make sure correct board is selected)
- if not working, try other projects. Do they work? Maybe reinstalling IDE.

eddcaton, 2020-09-30:
Thanks for the reply. 

I have updated my IDE to the latest version, downloaded the new BWC_v1.28 file that was uploaded today. I am now getting a lot of JSON errors particularly in the filestuff.ino and also the wifi stuff.ino.

I have also uploaded the blink example sketch to the Node MCU and it worked fine. 

There is a high possibility i am being an idiot and missing something...

visualapproach, 2020-09-30:
If you were I don’t think you’d find this project, so let’s rule that out.
Can you post a screenshot of the IDE so I can see all tabs. (Shift prtscr, paste in paint and save, then drag file to bottom of the reply-window)

eddcaton, 2020-09-30:
[image]

eddcaton, 2020-09-30:
[image]

visualapproach, 2020-09-30:
[test.zip](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/5307796/test.zip)

try this. What OS are you on?

eddcaton, 2020-09-30:
Hey, 

I am using PC to do the compiling and uploading. Just removing into the PC with it on tonight as I am away from home.

the test.zip just threw up my previous errors:
`ESP12E_Bestway_controller_v1.27p2:28: error: 'startup' was not declared in this scope
startup();

ESP12E_Bestway_controller_v1.27p2:151: error: 'schedule' was not declared in this scope
schedule(); //check automated tasks

ESP12E_Bestway_controller_v1.27p2:158: error: 'sendWSmessage' was not declared in this scope
sendWSmessage(); //to webclients

ESP12E_Bestway_controller_v1.27p2:159: error: 'savelog' was not declared in this scope
savelog(); //saves status variables to LittleFS

ESP12E_Bestway_controller_v1.27p2:160: error: 'saveappdata' was not declared in this scope
saveappdata(); //saves uptime, heating time etc to LittleFS

ESP12E_Bestway_controller_v1.27p2:323: error: 'enterAPmode' was not declared in this scope
enterAPmode();`

eddcaton, 2020-09-30:
Not sure if it makes any difference but.
When I open ESP12E_Bestway_controller_v1.27p2.ino there are no other tabs in the IDE.
But when i open BWC_v1.28.ino all the tabs are there (like my screen shot)

visualapproach, 2020-09-30:
Seems weird! Try reinstalling ide. Could also try copying all tabs into one. In the order of #include then in the order from left to right if you get what I mean.

eddcaton, 2020-10-01:
Hey After re-install I am getting the JSON errors again. 

ilestuff:11:3: error: 'StaticJsonDocument' was not declared in this scope
   StaticJsonDocument<1024> doc;
   ^
filestuff:11:28: error: 'doc' was not declared in this scope
   StaticJsonDocument<1024> doc;
                            ^
filestuff:14:3: error: 'DeserializationError' was not declared in this scope
   DeserializationError error = deserializeJson(doc, file);
   ^
filestuff:14:24: error: expected ';' before 'error'
   DeserializationError error = deserializeJson(doc, file);
                        ^
filestuff:15:7: error: 'error' was not declared in this scope
   if (error) {
       ^
C:\Users\Edd\Documents\Arduino\libraries\BWC_v1.28\filestuff.ino: In function 'void saveConfiguration()':
filestuff:50:3: error: 'StaticJsonDocument' was not declared in this scope
   StaticJsonDocument<1024> doc;
   ^
filestuff:50:28: error: 'doc' was not declared in this scope
   StaticJsonDocument<1024> doc;
                            ^
filestuff:63:30: error: 'serializeJson' was not declared in this scope
   if (serializeJson(doc, file) == 0) {
                              ^
C:\Users\Edd\Documents\Arduino\libraries\BWC_v1.28\filestuff.ino: In function 'bool loadmqtt()':
filestuff:171:3: error: 'StaticJsonDocument' was not declared in this scope
   StaticJsonDocument<1024> doc;
   ^
filestuff:171:28: error: 'doc' was not declared in this scope
   StaticJsonDocument<1024> doc;
                            ^
filestuff:174:3: error: 'DeserializationError' was not declared in this scope
   DeserializationError error = deserializeJson(doc, file);
   ^
filestuff:174:24: error: expected ';' before 'error'
   DeserializationError error = deserializeJson(doc, file);
                        ^
filestuff:175:7: error: 'error' was not declared in this scope
   if (error) {
       ^
C:\Users\Edd\Documents\Arduino\libraries\BWC_v1.28\filestuff.ino: In function 'void savemqtt()':
filestuff:217:3: error: 'StaticJsonDocument' was not declared in this scope
   StaticJsonDocument<1024> doc;
   ^
filestuff:217:28: error: 'doc' was not declared in this scope
   StaticJsonDocument<1024> doc;
                            ^
filestuff:231:30: error: 'serializeJson' was not declared in this scope
   if (serializeJson(doc, file) == 0) {
                              ^
C:\Users\Edd\Documents\Arduino\libraries\BWC_v1.28\wifistuff.ino: In function 'void sendMessage(int)':
wifistuff:214:3: error: 'StaticJsonDocument' was not declared in this scope
   StaticJsonDocument<1200> doc; //I hope this is enough
   ^
wifistuff:214:28: error: 'doc' was not declared in this scope
   StaticJsonDocument<1200> doc; //I hope this is enough
                            ^
wifistuff:235:33: error: 'serializeJson' was not declared in this scope
   if (serializeJson(doc, jsonmsg) == 0) {
                                 ^
C:\Users\Edd\Documents\Arduino\libraries\BWC_v1.28\wifistuff.ino: In function 'void handleGetConfig()':
wifistuff:266:5: error: 'StaticJsonDocument' was not declared in this scope
     StaticJsonDocument<1024> doc;
     ^
wifistuff:266:30: error: 'doc' was not declared in this scope
     StaticJsonDocument<1024> doc;
                              ^
wifistuff:281:35: error: 'serializeJson' was not declared in this scope
     if (serializeJson(doc, jsonmsg) == 0) {
                                   ^
C:\Users\Edd\Documents\Arduino\libraries\BWC_v1.28\wifistuff.ino: In function 'void handleSetConfig()':
wifistuff:297:5: error: 'StaticJsonDocument' was not declared in this scope
     StaticJsonDocument<1024> doc;
     ^
wifistuff:297:30: error: 'doc' was not declared in this scope
     StaticJsonDocument<1024> doc;
                              ^
wifistuff:299:5: error: 'DeserializationError' was not declared in this scope
     DeserializationError error = deserializeJson(doc, message);
     ^
wifistuff:299:26: error: expected ';' before 'error'
     DeserializationError error = deserializeJson(doc, message);
                          ^
wifistuff:300:9: error: 'error' was not declared in this scope
     if (error) {
         ^
C:\Users\Edd\Documents\Arduino\libraries\BWC_v1.28\wifistuff.ino: In function 'void handleGetMqtt()':
wifistuff:329:5: error: 'StaticJsonDocument' was not declared in this scope
     StaticJsonDocument<1024> doc;
     ^
wifistuff:329:30: error: 'doc' was not declared in this scope
     StaticJsonDocument<1024> doc;
                              ^
wifistuff:344:35: error: 'serializeJson' was not declared in this scope
     if (serializeJson(doc, jsonmsg) == 0) {
                                   ^
C:\Users\Edd\Documents\Arduino\libraries\BWC_v1.28\wifistuff.ino: In function 'void handleSetMqtt()':
wifistuff:364:5: error: 'StaticJsonDocument' was not declared in this scope
     StaticJsonDocument<1024> doc;
     ^
wifistuff:364:30: error: 'doc' was not declared in this scope
     StaticJsonDocument<1024> doc;
                              ^
wifistuff:366:5: error: 'DeserializationError' was not declared in this scope
     DeserializationError error = deserializeJson(doc, message);
     ^
wifistuff:366:26: error: expected ';' before 'error'
     DeserializationError error = deserializeJson(doc, message);
                          ^
wifistuff:367:9: error: 'error' was not declared in this scope
     if (error) {
         ^
Multiple libraries were found for "WiFiUdp.h"
 Used: C:\Users\Edd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
'StaticJsonDocument' was not declared in this scope

visualapproach, 2020-10-01:
I don't know if there is a reason your files are located in ".../libraries"? Did you try to install it as a library or something? If so, don't. Just put the BWC_v1.28 folder somewhere like "/sketches" and doubleclick the BWC_v1.28.ino file.
Wish you luck.

eddcaton, 2020-10-01:
Hey, 

I have moved the BWC_v1.28 folder outside of the libraries file and it not just sits in the Arduino folder in documents. 
Still getting 'was not declared in this scope' for the filestuff and wifistuff

visualapproach, 2020-10-01:
I don’t have any more ideas I’m afraid.

eddcaton, 2020-10-01:
We have something working!

So I completely uninstalled IDE and then installed the windows 10 APP from the windows app store. Now it has uploaded fine. 

I have connected to the Node MCU, put my routers login details into the web browser and it is has now been give a IP address and i can ping it fine.

When i put the ip address int my browser it says : 404: File Not Found

Any ideas?

visualapproach, 2020-10-01:
Good. You need to upload the LittleFS files also

closed 2020-10-26 by visualapproach