visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #29: blynk

visualapproach, 2020-10-02:
What is a good way of controlling the SPA or any home automation things from an iPhone?
- I don't want to expose the ESP8266 webserver directly to the internet due to lack of security.
- I want an app-like interface, like the homepage I already built into the ESP8266.
- I have set up a Raspberry Pi 4 B with Docker + MQTT + node red (+grafana + portainer). It works great logging and graphing pool temperature.
I looked at the Blynk app, which @877dev seems to use but if that's the way to go I would like some help setting it up. I have the Blynk app installed on the phone, but there was no RPi4/node red to choose and it's all about controlling pins rather than getting JSON data in and sending values back.
How do you home automation guys do it from outside your home network?

/A home automation wannabe with just a hot tub and a lamp to control :rofl:

877dev, 2020-10-02:
Hey there, 

Now that you have virtually identical setup to me, you may as well set Blynk up! 👍🏻 
Are you using IOTstack btw?

Here's my latest view, which sync's with the webserver, and the LCD is live and can display custom messages if wanted:

[image]

I actually just submitted a pull request to IOTstack with some improvements to Blynk local server. But if you want to have it internet facing, you can just use Blynk with their cloud. Just start from [here](https://gist.github.com/877dev/cfa52044336e1e44d74156ce8b1458aa#iosandroid-app-setup) and use Blynk instead of custom when setting up.

877dev, 2020-10-02:
In Blynk app at top right, scan this QR code for a template of mine.

[image]

In node red, import this flow:
`[{"id":"34cc25e0.5cbb5a","type":"mqtt in","z":"5a9d8292.529c2c","name":"","topic":"BW-1.27_MQTT/message","qos":"1","datatype":"json","broker":"","x":520,"y":1900,"wires":[["36ddc5d6.3661ea","2282e6ef.0a3daa","df534aa.1639bb8"]]},{"id":"615889ab.0cddf8","type":"blynk-ws-out-lcd","z":"5a9d8292.529c2c","name":"Hot tub LCD","pin":"70","client":"","x":1330,"y":1780,"wires":[]},{"id":"2282e6ef.0a3daa","type":"function","z":"5a9d8292.529c2c","name":"Prefix current and target temps","func":"//Top line\nmsg.text  = \"Current temp:\" + msg.payload.temp + \"    \";\n//Bottom line\nmsg.text1 = \"Target temp :\" + msg.payload.target + \"    \";\nreturn msg;\n","outputs":1,"noerr":0,"x":860,"y":1780,"wires":[["615889ab.0cddf8"]]},{"id":"c3f3c100.8f2b8","type":"blynk-ws-in-write","z":"5a9d8292.529c2c","name":"Write event (all pins)","pin":"0","pin_all":true,"client":"","x":550,"y":2210,"wires":[["735b9924.e240e8"]]},{"id":"735b9924.e240e8","type":"function","z":"5a9d8292.529c2c","name":"Check pins 71-79 and change payload","func":"const pin = msg.pin;\n\nif ((pin > 70) && (pin < 80))   //filters out all other Blynk pins\n{\nswitch (parseInt(pin)) {    //convert string to integer for ease\n  case 71:\n    msg.payload = \"0x5\";    //power\n    break;\n  case 72:\n    msg.payload = \"0x6\";    //up\n    break;\n  case 73:\n     msg.payload = \"0x7\";   //down\n    break;\n  case 74:\n    msg.payload = \"0x8\";    //filter\n    break;\n  case 75:\n    msg.payload = \"0xA\";    //heater\n    break;\n  case 76:\n    msg.payload = \"0xB\";    //units\n    break;\n  case 77:\n    msg.payload = \"0xC\";    //air\n    break;\n  case 78:\n    msg.payload = \"0xD\";   //timer\n    break;\n  case 79:\n    msg.payload = \"0xE\";   //lock\n  }\n}\n\nelse {\n    msg.payload = \"0xFF\";   //unused value, blank payload switches heater off for some reason\n}\n\nreturn msg;\n","outputs":1,"noerr":0,"x":830,"y":2210,"wires":[["7f7aa9a0.25d938","81a2291b.849d28"]]},{"id":"7f7aa9a0.25d938","type":"mqtt out","z":"5a9d8292.529c2c","name":"","topic":"BW-1.27_MQTT/command","qos":"1","retain":"","broker":"","x":1140,"y":2210,"wires":[]},{"id":"81a2291b.849d28","type":"debug","z":"5a9d8292.529c2c","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","x":1090,"y":2260,"wires":[]},{"id":"b3d1fe9a.5e30a","type":"comment","z":"5a9d8292.529c2c","name":"Monitor Blynk pins and change payload to suit","info":"","x":830,"y":2150,"wires":[]},{"id":"6fbacabc.224d84","type":"blynk-ws-out-write","z":"5a9d8292.529c2c","name":"Blynk (dynamic pin by code)","pin":"0","pinmode":"1","client":"","x":1310,"y":1900,"wires":[]},{"id":"36ddc5d6.3661ea","type":"debug","z":"5a9d8292.529c2c","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","x":770,"y":1970,"wires":[]},{"id":"df534aa.1639bb8","type":"function","z":"5a9d8292.529c2c","name":"NEW -Check incoming and send to appropriate Blynk pin","func":"const pins = {\n    power: 71,\n    filter: 74,\n    heat: 75,\n    celsius: 76,\n    air: 77,\n    locked: 79,\n}\n\n\nconst msgPower = {\n    payload: msg.payload.power === true ? 1  :  0,\n    pin: pins.power,\n}\n\nconst msgFilter = {\n    payload: msg.payload.filter === true ? 1  :  0,\n    pin: pins.filter,\n}\n\nconst msgHeat = {\n    payload: (msg.payload.heating || msg.payload.heater) === true ? 1  :  0,\n    pin: pins.heat,\n}\n\nconst msgCelsius = {\n    payload: msg.payload.celsius === true ? 1  :  0,\n    pin: pins.celsius,\n}\n\nconst msgAir = {\n    payload: msg.payload.air === true ? 1  :  0,\n    pin: pins.air,\n}\n\n\nconst msgLocked = {\n    payload: msg.payload.locked === true ? 1  :  0,\n    pin: pins.locked,\n}\n\n\nreturn [ [ msgPower, msgFilter, msgHeat, msgCelsius, msgAir, msgLocked ] ];\n","outputs":1,"noerr":0,"x":920,"y":1900,"wires":[["6fbacabc.224d84"]]},{"id":"f22ba8b.e4d7b58","type":"comment","z":"5a9d8292.529c2c","name":"MQTT from hot tub","info":"","x":520,"y":1870,"wires":[]},{"id":"5f6c6850.65fba8","type":"comment","z":"5a9d8292.529c2c","name":"Filter payloads and pass onto appropriate Blynk virtual pin","info":"","x":920,"y":1860,"wires":[]},{"id":"872ec3af.3943f","type":"comment","z":"5a9d8292.529c2c","name":"Pass temperatures to Blynk LCD widget","info":"","x":870,"y":1740,"wires":[]},{"id":"8bc9493f.4919c8","type":"comment","z":"5a9d8292.529c2c","name":"Blynk LCD widget","info":"","x":1330,"y":1740,"wires":[]},{"id":"c610390d.7133c8","type":"comment","z":"5a9d8292.529c2c","name":"To all other pins","info":"","x":1320,"y":1860,"wires":[]}]`

You need to edit MQTT broker settings, and Blynk server settings.

visualapproach, 2020-10-02:
Wow, thanks a lot for your answer! The QR-code told me I had not enough energy. I can agree, but DOH! Okay, I guess I need to set up a local server, and I hope I will be able to reach it through my router somehow. I call you back when I have digested it all. Most likely with more questions ;-)
And yes I installed the IOTstack.

visualapproach, 2020-10-02:
First issue: no config folder is created when Blynk-server installed from bash menu.sh

877dev, 2020-10-02:
Yes that's what my PR is to fix. 

If you look here:
https://gist.github.com/877dev/cfa52044336e1e44d74156ce8b1458aa#file-blynk-setup-guide-md

In Portainer stop Blynk and delete the image. 

Create the Blynk config dir. 

Create server.properties and mail.properties in that dir and edit as needed. 

Add the extra line in my service.yml to your docker-config.yml file in IOTstack dir. 

Up the stack to recreate Blynk container and run. 

Should be working now. 

I'm not at pc so some info might be iffy, let me know 👍

visualapproach, 2020-10-03:
Okey, thanks, I'm traveling so I'll check it when I'm back home again

877dev, 2020-10-03:
Actually I thought of an easier way, I'll post it later today if I have time after work.

visualapproach, 2020-10-03:
No rush, I’m not home until tomorrow evening.

877dev, 2020-10-03:
I'm assuming you know your way around a command line, let me know if you get stuck :)

## Remove old data:
1) Stop the Blynk container `docker-compose -f ~/IOTstack/docker-compose.yml stop blynk_server`
2) Delete the Blynk container `docker-compose -f ~/IOTstack/docker-compose.yml rm -f blynk_server`
3) Delete all old data `sudo rm -rf ~/IOTstack/volumes/blynk_server` 
4) In Portainer, go to images and delete the old Blynk image (may not need to do this, but just in case)

## Prepare my new files:

Extract and copy the zip files to  `~/IOTstack/.templates
[IOTstack.zip](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/5321949/IOTstack.zip)

You may find it easier to create them this way:
`sudo nano ~/IOTstack/.templates/blynk_server/directoryfix.sh` to create the file, and copy/paste the contents in (right click to paste, ctrl-x then Y to save).

Same for Dockerfile and service.yml.

Now run this command to make the script executable:
'sudo chmod +x  ~/IOTstack/.templates/blynk_server/directoryfix.sh'

## Recreate Blynk container with config files:

Run the menu again `./menu.sh`  > Build Stack and make sure Blynk is selected.
Continue, selecting "Do not overwrite", except for Blynk where you select "Pull full service from template".

From here you are up to date, and can just follow my guide from [Configuring for first run](https://gist.github.com/877dev/cfa52044336e1e44d74156ce8b1458aa#configuring-for-first-run)

Let me know if you spot any improvements.

visualapproach, 2020-10-04:
Hello again! Very kind of you to prepare those files. I followed the instructions above, but I didn't edit the mail.properties because I don't have gmail, so I didn't bother. Then it says to edit server.properties "like you would a normal local server". Yeah, well...  I didn't change anything. Started the stack and portainer shows the blynk server, but I get "ERR_EMPTY_RESPONSE" on IP:9443/admin. Maybe I should have edited those files, but I don't know what I am doing.

visualapproach, 2020-10-04:
maybe I should edit this line:
table.rows.pool.size=100
to 1100, since that's the size of my pool? (just kidding)

877dev, 2020-10-04:
Hi,
Did you check Portainer logs and Blynk logs?

https://gist.github.com/877dev/cfa52044336e1e44d74156ce8b1458aa#troubleshooting

Also you may need to type in the http:// part before the IP address for some reason. 

You don't need to edit server.properties it should just work, no matter the pool size!

visualapproach, 2020-10-04:
Yes, but didn’t see anything suspicious. There was something fishy with the Pi, it hang. Blynk reported running but couldn’t see the admin page even with http://. 
I’ll leave it for today. Cheers

visualapproach, 2020-10-05:
Yay! I'm getting somewhere now. At least I'm looking at the Dashboard now :-)

visualapproach, 2020-10-05:
Ok, I have the APP running and working now on my iPhone. Still inside my network but one step at a time. I wouldn't made it this far if you didn't support me @877dev 
Is there an easy way to change custom server from the app?

877dev, 2020-10-05:
You logged into your custom server like this?

[image]

visualapproach, 2020-10-06:
I didn't see your reply until now, sorry. 
Actually, after watching Andreas Spiess's video I installed pivpn and it is pretty cool! I went with the WireGuard option and it was relatively straightforward to make it work on iPhone. Now when I activate VPN it is like I'm inside my own network and can browse to 192.168.xxx.xxx (for instance my hot tub web page or blynk app)

visualapproach, 2020-10-06:
Blynk account: I don't know yet, there is a lot of info there and I'm on the iPhone now so I will look at it when I have more time. Thx

877dev, 2020-10-07:
Wireguard sounds interesting, I will have to check that out. Maybe you can help me with that if I get stuck 😁

visualapproach, 2020-10-07:
Of course!

visualapproach, 2020-10-07:
I followed the instructions at https://www.pivpn.io/
When asked for upstream DNS IP I entered my isp’s DNS which I found in my router. If using pihole you should probably use a special IP. I don’t know if there is a docker container for pivpn, I just installed it as described in the link

877dev, 2020-10-08:
I made a draft guide on using Wireguard, it was very easy once I figured a few things out:
https://gist.github.com/877dev/95184102400e9055f2775fdce7077a21

visualapproach, 2020-10-18:
Sorry for not replying sooner, I've been working with v2 all free time. Good work on the WireGuard tut. I installed pivpn outside of docker and this is great for when I have time or need to change it.

877dev, 2020-10-19:
No worries, I am very busy at the moment with work and sorting school things out for my kids. Will try v2 when things quieten down 👍

closed 2023-11-23 by visualapproach