visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #31: Question

garrydwilms, 2020-10-10:
Hi. 
This project is great and looks like a lot of work has been put into the decoding. 
I'm still waiting on my level shifter so not fitted mine yet but it's built and flashed and appears to be running and serving webpages. 
Just a quick question if you don't mind, the hot tubs go into idle after 48hours of no use. 
Will this keep it active?. Do you think it would be useful / easy enough to add this as a feature, so it presses a button every day if not used to keep alive or not depending on preference. 

I may look to add it myself as I'm away a lot and if I forget about it, it's cold when I get back. 

Thanks again for your work. 
Garry

visualapproach, 2020-10-10:
Hi @garrydwilms you are welcome and thank you.
I have never got my tub to shut off automatically before I installed this, but keeping it alive is possible as is. Just use auto mode and it should send commands every day to wake it.

877dev, 2020-10-10:
For non-auto users, I guess a simple unlock/lock command every night would keep it alive?

garrydwilms, 2020-10-10:
Thanks folks. 

Maybe it's as easy as this if you don't want to use auto??, in timers.ino :

//executed daily
void dayTimer() {
  saveAppdataFlag = true; //Flash memory only lasts for so many writes (~100 000?)
  //event and appdata is saved every time something changed (check handleData())
unlockDevice(); //trigger an unlock to keep alive
}

visualapproach, 2020-10-10:
Yes :)

877dev, 2020-10-10:
Nice!

visualapproach, 2020-10-26:
Issue is obsolete in v2 where you can add any command to a queue. Config page-> add command "Set pump, 1, 0, 86400" will turn on pump every day.

closed 2020-12-12 by visualapproach