visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #33: Disp not turning on and nothing happening

Naesstrom, 2020-10-13:
I finally got my levelshifter today and mounted everything, flashing went fine once I updated all the libraries 🗡️ 
[image]

Problem is that when I connected it to the unit the display doesn't turn on, the ESP works and I can connect to it but pressing power etc. doesn't turn anything on either?
Disp goes to disp and CIO to the controller so everything seems to be right, if the connections was turned around ie. not 1-1, 2-2 etc. I guess it shouldn't even power on.
[image]

visualapproach, 2020-10-13:
Have you checked that the pin outs in the sketch matches the pin outs on the pcb?

Naesstrom, 2020-10-14:
I'll check, the pcb is based straight off from your schematic tough
[image]

visualapproach, 2020-10-14:
Ok, should not be the problem then I hope. But check as good as you can that there is no shorts between tracks and that there is connectivity where it suppose to. Then I would try the hw-tester program. And if you are sure the connections is correct you could bypass the resistors and see if it makes any difference.

Naesstrom, 2020-10-15:
Yeah something must have been wrong with the PCB, couldn't get a measure between to points and after adding a extra cable it seems to be working!

closed 2020-10-15 by Naesstrom