visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #34: Reboot lost cumulative heating time.

garrydwilms, 2020-10-14:
Hi,
Had this running for a few days now and it's great. It has massively changed the way I manage my temperature as I'm away a lot so can have it ready when I need it without wasting electric. 
One issue I have had is that the system reboots a few times a day, which is fine most of the time, but today it wiped out my cumulative heating time and hence cost data (which is now working OK so not sure what was wrong initially). 
So, is it possible to reload the cumulative heating time from the event log on reboot?
I might look into this as I've not coded for awhile but starting to get the bug again. 

Thanks for the project again, I'm really enjoying playing with it. 

Garry

visualapproach, 2020-10-14:
Hi there!
Glad you have joined and like the project!
It *should* reload it when rebooting. If it doesn't ... hm I don't know why. And regarding the cost, it works for me, but I've had another report of it not working. There is no easy way (yet) to retrieve the data, but I'll keep it in mind. I'm working on a new version and I may as well do something about that. One of my goals with the new version is to make it easier for other people to build on. It will be a library with some example files wich at least looks cleaner. There will also be a much better way to schedule tasks (IMO).
I hope many issues will be gone, but we will have to see until you guys test it out.

garrydwilms, 2020-10-14:
Yeah it was me. But like I say, it's ok now. I was probably just being dumb 🤷🏻‍♂️

visualapproach, 2020-11-21:
I found that Ticker class can't handle more than 71 min periods so it never saved settings.txt which contains uptimes etc. Updated the code now, so cumulative times will be remembered, except for up to three hours prior to the restart.

closed 2020-11-21 by visualapproach