visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #36: Display controller datasheet

staylo, 2020-10-22:
I took a quick look at the ICs on the display PCB. The controller handling the display and buttons is a Shenzhen Titan Micro Electronics TM1628 (It reads as 'TM1628I' to me, but I can't find any details of such an IC).

The datasheet is available from a google search. I think the pull up resistors / anti-noise capacitor circuit in the application circuit matches what's on the display PCB, but it's hard to tell because the traces to the IC are on the reverse side (encased in expoxy)

The only bad news is that from the datasheet data appears to be sent LSB-first, not MSB-first :)

visualapproach, 2020-10-22:
Aha. If you mean it is bad because then it might be the wrong data sheet, I wouldn’t worry about that. The communication doesn’t represent anything that I’m aware of that would make a difference. It’s mostly bitpatterns that can be read from either direction. It may very well be LSB first. I have a feeling that the header packets may represent a memory address to write or read, which would require the right bit order if we cared, but in this case we just replicate it without worrying about the specific values.

visualapproach, 2020-10-22:
Interesting find Simon. There appears to be more brightness levels than I thought but it looks like it fits (backwards) so it is probably a useful data sheet.

visualapproach, 2020-10-25:
Hi again, I updated the code to work in LSBF mode. It doesn't make any difference at all, but it will make things easier if people want to modify it and follow along with the data sheet. Also, some things like the brightness values makes more sense now. The only strange thing is that the CIO sends the _command1_ 0x01 which is not an option according to data sheet. Maybe a special mode. Cheers / Thomas

staylo, 2020-10-25:
Ah, cool :+1: 

On closer inspection the datasheet I found is a real mess. I'm guessing you're looking at the same one. The block diagram doesn't match the pin description, which doesn't match the pin description on the application circuit.. jeez!

I can't find a revised english datasheet, but as far as I can tell the [chinese v1.1 datasheet](http://titanmec.com/index.php/en/project/download/id/522.html) resolves the matter with a table on page 3:

0x00 - 4 grids 13 segments
0x01 - 5 grids 12 segments
0x10 - 6 grids 11 segments
0x11 - 7 grids 10 segments

Notice the description's numbers don't match the english datasheet, which is in error. (This command is controlling whether the output pins 18, 19 and 20 are used for segments or for keypad grid lines... there are only 17 pins to be shared between these two functions so the original datasheet's description of '7 grids 11 segments' is impossible)

No idea where the SEG11 pin has vanished to though! Did they screw up the numbering on the first datasheet and had to stick with it, or is that a weird convention of some kind?

closed 2023-12-04 by github-actions[bot]