visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #37: Node red to influx error

Naesstrom, 2020-10-25:
Looks like somethings wrong with included node-red example. the influxdb node gives the following error:

`Error: A 400 Bad Request error occurred: {"error":"unable to parse 'pooldata air=undefined,celsius=undefined,ch1=undefined,ch2=undefined,ch3=undefined,content=undefined,filter=undefined,heater=undefined,heating=undefined,locked=undefined,power=undefined,target=undefined,temp=undefined': invalid boolean"}`

visualapproach, 2020-10-25:
You have to create an influxdb database, and set the database node accordingly. Also you have to change broker node setting to your own settings. Unfortunately it is not just drag and drop with the nodes, because they are dependent on many other things in your local setup. And you are probably better with node red than I am ;-)

Naesstrom, 2020-10-25:
yeah I've done that, using influx for other things so it should be correct!

Do you have the function working on your node-red?

visualapproach, 2020-10-25:
Yes it is working

closed 2022-03-09 by visualapproach