visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #39: Frequent resets (websocket interface?)

staylo, 2020-11-06:
I've finally got my ESP8266 installed. I'm encountering resets that seem to be often but not always triggered when using the web interface on my Android phone, less often when accessing from Firefox, but I haven't been able to pin down the pattern or cause yet. Is the code on github's considered current? If so I'll get a USB extension cable hooked up to the pump so that I can look at the stack dump. I wasn't sure if anything else is going on behind the scenes :)

Is anyone else is encountering resets around the time that they access the web interface?

visualapproach, 2020-11-06:
If you check out the bootlog.txt, is it sw watchdog resets? It can trigger a reset if some function takes too long. I have turned off the sw wdt in bwc.loop but not the other functions like mqtt and web server etc. It seems it can reset when the connection is slow on either end.

staylo, 2020-11-06:
Yes, they're software watchdog resets, but I disabled the software watchdog and got some hardware watchdog resets. 

WebSockets.h defines a fixed 5000ms timeout, I think the software watchdog is set to 3200ms. So maybe an upstream issue. :)

Currently testing with the websockets calls moved into the main loop. I've disabled the software watchdog and started feeding the hardware watchdog before and after each websocket call. Looks messy but I haven't been able to trigger a reset since.

I'll get the USB cable hooked up and do some further testing.

visualapproach, 2020-11-06:
Great, you are welcome to post your findings. Some things are quite difficult to find. For example mqtt re-connect did not work if launched from a timer function. Really strange.

staylo, 2020-11-06:
Yes, it's a nightmare with hardware watchdog resets especially because you can't see the stack that caused it. I'm still getting some occasional resets, I'll see if I can pin down the cause.

I'd recommend moving any complex code out of the ticker / timer / interrupt functions, there appears to be some known issues:

> It is currently not recommended to do blocking IO operations (network, serial, file) from Ticker callback functions. Instead, set a flag inside the ticker callback and check for that flag inside the loop function.
[](https://arduino-esp8266.readthedocs.io/en/latest/libraries.html)

Could be affecting SaveSettings?

visualapproach, 2020-11-06:
Yes, I think savesettings is called from a timer iirc. Maybe try a flag and add it to the bwc loop.

staylo, 2020-11-07:
The only way I can eliminate hardware watchdog resets is by reducing  WEBSOCKETS_TCP_TIMEOUT to 2000 in WebSockets.h, so I think this is probably an upstream issue. I'll try and identify the issue on the bench with a bare ESP and raise an issue on their repo. 

As far as the spa software goes, with the websockets implementation any timeouts block heavily, so it may be that in the long term a switch to asynchronous mode would be beneficial - but I don't know whether there's a resource cost.

Moving savesettings into the loop seems to have had an effect. Recorded times (uptime, heating etc) were resetting to 0 on reboot usually but now seem to be saving correctly. I'll test further.
-

bertieuk, 2020-11-23:
I get these all the time too :-(

From bootlog.txt:
`{"BOOTINFO":"Software Watchdog 2020-11-23 15:50:29"}
{"BOOTINFO":"Software/System restart 2020-11-23 16:19:18"}
{"BOOTINFO":"Software Watchdog 2020-11-23 16:21:55"}
{"BOOTINFO":"Software Watchdog 2020-11-23 16:25:07"}
{"BOOTINFO":"Software Watchdog 2020-11-23 16:35:28"}
{"BOOTINFO":"Software Watchdog 2020-11-23 16:38:13"}
{"BOOTINFO":"Exception 2020-11-23 16:39:30"}
{"BOOTINFO":"Software Watchdog 2020-11-23 16:47:22"}
{"BOOTINFO":"Software/System restart 2020-11-23 16:47:47"}
{"BOOTINFO":"Software Watchdog 2020-11-23 16:52:42"}
{"BOOTINFO":"Software Watchdog 2020-11-23 16:55:20"}
{"BOOTINFO":"Software Watchdog 2020-11-23 17:02:20"}
{"BOOTINFO":"Software/System restart 2020-11-23 17:02:48"}
{"BOOTINFO":"Software Watchdog 2020-11-23 17:06:19"}
{"BOOTINFO":"Software Watchdog 2020-11-23 17:10:24"}
{"BOOTINFO":"Software Watchdog 2020-11-23 17:20:22"}
{"BOOTINFO":"Software Watchdog 2020-11-23 17:24:18"}
{"BOOTINFO":"Software Watchdog 2020-11-23 17:25:17"}
{"BOOTINFO":"Software Watchdog 2020-11-23 17:26:41"}
{"BOOTINFO":"Software Watchdog 2020-11-23 17:31:13"}
{"BOOTINFO":"Software Watchdog 2020-11-23 17:33:54"}
{"BOOTINFO":"Software Watchdog 2020-11-23 17:37:12"}
{"BOOTINFO":"Software Watchdog 2020-11-23 17:49:36"}
{"BOOTINFO":"Software Watchdog 2020-11-23 17:50:48"}
{"BOOTINFO":"Software Watchdog 2020-11-23 17:54:53"}
{"BOOTINFO":"Software Watchdog 2020-11-23 18:00:53"}
{"BOOTINFO":"Software/System restart 2020-11-23 18:03:00"}
{"BOOTINFO":"Software Watchdog 2020-11-23 18:24:42"}
{"BOOTINFO":"Software Watchdog 2020-11-23 18:39:15"}
{"BOOTINFO":"Software Watchdog 2020-11-23 19:39:58"}
{"BOOTINFO":"Software Watchdog 2020-11-23 19:49:24"}
{"BOOTINFO":"Software Watchdog 2020-11-23 19:53:40"}
{"BOOTINFO":"Software Watchdog 2020-11-23 20:16:14"}
{"BOOTINFO":"Software Watchdog 2020-11-23 20:26:53"}
{"BOOTINFO":"Software Watchdog 2020-11-23 20:37:57"}
{"BOOTINFO":"Software Watchdog 2020-11-23 20:51:53"}
{"BOOTINFO":"Software Watchdog 2020-11-23 21:13:59"}
{"BOOTINFO":"Software Watchdog 2020-11-23 22:12:14"}
{"BOOTINFO":"Software/System restart 2020-11-23 22:12:39"}
{"BOOTINFO":"Software Watchdog 2020-11-23 22:16:53"}
{"BOOTINFO":"Software Watchdog 2020-11-23 22:21:29"}`

And this is all the ESP, no intervention from me. I updated to the latest code at ~16:30 today, so these are all after that, but I have the same for previous days, at all times day + night.
I have a MagicMirror, with the Spa Control page as a widget. The mirror refreshes this every 2 mins. However, the mirror is on a PIR motion sensor so isn't on all the time. I thought it could be this, but doesn't explain the resets between midnight and ~7am when no one is awake activating the mirror :-/

I am running the 'Basic' 2.0.0 code.

Also, I've found the ESP or Spa Control become unusable/hang if it can't see the internet. Any way to fix this?

Thanks!

visualapproach, 2020-11-24:
@bertieuk Hi! The basic.ino was a little behind but I updated it a little now.
Next thing you should do is go to Arduino/libraries/arduinoWebSockets-master/src and edit "WebSockets.h". Around line 88 change the TIMEOUT value to 2000 (from 5000). (Note, this is not my library, so if you use it in other projects it will also be affected) 
Save and recompile basic.ino.

Regarding bad WiFi: In basic.ino find the function 
```
void startWiFi() { // Start a Wi-Fi access point, and try to connect to some given access points. Then wait for either an AP or STA connection
  WiFi.mode(WIFI_STA);
  WiFiManager wm;
  wm.autoConnect("AutoPortal");
  //WiFi.begin();
  Serial.println(F("Connecting"));
  while (WiFi.status() != WL_CONNECTED) {  // Wait for the Wi-Fi to connect
    delay(250);
    Serial.print('.');
    //checkAP_request();
  }
  Serial.println("\r\n");
  Serial.print(F("Connected to "));
  Serial.println(WiFi.SSID());             // Tell us what network we're connected to
  Serial.print(F("IP address:\t"));
  Serial.print(WiFi.localIP());            // Send the IP address of the ESP8266 to the computer
  Serial.println("\r\n");
}
```
and replace with
```
void startWiFi() { // Start a Wi-Fi access point, and try to connect to some given access points. Then wait for either an AP or STA connection
  WiFi.mode(WIFI_STA);
 // WiFiManager wm;
 // wm.autoConnect("AutoPortal");
  WiFi.begin("YOUR SSID", "YOUR PASSWORD");
  Serial.println(F("Connecting"));
  while (WiFi.status() != WL_CONNECTED) {  // Wait for the Wi-Fi to connect
    delay(250);
    Serial.print('.');
    //checkAP_request();
  }
  Serial.println("\r\n");
  Serial.print(F("Connected to "));
  Serial.println(WiFi.SSID());             // Tell us what network we're connected to
  Serial.print(F("IP address:\t"));
  Serial.print(WiFi.localIP());            // Send the IP address of the ESP8266 to the computer
  Serial.println("\r\n");
}
```

Doing so will (should) disable the autoportal which blocks the rest of the program from running.

bertieuk, 2020-11-24:
@visualapproach Thanks! Made those changes and uploaded ~17:18

Bootlog.txt since:
`{"BOOTINFO":"Software/System restart 2020-11-24 17:19:31"}
{"BOOTINFO":"Software/System restart 2020-11-24 17:24:34"}
{"BOOTINFO":"Software Watchdog 2020-11-24 17:28:10"}
{"BOOTINFO":"Software Watchdog 2020-11-24 17:31:15"}`

Still restarting :-( 

So, I disabled the widget on my MagicMirror and kept it awake. No resets for 15+ minutes!

I then left the Spa Control page open on my phone and realised it auto-refreshes itself - no need for the mirror widget to refresh it. So enabled the widget on the mirror with no auto-refresh, and no resets for ~30 mins!

visualapproach, 2020-11-24:
@bertieuk Okay, I hope it works Ok now. And that a poor WiFi isn't a showstopper anymore.
A magic mirror... wow that sounds like a fun gadget! You are very welcome to post a picture of it in the 'our cosy places' issue if you want.

closed 2020-12-12 by visualapproach

Jay-Ti, 2022-10-02:
> Next thing you should do is go to Arduino/libraries/arduinoWebSockets-master/src and edit "WebSockets.h". Around line 88 change the TIMEOUT value to 2000 (from 5000). (Note, this is not my library, so if you use it in other projects it will also be affected) Save and recompile basic.ino.

This seems like an oversight on the library side.
The Latest Version of the WebSockets Library allows for external definition of the TCP_Timeout (WebSockets.h Line 98).
For stability’s sake, there should be a mention in the documentation of this or the next Version could redefine the timeout.

Im new to github but will see to making a Pull Request.