visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #40: Solar panel example enables heater every reset

staylo, 2020-11-06:
The MQTT and All Features examples contain the code:

`const int solarpin = D0;    //no interrupt or PWM
..
pinMode(solarpin, INPUT_PULLUP);
..
  if (digitalRead(solarpin) == HIGH && runonce == true) {
    bwc.qCommand(SETHEATER, 1, 0, 0);       // cmd:set heater, to ON (1), immidiately, no repeat
..
  }`

D0 is not connected on the example schematic, so this code always sets the heater ON when booting.

visualapproach, 2020-11-06:
It's mostly there as an example or template to build upon. If you actually connect something to it, it should be pulled down or up from the external equipment so heater won't turn on at reset. But as I commented in code, it is untested by me. By the way- I think you should go with the mqtt with gpio version because the allfeats version saves eventlog often which may increase the chance for sw wdt reset.

staylo, 2020-11-06:
Ah, makes sense -  but I think it's an undocumented feature except for in the code comments at the moment :)

visualapproach, 2020-11-06:
Yeah you're right! Should probably put info in readme. I keep thinking that everyone is reading the code but that's ofc an inconvenient way to hunt for feats :)

closed 2020-12-12 by visualapproach