visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #41: v2.0.0 code

877dev, 2020-11-22:
Hey there, 

Finally got some time to check out this new version v2.0.0. Looks quite different, must've taken a lot of work! I've only had a quick scroll through but seems good.

Couple of thoughts:

1) Did not compile for me, I had to rename "BWC_globals.h" to "BWC_8266_globals.h" in the library. It could be changed elsewhere I'm sure, just a spelling error.

2) My pins are different to yours due to a different layout, I edited "BWC_8266.h", no biggie. But would be nice to be able to change them by just setting one flag somewhere.

3) MQTT credentials, can't see where these are inputted? There's no option on the web interface like before, and I don't see reference to "myMqttIP"or "myMqttPort" anywhere else. So of course I'm getting -2 connection error. Are you not using mqtt anymore?

visualapproach, 2020-11-22:
Hi! Thank you, yes it took some time :-)

1. Doh! Corrected now. I changed my local filenames while I was trying to port the project to ESP32*. I made some minor improvements that I wanted to upload and I forgot that file. 
2. Yes, the pinout is kind of hidden now. I may add parameters to the class.
3. It's in the example files "globals.h". Put BWC_v2.0.0 folder in "libraries" and open the examples from Arduino IDE menu. Then you just edit the globals.h tab.
4. You can easily add a repeating command to the queue from the config webpage (or via mqtt). For instance "set pump ON now and repeat every day"

I suggest you go with the example "basic with mqtt and gpio". If you are going to use the GPIOs you have to uncomment those parts in the code.

Cheers

877dev, 2020-11-22:
Ah yes I see the "globals.h" tab, somehow missed it! :)

I'll have a little play after this week when work is quieter.
Nice job! 👍🏻

visualapproach, 2020-11-22:
* I tried to see if I could port to ESP32 to learn something. It didn't go well ;-(
Cores panicked and it kept crashing until I gave up.

visualapproach, 2020-11-22:
2. Done

877dev, 2020-11-23:
Nice thank you. It looks like they can be defined in one line, which is excellent. Are they defined in the same order as per your comments above?

877dev, 2020-11-23:
Ok so now transferring over my node red flow to the new format, I have it working reading the values from mqtt.

But I cannot send mqtt commands currently, and I notice that even commands from the web interface do not work for me. So maybe that is work in progress?

I tried values of true and 1, and XTIME of 0 and current unix time...

e.g.
{"CMD":2,"VALUE":1,"XTIME":1606161550,"INTERVAL":0}
or
{"CMD":2,"VALUE":true,"XTIME":1606161550,"INTERVAL":0}
or
{"CMD":2,"VALUE":1,"XTIME":0,"INTERVAL":0}

visualapproach, 2020-11-23:
> Ok so now transferring over my node red flow to the new format, I have it working reading the values from mqtt.
>
> But I cannot send mqtt commands currently, and I notice that even commands from the [...]

Hi, It should be working as is. Do you send to default topic "**BW_2.0.0/command**"? With payload as above, which is correctly formatted. (I might have changed the topic base name.)

> I've been looking at your code, specifically "BWC_8266.cpp". Am I correct that if no parameters are passed to bwc.begin, via 
> the main sketch, it is classed as (void) and the default values are used? And if parameters ARE passed, then it skips the void 
> section and goes to the second function?
> 
> If so, I've never really understood that void could be used that way..

You are right about the parameters - The "void" is actually not the thing here. It just tells the compiler that no parameters are expected. It can actually be omitted in many cases.
`void myfunction(void){} `
is the same as 
`void myfunction(){} `
When writing a class you can make several functions with the same name, but with different parameter types or count, and the compiler will choose the one version that matches the parameters given. It's called overloading a function.
So in the function 
'void bwc.begin(void)' I set the pins to default value. But in 
'void bwc.begin(int, int, int, int, int, int, int)' I set pins according to the parameters.

877dev, 2020-11-23:
> Hi, It should be working as is. Do you send to default topic "**BW_2.0.0/command**"? With payload as above, which is correctly formatted. (I might have changed the topic base name.)

Turns out I had a spelling mistake :-)  I deleted my comment but you replied too soon lol. It's working now.

Overloading a function, that is interesting thanks for the info. And thanks for adding that in, makes my life easier with my dodgy pins.

I'll work on the node red flows next, for hooking my Blynk app to control the pump. Before it was toggled by sending i.e. 0x5 for power, now it needs a different payload per state. Pretty easy to hook up with nodes, but I will try and write a function to make it more eloquent. 

There's no up and down control anymore, but I can replace that with a slider. And the unlock button is no longer needed it seems. Finally timer button, but who uses the timer button anyway! :)

visualapproach, 2020-11-23:
Haha, I'm glad it worked out. But I wonder why the web interface didn't. Did you upload the new data folder to LittleFS?
And yes, I thought the UP/DOWN, ON/OFF and LOCK was obsolete now. And the factory timer was too confusing to use anyway.

I set up my node red such that when the set temperature is reached I get an email :-). Or when temp is below 10C. I must be old, thinking such things are cool...

877dev, 2020-11-24:
Yes I did upload LittleFS and have the new web interface. Not a big deal really. 

Any reason why power button is missing from the commands? For example if someone turns it off outside, does the code handle that and power it up?

visualapproach, 2020-11-24:
Yes I removed it because you don't really care if it's on or not. The off state just seems to dim the display and annoys people when they wanna do something like lower the temp or so. Same with the lock. So if you send a command, let's say "set temp to 40", the esp first turns on the power and then unlocks the device (if necessary) and then simulates UP until display reads 40.

visualapproach, 2020-11-24:
Btw, the web interface queue on the 'confit' page refreshes every 4 seconds. Maybe you was a little impatient?
Or is it the main page that doesn't work for you?

877dev, 2020-11-25:
No it definately does not update, this is the page I am referring to:

[image]

I click "Add command to queue" and it never appears..

visualapproach, 2020-11-28:
Okay, a couple of thoughts;
Is that a normal electric price in your area?
It may be clear to you already, but just in case - Adding a command with xtime in the past, or "now" will execute immidiately and will be deleted from the queue before it pops up on the page. If you add an interval more than zero, it will first execute, then add the interval time to the command. Meaning it should show up in the queue. Likewise for single events in the future.
You could check the browser console (Chrome: shift+ctrl+I) what is sent/received.

877dev, 2020-12-01:
> Is that a normal electric price in your area?
I assume so, it's £0.1295 per kwh, the interface only seems to allow two decimal places, so 0.13. Also there is a daily charge of £0.195 on top of that. 
Why is that good or bad? :-)

> Adding a command with xtime in the past, or "now" will execute immidiately
That wasn't obvious at first, but I figured it out. Didn't know that "now" works also, even better.

> If you add an interval more than zero, it will first execute, then add the interval time to the command. Meaning it should show up in the queue. Likewise for single events in the future.

Does not work for me, I'm using the web interface and selecting a few minutes in the future, also sending interval of 8600. This is console output, never looked in there before..

[image]

visualapproach, 2020-12-01:
Price- I was just curious. I checked my bill and my rate is 0.58 SEK (around £0.05) but there is a fixed price on top of that too. Difficult to compare.

"Now" meant current timestamp at the time you send the command. Not the literal string "Now".

The errors indicate that you don't have the correct config.HTML file. The line numbers don't match mine.
Download latest files and via Ip/upload.html choose the newest html files.

Wish you luck

877dev, 2020-12-03:
> The errors indicate that you don't have the correct config.HTML file. The line numbers don't match mine.
> Download latest files and via Ip/upload.html choose the newest html files.

Strange, I used the latest files from github. I just manually uploaded each file and have the same result.

Could it be something I have inadvertently changed in the sketch?

visualapproach, 2020-12-04:
Errors would occur if you changed the sketch to send data not matching the expected format. However those errors would occur at specific lines in the web page's source. Now, if you, instead of looking at the console tab in your browser,  go to the SOURCE tab. Select config.html. Here you can see what is ACTUALLY on your board. Compare with the config.html in latest example/data folder.

This is from the config.html file on this github. You see the "{LEN:0}" string is printed at line 112, as opposed to line 105 in your picture.
[image]

By clicking source, here we see that the console.log is at the line 112.

[image]

877dev, 2020-12-04:
Hey, just looking at this now coincidentally. Give me 10 mins to check.. :)

visualapproach, 2020-12-04:
Hmm, it expects a message like:  {"LEN":1,"CMD":[2],"VALUE":[0],"XTIME":[1607685540],"INTERVAL":[0]}
but only gets {"LEN":1}. Hence trying to parse CMD etc will throw a UNDEFINED error.
So check your code if it's sending all info. Original code responsible for this:

```
//this is the sketch file
//response to /getcommands/
void handleGetCommandQueue() { // reply with json document
  if (server.method() != HTTP_POST) {
    server.send(405, "text/plain", "Method Not Allowed");
  } else {
    String jsonmsg = bwc.getJSONCommandQueue();
    server.send(200, "text/plain", jsonmsg);
  }
}
```
 
```
//this is the .cpp file
String BWC::getJSONCommandQueue(){
  // Allocate a temporary JsonDocument
  // Don't forget to change the capacity to match your requirements.
  // Use arduinojson.org/assistant to compute the capacity.
  //feed the dog
  ESP.wdtFeed();
  DynamicJsonDocument doc(1024);
  // Set the values in the document
  doc["LEN"] = _qCommandLen;
  for(int i = 0; i < _qCommandLen; i++){
	  doc["CMD"][i] = _commandQ[i][0];
	  doc["VALUE"][i] = _commandQ[i][1];
	  doc["XTIME"][i] = _commandQ[i][2];
	  doc["INTERVAL"][i] = _commandQ[i][3];
  }

  // Serialize JSON to file
  String jsonmsg;
  if (serializeJson(doc, jsonmsg) == 0) {
    jsonmsg = "{\"error\": \"Failed to serialize message\"}";
  }
  return jsonmsg;
}
```

877dev, 2020-12-04:
Very strange, my sketch and .cpp are exactly the same...

visualapproach, 2020-12-04:
Yes, strange it is. In the console window, enter
{"CMD":2,"VALUE":0,"XTIME":0,"INTERVAL":0}
and press enter. See if that works.

regarding eventlog.txt - No it is not a problem. That file only exists if you create it by using saveEventlog() but that is not recommended (by me) because it was a debug function which will eventually fill up your flash.

visualapproach, 2020-12-04:
@877dev you should look at IP/cmdq.txt
Maybe something weird is saved there.

877dev, 2020-12-05:
> Yes, strange it is. In the console window, enter
> edited:
> {"CMD":2,"VALUE":1,"XTIME":0,"INTERVAL":0}
> and press enter. See if that works.

Seems to be accepted ok, but no bubble turn on, or show up in the command queue.
[image]

cmdq.txt just show `{"LEN":0}`

visualapproach, 2020-12-05:
OK, do you use my code or custom? I know you changed the pinouts, but anything else?

877dev, 2020-12-05:
I use the gpio code, changed pinouts and mqtt reporting timer to 5 seconds. But I tried with default 600s too. Not anything else I can think of. Also re-uploaded LittleFS just in case.

visualapproach, 2020-12-05:
Same code and same Files. Shouldn't be it.
What differs?
- Other hardware
- Dependancy libraries. Try updating ArduinoJSON, and other libraries.
- Time and timezone
- Browser? I use chrome/safari both works
- Board settings in arduino IDE?

visualapproach, 2020-12-05:
Does it work when pressing the pump display buttons? Or via MQTT?

877dev, 2020-12-07:
Sorry for delay, I've worked all weekend :(

OK so I checked all of this, same error I'm afraid.

Other hardware **NONE**
Dependancy libraries. Try updating ArduinoJSON, and other libraries. **UPDATED ALL**
Time and timezone **TRIED 0 AND 1**
Browser? I use chrome/safari both works **TRIED CHROME AND IE**
Board settings in arduino IDE? **SEEM CORRECT**
Does it work when pressing the pump display buttons? **YES ALL BUTTON WORK**
 Or via MQTT? **"NOW" COMMANDS WORK, BUT FUTURE COMMANDS DO NOT**

Also tried basic sketch, only thing I changed is the pins, same error. 

It's a shame no-one else can weigh in, as it might be just something on my end..
But not to worry, I can control the pump via mqtt and use timer nodes etc..

877dev, 2020-12-07:
One thing I noticed is the time is "American" format, i.e. MM/DD/YYYY

[image]

Whereas in the config it is EU format DD/MM/YYYY
[image]

But I think the main page sometimes changes to EU format after a while, or am I imagining that?

visualapproach, 2020-12-07:
No worries, I've been busy trying to repair my 2:nd broken pump :(. Stupid impeller needs to be exactly at the right position, not even half a mm off axis...
Afaik (not home now) the esp is sending timestamps and the browser is converting to where it thinks you are.
We will check another thing. Call you back.

877dev, 2020-12-07:
If it's too much trouble don't worry, it's more an annoyance it's not working than anything:)

Have you got / did you have Grafana working? I notice that when set to target of 40, the heater does not come back on until the actual temp drops down to 38. I wonder if that is by design or in your code?

I might setup a node red flow to kick the heater back in when it drops to 39, only during certain hours. Getting a bit chilly over here 😁￼

visualapproach, 2020-12-07:
Check in your library folder. Remove any old folder containing this project. Any #include statement may use the first matching file it finds, wich may be in another folder.
Yes my Grafana is working. 
I didn't code it to switch at 38/40 so it must be by design. I think I read that the material in the kit was made for 38 C, so I think they made it so that it will not exceed 40 degrees.

visualapproach, 2020-12-07:
Also, have you tried it on another 8266? On the bench, not connected to something other than USB.

877dev, 2020-12-08:
I only have one library folder. I will try it on another esp8266, I have a spare pump head unit I can plug it into.
Little things like this are quite annoying for some reason, messes with my ocd  ;-)

visualapproach, 2020-12-08:
I know the feeling

877dev, 2020-12-09:
Spare esp8266 not connected to anything, flashed fresh from basic sketch. Same issue, also the on board LED is flashing very fast (Wemos D1 mini), I'll check what's causing that now..

[image]

877dev, 2020-12-09:
Serial monitor output while error occurs, is the "Disconnected" anything?

[image]

visualapproach, 2020-12-09:
I'll have a look after work. LED is probably on One of the data pins.

closed 2020-12-09 by visualapproach