visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #47: 4 pin models

TimE90, 2020-12-20:
Hey @visualapproach!

Thanks for this great project. My family and friends having several Lay Z Spa models. Some having the 6 pin connector and others are the 4 pin models.

Do you think it‘s possible to extend functionality for the 4 pin models?

visualapproach, 2020-12-21:
Hi! Short answer: yes. Someone just need to have the time and motivation to do it. I guess the bulk of the project is reusable but I wouldn't say it's a quick fix. Not ruled out but don't expect anything this year.
Cheers

visualapproach, 2021-01-07:
I had a first look at a #54138 model with 4 wires. I have decoded some of the communication. Unfortunately it's not only the comms that differ. The display is actually controlling the CIO. But I made some progress. Reporting back another day, because I just got a shitload of Shellys I want to install first 🥰

visualapproach, 2021-01-09:
Some words about this:
They seem to make many different 4-pin models, each with a different protocol. I was looking at [https://github.com/mrQ000/layz-rc](url) but it wasn't the same as my test object. That is disappointing because that means there is no one "4-wire-fix" that suites everyone. So before you can use a 4 wire version, you have to be absolutely certain that you have compatible hardware and software.
I have borrowed a friend's hot tub, and by hijacking the serial wires between the display and pump, I can send commands to it from an arduino and also get the present temperature.
However, the display is also the brains. "Target temperature" is only known to the display internals, and never shown on the serial bus. We can ofc let our own device take over the command, but the display would not be in sync. The user has to decide whether the ESP or the display should regulate the temperature. 
This particular device has these wires; Black, orange/red, Yellow and green.
So, black is ground and red is 5 Volts you ask? No sir! After burning my finger on the arduino I realized something was wrong, and I measured the wires with the DMM again. And sure enough - black is 5V and red is GND. But of course! Why not?! 😩 
Yellow is DSP TX and green CIO TX.
Next, I'll probably hook up a 8266 and test if I can make something useful out of it.

visualapproach, 2021-01-17:
Progress report: I have a working prototype now. I can take over command from web interface (and mqtt, though not tested) and hardware is pretty much the same as before - 8266 and a levelshifter, and some wires. There is still some points on my todo-list before I can release the code if you or anyone are still interested.

visualapproach, 2021-01-22:
READ EVERYTHING BEFORE BUILDING! ONLY FOR #54138 (maybe #54123 with modification to "model.h" but that is not tested and it has different pinout.)

Hardware: NodeMCU ESP12E and a level converter. Prototype board/PCB and cables.
Pinout:
Black(+5 V)
Red (GND)
Yellow (DSP TX / CIO RX)
Green (DSP RX / CIO TX)
You MUST double check your hardware before connecting anything!

From CIO:
5 V to ESP Vin, Level converter HV, and DSP 5V pin.
GND to ESP GND, Level converter GND, and DSP GND.
CIO RX (yellow) to D3
CIO TX (green) to D2

From DSP:
5V see above
GND see above
DSP RX (green) to D7
DSP TX (yellow) to D6

Other:
ESP 3V3 to LLC LV

This SPA model is having a more "intelligent" display that sends commands to the pump unit. Therefore we cannot simulate button presses. Instead we take over control of the pump which can potentially be risky. It appears the pump has some safety features, but I cannot guarantee that the program covers all such features that the display might have. I have tried to incorporate safety, but I am not responsible for what happens if you use this. If the CIO send error messages the ESP stops acting like a controller and passes control to the display. No guarantee though since it can hang or break otherwise. Don't leave the device unsupervised until you are confident that everything works. And even then, DO NOT BLAME ME!

Arduino users: Sorry, but I wrote this in platformIO on Visual Studio Code. If you desperately needs an arduino IDE version it should be relatively easy to convert. MQTT is commented out in the code. To enable you need to uncomment and set your credentials in 4_wire.h

visualapproach, 2021-01-22:
Also, it is an early version. Only tested on the bench.

TomK, 2021-01-24:
Apparently I have a 4 pin version (Palm Springs).  Already bought all the bits to build the 6pin, but thankfully haven't assembled yet. I'm using perfboard for now.
Reading above it looks like I should be using a 4 channel LLC now, is that correct?

visualapproach, 2021-01-24:
> Apparently I have a 4 pin version (Palm Springs).  Already bought all the bits to build the 6pin, but thankfully haven't assembled yet. I'm using perfboard for now.
> 
> Reading above it looks like I should be using a 4 channel LLC now, is that correct?

You can just leave the spare channels unconnected if it has more than 4. And check vcc and gnd with a DMM, do not trust the pin order or color.

TomK, 2021-02-01:
Out of interest, which version of the control panel did you write this against?  I have two different 4pin control panels which i've tried on my spa (without this wifi module, yet).  One control has a "test" button, my original does not.

There's clearly some disparity between the two because pressing the "bubbles" button on the new panel turns off the heater and pump, where my original control panel leaves them on (as expected).

Once i've put this together, my plan is to run it with my original (no test button) panel.  If this works correctly then i could make a branch in this codebase to interface the "new" panel with the "old" pump (assuming it doesn't just work, somehow).

visualapproach, 2021-02-01:
54138. No test button iirc, and bubbles turns heater off. Which I tried to mimic in software. I guess it might be a power restriction to stay in spec. Better not change it in that case

TomK, 2021-02-01:
Ah, I also just realised the # numbers you mentioned above are bestway model numbers:

Lay-Z Spa Hawaii Hydrojet Pro – BW54138
Lay-Z-Spa Miami AirJet – BW54123

So, i assume you wrote it against the hydrojet.  It's unlikely to work for my Palm Springs as-is, so i'll expect to do some modifications.  How would you like me to structure contributions?  In a new subfolder of Code?

TomK, 2021-02-01:
For posterity, I have `Lay-Z Spa Palm Springs – BW54129`

visualapproach, 2021-02-01:
It would be nice to have one code for all 4-pin models. At least if only the bit/byte pattern is differing. I prepared a header file for that, to be extended with more models. If you can comply with that I suggest you fork the repo and edit, then make a pull request to me.
If you have to do breaking updates (no longer compatible with earlier models) I think it's better you make a separate version, zip it and attach to a issue so I can put it in a new folder.
(Managing GitHub isn't my strongest skill so I'm open for suggestions)

jenswalit, 2021-04-18:
Hello i have the Vegas pool with 4Pin 54112. Can I use your Project? Or What must be edit. Thanks for you work.

visualapproach, 2021-04-20:
> Hello i have the Vegas pool with 4Pin 54112. Can I use your Project? Or What must be edit. Thanks for you work.

Hey @jenswalit  ! At present, there is a 4 wire version that works with 54138 (tested) and possibly 54123 (not tested). Anything else may or may not work. To find out, you need to determine which one of the four wires is +5V, and which is GND. This is doable with a DMM (digital multimeter). 
Then you have to determine which of the two remaining wires is CIO TX. That might be doable with a DMM by disconnecting the display and see which wire will show a small voltage when powering up the pump. Now you should know the pinout on your device and compare it to the pinout in the description (see BW4W code folder) then at this point you may take a chance and just see if it works.
A little involved so if you are not confident in electronics maybe it's good to let someone else try it out first. On the same model I mean. Hopefully they will report here if that happens.
And even if the pinout is known, I can't tell if the serial protocol is the same between different models.

Inazone, 2021-04-21:
Hi there, if we wanted to investigate this further, how would we go about doing this? I'm no electronics expert but have been using multi meters for years but appreciate that you can only do so much with a MM. What other bit of kit can we use? I think this is a great project and if I can help by probing my Maldives pump then I would be happy to help.

jenswalit, 2021-04-21:
many thanks. I've been involved with electronics since I was a child. I will try to use the project for my pump. I saw in another project that I can also read serially with an 8266. If that doesn't work, I'll buy a logic analyzer and try to get the information. Maybe we are lucky and get on and can move the project further in the direction of 4Pin.

visualapproach, 2021-04-21:
A logic analyzer is really useful because you don't need to know baudrate or protocol beforehand. But in this case we can guess that your models use the same serial mode: 9600 baud, 8N1. So a 5V arduino UNO or similar can be used (or an 8266, but you need a level converter). Both the CIO and the DSP sends packets of 7 bytes every half second. Capture those bytes and post them here and we can compare. (You need to write a simple program to get the byte values, rather than ascii code, to show in the serial monitor. Ask if you need help with this)

jenswalit, 2021-04-26:
Hello visualapproach, 
my 4Pin Pump works perfect. 
But now i see thas your MQTT Code ist diffrent to the 4Pin Code. Do you upgrade this code in future?
I use MQTT to see some Data about Temp and Status in my Home Automation System.

visualapproach, 2021-04-26:
> Hello visual approach,
> My 4Pin pump works perfectly.
> But now I see that your MQTT code is different from the 4Pin code. Will you update this code in the future?
> I am using MQTT to show some temperature and status data in my home automation system.
> For your information I use the code from the NO54123 but I have not commented out the code in the model.h because it will not work. In the future I will build a definon for my 54112 so that everything is clean. I also have problems using the resistors and have not used them now. I also had your plate made and soldered everything nicely. I still have some left if someone wants some, I can give them some.
> 
> Now the weather just has to be a little better. Dan off to the pool.
> 
> Thanks for your work.

Hi @jenswalit , Glad to hear from you! What MQTT differences do you mean? The base topic is different, but is that a problem? Also, since the pumps have physical differences (Jets for instance) the MQTT messages will contain a few different states. I think it's mostly compatible, but not identical. Can you elaborate more?

I'd be happy to know more about your pump model to add to the table in the docs. I reckon it's a 54112 with 4 wires. Is it a "Vegas"? Hydrojets? Square-ish pump?
The file "model.h" should look like one of these:

```
#define NO54138 //this is the only option for now. Maybe there will be other pumps supported in the future.

//WARNING: THIS DEVICE HAS DIFFERENT PINOUTS!!! CHECK BEFORE USING
//NOT TESTED BY ME!!
//#define NO54123 //not tested. ref https://github.com/mrQ000/layz-rc link to presentation slides
```

or

```
//#define NO54138 //this is the only option for now. Maybe there will be other pumps supported in the future.

//WARNING: THIS DEVICE HAS DIFFERENT PINOUTS!!! CHECK BEFORE USING
//NOT TESTED BY ME!!
#define NO54123 //not tested. ref https://github.com/mrQ000/layz-rc link to presentation slides
```
Which one works for you?

Cheers

Inazone, 2021-04-26:
> > Hello visual approach,
> > My 4Pin pump works perfectly.
> > But now I see that your MQTT code is different from the 4Pin code. Will you update this code in the future?
> > I am using MQTT to show some temperature and status data in my home automation system.
> > For your information I use the code from the NO54123 but I have not commented out the code in the model.h because it will not work. In the future I will build a definon for my 54112 so that everything is clean. I also have problems using the resistors and have not used them now. I also had your plate made and soldered everything nicely. I still have some left if someone wants some, I can give them some.
> > Now the weather just has to be a little better. Dan off to the pool.
> > Thanks for your work.
> 
> Hi @jenswalit , Glad to hear from you! What MQTT differences do you mean? The base topic is different, but is that a problem? Also, since the pumps have physical differences (Jets for instance) the MQTT messages will contain a few different states. I think it's mostly compatible, but not identical. Can you elaborate more?
> 
> I'd be happy to know more about your pump model to add to the table in the docs. I reckon it's a 54112 with 4 wires. Is it a "Vegas"? Hydrojets? Square-ish pump?
> The file "model.h" should look like one of these:
> 
> ```
> #define NO54138 //this is the only option for now. Maybe there will be other pumps supported in the future.
> 
> 
> //WARNING: THIS DEVICE HAS DIFFERENT PINOUTS!!! CHECK BEFORE USING
> //NOT TESTED BY ME!!
> //#define NO54123 //not tested. ref https://github.com/mrQ000/layz-rc link to presentation slides
> ```
> 
> or
> 
> ```
> //#define NO54138 //this is the only option for now. Maybe there will be other pumps supported in the future.
> 
> 
> //WARNING: THIS DEVICE HAS DIFFERENT PINOUTS!!! CHECK BEFORE USING
> //NOT TESTED BY ME!!
> #define NO54123 //not tested. ref https://github.com/mrQ000/layz-rc link to presentation slides
> ```
> 
> Which one works for you?
> 
> Cheers
> 
> PS Näsström should have creds for that PCB. I have another variant of that coming after testing is done. It's basically the same but slots for 4-wire cables as well, and with the cheaper LLC, and no resistors. Well actually one resistor. DS

I'd be happy to test the pcb for the 4 wire setup. Happy to cover costs for sending the board etc...

jenswalit, 2021-04-27:
On my Website the MQTT is empty.

Can i debug tis MQTT messages?

And my Pool is a VEGAS i have add some definitions in the model.h an globals.

jenswalit, 2021-04-29:
Hallo nochmal.

Wie lange läuft die Pumpe nach?
in dem Code sehen ich Zeitstempel + 10.  Sind das Secunden?

`//let pump run a bit to cool element
          if(_cio.states[HEATSTATE] && !_commandQ[0][1]) {
            qCommand(SETHEATER, 0, 0, 0);
            qCommand(SETPUMP, 0, _timestamp + 10, 0);
          } else {
            _cio.states[PUMPSTATE] = _commandQ[0][1];
          }
					break;`

visualapproach, 2021-04-29:
Hallo
Das ist sekunden ja. So die pumpe läuft 10 s after 'jetzt/now' but I'm better in C++ than german so I'll stick to english :)
I'm still not sure of what you mean with your mqtt problem.

jenswalit, 2021-04-29:
In MQTT I mean that I only get the status, the Mac address and the connect status. I would like to get more to the broker. On the website I only see []. when restarting I see [DISPLAY] or in the event of an error [E00] but nothing else. Is it correct that way?

visualapproach, 2021-04-29:
> In MQTT I mean that I only get the status, the Mac address and the connect status. I would like to get more to the broker. 

You should get states in topic BW4W/message (iirc). Never really checked this version so I might have a look.

> On the website I only see []. when restarting I see [DISPLAY] or in the event of an error [E00] but nothing else. Is it correct that way?

That's a legacy from the 6 wire interface. I kept it as a simple way of showing errors. There is no way of knowing what's on the actual display. So it correct in that aspect.

> And can I simply use the MQTT website from the DATA of the 6Pin project for 4Pin too? or is there too much different.

Nope, the code has to be updated also... but I hear you

visualapproach, 2021-04-30:
@jenswalit I checked now, and MQTT is off by default in this version. Have you uncommented the MQTT code? I've marked the lines below with a * where you have to remove the slashes

```
void setup() {

  // put your setup code here, to run once:
//  pinMode(solarpin, INPUT_PULLUP);
//  pinMode(myoutputpin, OUTPUT);
//  digitalWrite(myoutputpin, LOW);
  Serial.begin(115200);
  bwc.begin(); //no params = default pins
  startWiFi();
  startOTA();
  startServer();
  startWebSocket();

*  // startMQTT();
*  // updateMqttTimer.attach(600, sendMQTTsetFlag); //update mqtt every 10 minutes. Mqtt will also be updated on every state change
  updateWSTimer.attach(2.0, sendWSsetFlag);     //update webpage every 2 secs plus state changes

  pinMode(D4, OUTPUT);  //built in LED for some feedback
  digitalWrite(D4, LOW);

}

void loop() {

  webSocket.loop();             // constantly check for websocket events
  server.handleClient();        // run the server
  ArduinoOTA.handle();          // listen for OTA events
*  //if (!MQTTclient.loop()) MQTT_Connect();           // Do MQTT magic
  bwc.loop();                   // Fiddle with the pump computer
  if (bwc.newData()) {
    sendMessage(1);//ws
*    //sendMessage(0);//mqtt
    //bwc.saveEventlog();       //will only fill up and wear flash memory eventually
  }
  if (sendWSFlag) {
    sendWSFlag = false;
    sendMessage(1);//ws
  }
*  // if (sendMQTTFlag) {
*  //   sendMQTTFlag = false;
*  //   sendMessage(0);//MQTT
*  // }
```

4ae16e, 2021-05-21:
Hi, I have 54123 version. I can test it and say my progress here.. but.. I have problem convert code from PlatformIO to Adruino IDE. I can see code, compile it but without any result.. serial debug wrote only one empty line..
thanks for help with this

visualapproach, 2021-05-21:
> Hi, I have 54123 version. I can test it and say my progress here.. but.. I have problem convert code from PlatformIO to Adruino IDE. I can see code, compile it but without any result.. serial debug wrote only one empty line..
> thanks for help with this

Hi,
Just rename "main.cpp" to "src.ino" and install the Arduino libraries listed in "platformio.ini". Then doubleclick on src.ini
Please download the version from the development branch. And while you are there, you can checkout the readme.md in the code folder :-)

4ae16e, 2021-05-21:
> > Hi, I have 54123 version. I can test it and say my progress here.. but.. I have problem convert code from PlatformIO to Adruino IDE. I can see code, compile it but without any result.. serial debug wrote only one empty line..
> > thanks for help with this
> 
> Hi,
> Just rename "main.cpp" to "src.ino" and install the Arduino libraries listed in "platformio.ini". Then doubleclick on src.ini
> Please download the version from the development branch. And while you are there, you can checkout the readme.md in the code folder :-)
> Also, move the data folder to the src folder.

I had problem only that data folder :-)  I forgot move it in to src folder.. I will try it again.
I have necessary  libs.. maybe problem will be somewhere in versions.. thanks for now

4ae16e, 2021-05-21:
OK, flashed from developent branch looks good. After flash I must wait few minutes to create AP, I don't know why.. but after connection it seems work. I will try connect it to unit. I will report progress..

4ae16e, 2021-05-22:
So, i checked wiring three times. I have 8ch lvl converter.. On low side 680ohm resistors on signal wires. And nothing. Esp not boot, it looks like problem with supply.. Maybe in 54123 version is another order of cables..

visualapproach, 2021-05-22:
> So, i checked wiring three times. I have 8ch lvl converter.. On low side 680ohm resistors on signal wires. And nothing. Esp not boot, it looks like problem with supply.. Maybe in 54123 version is another order of cables.. 

You should check +5V and GND before connecting or things may break. According to [](https://github.com/mrQ000/layz-rc) it's not the same wire layout. Check the link at the bottom of that page

TheCurly, 2021-06-16:
No, I'll give it a try!

TheCurly, 2021-06-16:
Just tried the development branch, I now get an E13 error shortly after powering on.

visualapproach, 2021-06-16:
> Just tried the development branch, I now get an E13 error shortly after powering on.

Okay, that's com error. Guess you have to go back to previous code (master) and I'll see if there is something I can do.

TheCurly, 2021-06-16:
> > Just tried the development branch, I now get an E13 error shortly after powering on.
> 
> Okay, that's com error. Guess you have to go back to previous code (master) and I'll see if there is something I can do.

Thanks! I'll leave the remote out for now and wait for your reply. Let me know if I can help to test anything.

visualapproach, 2021-06-16:
> Thanks! I'll leave the remote out for now and wait for your reply. Let me know if I can help to test anything.

Hi, I found and squashed a bug. Please try again now.

TheCurly, 2021-06-16:
> Hi, I found and squashed a bug. Please try again now.

I feel silly, forgot to change the code to D2 and D3 to match the pinout I soldered! But after changing that, both the remote and the display could not control the pump, heater, etc.  After you squashed the bug it is working again, I'll let it run and see if the E01 pops up again.

TheCurly, 2021-06-18:
Works quite a bit better, although I am getting E13 errors every once and a while. Looks like this happens mostly when using the web interface. Can it be the Wemos D1 Mini compared to the NodeMCU? I would think it doesn't make a difference since they both use the same ESP8266 chip. Could it be timing related, maybe the response to the DSP is not exactly right when the controller is serving the web page?

visualapproach, 2021-06-18:
Do you get E13 when 'take control' is disabled also? If not there might be something wrong in the code.
If it is caused by cpu too busy you can try over clocking it to 160MHz. There is a flag you can add in platformio.ini

TheCurly, 2021-06-18:
> Do you get E13 when 'take control' is disabled also? If not there might be something wrong in the code.
> If it is caused by cpu too busy you can try over clocking it to 160MHz. There is a flag you can add in platformio.ini

It has also happened when 'take control' is disabled and I just refresh the web page. I am using Arduino IDE, but the CPU was set to 80MHz, I'll try 160 instead to see if that helps.

TheCurly, 2021-06-21:
Not really any difference with 160MHz over the weekend, I got a few E13 errors, even without taking control and just powering it up and turning on the pump from the display.

visualapproach, 2021-06-21:
Is it possible that the WiFi is poor when the errors appear?

TheCurly, 2021-06-21:
> Is it possible that the WiFi is poor when the errors appear?

Not sure how good the reception is on the Wemos, but it is only about 4-5m from the wifi router with just a glass door in between (double glass but no reflective coating or foil or anything). 
Can a WiFi disconnect cause E13 errors? Could you make the controller continue to work without or with an intermittent or bad WiFi connection?

visualapproach, 2021-06-21:
> Not sure how good the reception is on the Wemos, but it is only about 4-5m from the wifi router with just a glass door in between (double glass but no reflective coating or foil or anything).
>
> Can [...]

It will enter AP mode which halts the ordinary program and that will definitely cause E13. That's the reason I tried to use the alpha version of WiFi manager. With non blocking AP. But it didn't work. The portal didn't show up. 
Now, we don't know if this is the real problem but I suggest you try without the auto portal. The WiFi credentials should still be remembered. I'll post what you need to do shortly.

visualapproach, 2021-06-21:
> It will enter AP mode which halts the ordinary program and that will definitely cause E13. That's the reason I tried to use the alpha version of WiFi manager. With non blocking AP. But it didn't [...]

Okay, edit main.cpp line 317 like this:
```
  //wm.autoConnect("AutoPortal");
  WiFi.begin();
```
Upload and give it a try. Don't upload the filesystem again. If it works without E13 it is awesome, otherwise we know it's something else 😄

TheCurly, 2021-06-21:
I think I can confirm it is WiFi related, the display goes into E13 error a few seconds after I switch off the WiFi.
I uploaded the edited code as a bin file through the firmware update page, but it does not seem to make any difference. At the moment it is wet outside so I can't (safely) open up the pump to program the controller directly and verify that the programming succeeded.

visualapproach, 2021-06-21:
No need to go out! To upload new firmware over the air you edit platformio.ini and remove the semicolons at the bottom rows. Change the IP to whatever your tub has. Then click the upload (right arrow) icon.

TheCurly, 2021-06-21:
So far I have been using Arduino IDE, but I'll check out PlatformIO. I exported the compiled binary from Arduino IDE and uploaded that via the firmware updater page, not sure if that works.

visualapproach, 2021-06-21:
There is no firmware update page. Maybe you think of upload.html? It's for uploading regular files like index.html and so on. But you can use arduino ide also, just select the IP of your device instead of COMx

TheCurly, 2021-06-21:
> There is no firmware update page. Maybe you think of upload.html? It's for uploading regular files like index.html and so on. But you can use arduino ide also, just select the IP of your device instead of COMx

There is an update.html page that says "Firmware updater", but I'll try to update through Arduino over the network. Did not know that was possible!

visualapproach, 2021-06-21:
> There is an update.html page that says "Firmware updater", but I'll try to update through Arduino over the network. Did not know that was possible!

Haha, I forgot. It was added by someone else and I haven't tested it. Not sure what kind of .bin-file it's supposed to take.

TheCurly, 2021-06-21:
After about an hour, the display is now showing En8 and beeping again. Never seen that before and I can't immediately find what it means.

TheCurly, 2021-06-22:
Made a quick sketch to check the WiFi signal and the Wemos reports RSSI values ranging from -49 to -52dB, that should be good enough for a stable connection.

visualapproach, 2021-06-23:
> Made a quick sketch to check the WiFi signal and the Wemos reports RSSI values ranging from -49 to -52dB, that should be good enough for a stable connection.

Okay. Unless it's code related I suspect it might be impedance mismatch or something. I'm NOT an electronic engineer, but I think it may be useful to put resistors from LLC to RX and TX. That would be four in total. Like:
ESP-----LLC----500Ohms---DSP RX
ESP-----LLC----500Ohms---DSP TX
ESP-----LLC----500Ohms---CIO RX
ESP-----LLC----500Ohms---CIO TX
Or it might be useless. But it could be worth a try thou. The value 500 Ohms is just a guess. But say 100 to 800 should be a good start.

TheCurly, 2021-07-04:
> Okay. Unless it's code related I suspect it might be impedance mismatch or something. I'm NOT an electronic engineer, but I think it may be useful to put resistors from LLC to RX and TX. That would [...]

I added the 4 resistors (560 ohm) a few days ago, but I am still getting intermittent E13 errors. Looks like the frequency of the errors goes up when I control the spa through the web page. If I just leave it alone and turn it on through the display it runs for half a day or so. Could it be something in the code?

visualapproach, 2021-07-04:
Yeah, it could be anything I guess. Did you try with no autoportal?
One could see what happens if just bypassing the traffic. I may make code for that, but it will take a while. I have a lot of other things needing my attention now.

TheCurly, 2021-07-04:
Yes, I tried without autoportal but it did the same thing. My wife is getting a little annoyed by the beeping but take your time :)

TheCurly, 2021-08-03:
I have had the spa running for more than a week with the bypass program, and I only got 1x an E13 error. The 560ohm resistors are still in place so I am thinking about taking them out and repeating the same test with the bypass program. The Wemos D1 mini is running at 80MHz.

[17 more replies]

closed 2022-03-09 by visualapproach