visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #48: Compilation errors

TimE90, 2020-12-23:
Hey,

could you help with these errors in compilation process? 

```
Arduino: 1.8.13 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), dtr (aka nodemcu), 26 MHz, 80MHz, DOUT (compatible), 4MB (FS:2MB OTA:~1019KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

/Users/tim/Documents/Arduino/libraries/BWC_v2.0.0/BWC_8266.cpp: In member function 'void BWC::begin()':
/Users/tim/Documents/Arduino/libraries/BWC_v2.0.0/BWC_8266.cpp:346:13: error: 'D1' was not declared in this scope
  _cio.begin(D1, D7, D2);
             ^
/Users/tim/Documents/Arduino/libraries/BWC_v2.0.0/BWC_8266.cpp:346:17: error: 'D7' was not declared in this scope
  _cio.begin(D1, D7, D2);
                 ^
/Users/tim/Documents/Arduino/libraries/BWC_v2.0.0/BWC_8266.cpp:346:21: error: 'D2' was not declared in this scope
  _cio.begin(D1, D7, D2);
                     ^
/Users/tim/Documents/Arduino/libraries/BWC_v2.0.0/BWC_8266.cpp:347:13: error: 'D3' was not declared in this scope
  _dsp.begin(D3, D5, D4, D6); 
             ^
/Users/tim/Documents/Arduino/libraries/BWC_v2.0.0/BWC_8266.cpp:347:17: error: 'D5' was not declared in this scope
  _dsp.begin(D3, D5, D4, D6); 
                 ^
/Users/tim/Documents/Arduino/libraries/BWC_v2.0.0/BWC_8266.cpp:347:21: error: 'D4' was not declared in this scope
  _dsp.begin(D3, D5, D4, D6); 
                     ^
/Users/tim/Documents/Arduino/libraries/BWC_v2.0.0/BWC_8266.cpp:347:25: error: 'D6' was not declared in this scope
  _dsp.begin(D3, D5, D4, D6); 
                         ^
Mehrere Bibliotheken wurden für "WiFiUdp.h" gefunden
 Benutzt: /Users/tim/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WiFi
 Nicht benutzt: /Applications/Arduino.app/Contents/Java/libraries/WiFi
exit status 1
Fehler beim Kompilieren für das Board Generic ESP8266 Module.

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.
```

visualapproach, 2020-12-23:
As stated in FAQ https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/discussions/46#discussion-64636
Compilation errors has been solved by reinstalling the Arduino IDE, or using another computer. (Assuming you have not changed the code)

TimE90, 2020-12-23:
You can close this issue. I changed the board to the correct module.

closed 2020-12-23 by TimE90