visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #49: No data / controls not working

TimE90, 2020-12-27:
Hey!

I'v tried to get this to work with the 6-pin Palm Springs of my father. Board and WiFi are working but spa control shows no data and the display/controls aren't working. I've attached two photos of my board and wiring.

There are no shorts on the board and I've checked wirings multiple times. What I could detect with multimeter is that LV1,2,3(VA1,2,3) of my level shifter always <1V but LV(5,6,7,8) is 3.3v. On HV1,2,3(VB1,2,3) I have measured 4.5-4.9v and on HV5,6,7,8(VB5,6,7,8) there are constant 4.9V.

Have I swapped the display and pump connector? Or have I overlooked something? Maybe the level shifter (https://www.amazon.de/gp/product/B07N7FFY2Q/ref=ppx_yo_dt_b_asin_title_o05_s03?ie=UTF8&psc=1) is defective?

Thank you for the project and your support!

[image]

[image]

visualapproach, 2020-12-27:
Hey,
Check if the ESP is in AP mode.
Do you have a working setup at your own ? Maybe try that device in daddy's tub.
It appears in your picture that the display is working.
/Thomas

TimE90, 2020-12-27:
Hey!

No, it’s my first and only setup. The ESP is connected with WiFi and reachable over web.

The display is not working. It only shows the last value when direct connected with the pump. It doesn’t refresh or respond.

I know it’s difficult to debug. But maybe you have a good hint what’s not working when ESP and display aren’t receiving any data?

visualapproach, 2020-12-27:
Hard to tell from the pic but is 3.3 V connected to the LLC?

TimE90, 2020-12-28:
yap, it’s connect on the bottom of the PCB. Measured all levels and I think they are correct.

Did I connect the display and CIO in right direction?

Do you have other advices for debugging?

Thanks for your help!

[image]

TimE90, 2020-12-28:
What else I noticed is that when I turn on the power and the ESP hangs in between the display does not turn on. If I then connect the display directly to the pump and then again the ESP in between the display is goes on. However, no new data comes to / from the display and also the ESP shows no data in WiFi control. Please see the attached video. This is not normal behavior or?

When measuring VA1,2,3 is always only about 40mV. Where VB1,2,3 are 4.5-4.7V. Shouldn't the level shifter make 3.1-3.3V from 4.5-.4.7V?

Video: https://www.dropbox.com/s/ytt413138dncrwg/Video%2028.12.20%2C%2022%2042%2027.mp4?dl=0

visualapproach, 2020-12-28:
Dsp/Cio in the right direction: I can't tell from your pictures but I guess so since the display is on. At least the pwr is correct.
Measuring voltages on data pins only tell you so much. The DMM probably average over some time when it is switching rapidly (like 100000 times per second) between 0 and Vcc.
Best option is to use a logic analyzer (can be built from arduino if you don't have one)
That LLC may well be the problem. I don't know how that works but it seems the OE pin should be used in a particular way. Maybe try a simpler model with transistors and 10kOhms resistors instead?

TimE90, 2020-12-29:
You were right 🥇  The OE pin should be 3.3V instead of GND!

Thank you very much and have a happy new year! Maybe we can try to get the 4 pin model to work next year :)

closed 2020-12-29 by TimE90

visualapproach, 2020-12-29:
Super! Happy New Year!