visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #50: Heating always till 40°

Bankaifan, 2021-01-01:
Hi,
it seems, the target temperature is ignored when heating.
I always had 39° set and it heatet till 40° - nothing special, I live with it.

But now I set the target temp to 38° - the pool is still heating till 40°.
But while cooling down - the target temp works. 
It is always always target - 2.
So - with target 39 the pool heated up till 40 and cooled down to 37, then heated up again.
Now - with target 38 the pool heats up till 40 and cool down to 36. 

Any ideas how to explain this behaviour?

visualapproach, 2021-01-01:
Hi, the ESP is only setting the target temp by simulating you pressing up/down buttons. The actual regulation is done by the pump mcu. My pump stays within target +/- 1 C. Unless you stop the filter pump in which case the residual heat in the heater element will raise the temperature of the small amount of water in the pump by one or two degrees. If you keep the filter pump on and keep a constant target temp it should behave exactly as without the ESP. If it does not, I can't explain it.

wyliecoyoteuk, 2021-02-18:
That is normal behaviour for older spa controllers, which have a hysteresis of +/- 2C, but the maximum temperature they can display is 40C. If you set it to 40C, it will actually go to 42C, then drop to 38C.
Newer pumps (with "Freezeshield")  have a +/- 1C hysteresis.

closed 2023-11-23 by visualapproach