visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #54: Support for 2021 units

duststorm, 2021-01-11:
Hi,

It seems that many of the new 2021 models Lay-Z spa come with a new heating/pumping unit.

I don't know what its internals look like, but it looks like this on the outside.
[image]
[image]

Has anyone already experience with these? I'd love to have a look inside, and know whether they are easily compatible.

dwoosnam, 2021-04-02:
I have this pump that came with a Lay-Z-Spa Rio. I'm currently having some difficulty in getting the display to turn on when using this circuit - I have posted more in the "Discussions/chat."

[image]
[image]
[image]
[image]
[image]
[image]

dwoosnam, 2021-04-03:
### A Look inside the 2021 pump
**So Remember, this is the 2021 pump that's boxier in shape compared to the egg pump.**

[image]

Things seem a lot more compact inside compared to the egg shape pump. 
[image]

The 6 pin contacts
[image]

Coloured cables are display side, white with red are pump side.
[image]

I opened up the touch display and checked the connections, and it seems like my 5v and GND should match up (my red and orange in the next pic.
[image]

dwoosnam, 2021-04-03:
### [Following on from Discussion/Chat - EDITED]

Getting so close with my 2021 "Cube" pump.

I am currently using this LLC - [https://amzn.to/3sNUL8S](https://amzn.to/3sNUL8S)
[image]

I have connected VA and OE on the LLC to 3.3v on the NodeMCU and after a short while (I imagine the NodeMCU boot time?) now have display (which scrolls "hello") and functionality of the buttons via the pump control panel! 

**The function of the pump control panel buttons works as intended.** - although slightly delayed
Bubbles = Bubbles
Heater = Heater
Pump = Pump
C/F = Unit Toggle switches back and forth (can we make C the default?)
Etc....

When using the control panel buttons on the pump, the web interface reacts accordingly.
**PUMP buttons >>> Web Interface actions**
Bubbles = Bubble toggle switches on and turns green
Heater = Heater toggle turns on
C/F = Unit (c/f) toggle switches accordingly

However, manually changing the toggles from the web interface causes some unusual effects.

**WEB Interface control  >>>  PUMP actions**
Unit (F/C) = Toggles the heater
Bubbles = Toggles the timer button
Pump = Toggles the pump/filter (yay!)
Heater = Toggles the bubbles

Again, any helpful hints would be most welcomed.

visualapproach, 2021-04-03:
Do you have latest version on both the code AND the webpages uploaded to the ESP?

dwoosnam, 2021-04-03:
As far as I'm aware (n00b here).

I copied the BWC_v2.0.0 folder into my Arduino/libraries folder. I'm then using the MQTT image to flash.

visualapproach, 2021-04-03:
Just to be sure, with the example .ino open, upload the data also:

[image]

If it is still the same, do this in Chrome browser:

go to page YOUR_ESP_IP/WebSocket.js
Copy the page and paste in a new comment in this issue.

dwoosnam, 2021-04-03:
I remember doing the first part as the website was just giving me a 404 when I tried before doing that.

```
//
var connection;
connect();

function connect(){
	connection = new WebSocket('ws://'+location.hostname+':81/', ['arduino']);
	connection.onopen = function () { 
		document.getElementById('header').style = "background-color: #00508F";
	};
	connection.onerror = function (error) {
		console.log('WebSocket Error ', error);
		document.getElementById('header').style = "background-color: #FF0000";
		connection.close();
	};
	connection.onclose = function(){
		console.log('WebSocket connection closed, reconnecting in 5 s');
		document.getElementById('header').style = "background-color: #FF0000";
		setTimeout(function() {
			connect();}, 5000);
	};	
	connection.onmessage = function(e){
		handlemsg(e);}
} 

String.prototype.pad = function(String, len) { 
                var str = this; 
                while (str.length < len) 
                    str = String + str; 
                return str; 
            }

function handlemsg(e) {  
	var msgobj = JSON.parse(e.data);
	var mycolor;
	console.log(msgobj);
	
/*     
	doc["LCK"] = _cio.states[LOCKEDSTATE];
    doc["PWR"] = _cio.states[POWERSTATE];
    doc["UNT"] = _cio.states[UNITSTATE];
    doc["AIR"] = _cio.states[BUBBLESSTATE];
    doc["GRN"] = _cio.states[HEATGRNSTATE];
    doc["RED"] = _cio.states[HEATREDSTATE];
    doc["FLT"] = _cio.states[PUMPSTATE];
    doc["TGT"] = _cio.states[TARGET];
    doc["TMP"] = _cio.states[TEMPERATURE];
    doc["CH1"] = _cio.states[CHAR1];
    doc["CH2"] = _cio.states[CHAR2];
    doc["CH3"] = _cio.states[CHAR3]; */
	if(msgobj.CONTENT == "OTHER"){
		//if(msgobj.MQTT){
			document.getElementById('mqtt').innerHTML = "MQTT:" + msgobj.MQTT.toString();
		//}
	}
	if(msgobj.CONTENT == "STATES"){
		document.getElementById('atlabel').innerHTML = msgobj.TMP.toString();
		document.getElementById('ttlabel').innerHTML = msgobj.TGT.toString();
		var element = document.getElementById('temp');
		if (element.value == 0) element.value = msgobj.TGT;
		document.getElementById('sliderlabel').innerHTML = element.value.toString();
		document.getElementById('AIR').checked = msgobj.AIR;
		document.getElementById('UNT').checked = msgobj.UNT;
		if(msgobj.RED) {
			mycolor = "background-color: #FF0000";
			}
		else {
			mycolor = "background-color: #00FF00";
[... 139 lines cut ...]
	sendobj["INTERVAL"] = 0;
	connection.send(JSON.stringify(sendobj));
	console.log(JSON.stringify(sendobj));
}
function reboot() {
	var sendobj = {};
	sendobj["CMD"] = rebootesp;
	sendobj["VALUE"] = 0;
	sendobj["XTIME"] = 0;
	sendobj["INTERVAL"] = 0;
	connection.send(JSON.stringify(sendobj));
	console.log(JSON.stringify(sendobj));
}

function zero() {
	var sendobj = {};
	sendobj["CMD"] = resettimes;
	sendobj["VALUE"] = 0;
	sendobj["XTIME"] = 0;
	sendobj["INTERVAL"] = 0;
	connection.send(JSON.stringify(sendobj));
	console.log(JSON.stringify(sendobj));
}

```

visualapproach, 2021-04-03:
How bizarre. In chrome again - shift+ctrl I (right click/inspect) and click the tab "console". Press UNIT on webpage and there should be a message in the console "CMD:1, VALUE:1, XTIME:0, INTERVAL:0"
Can you confirm that please. CMD should be "1".

visualapproach, 2021-04-03:
When you press buttons on the pump with the ESP connected, does the pump behave as expected? (Bubbles = bubbles and so on)

dwoosnam, 2021-04-03:
It's cold out, so I've closed the pump back up and I'm just powering the NodeMCU via MicroUSB on my desk now, so this might affect readings.

But, as requested....

Unit:
`{"CMD":1,"VALUE":true,"XTIME":1617474336,"INTERVAL":0}`

Bubbles:
`{"CMD":2,"VALUE":true,"XTIME":1617474371,"INTERVAL":0}`

Pump: 
`{"CMD":4,"VALUE":true,"XTIME":1617474403,"INTERVAL":0}`

Heater:
`{"CMD":3,"VALUE":true,"XTIME":1617474423,"INTERVAL":0}`

When using the buttons on the pump, they work as expected. And, the strange thing is, the related items on the web interface change to match.

eg.
- Press bubbles button on pump
- Bubble motor comes on
- Web interface changes to show bubbles are on.

If I then click the bubbles toggle on the web interface, the bubble motor stays on and the display/control panels acts as though I've pressed the timer button.

visualapproach, 2021-04-04:
[dwoosnam.zip](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/6254365/dwoosnam.zip)

Okay, replace your files in the library/BWC... with attached files. (Unzip to the library/BWC... directory)
Recompile and upload.
Then with everything connected, go to chrome console as above. Press a button on the web interface, like UNIT. Note what it says in console "OTHER.....lastButton:"
Press UNIT on the pump and look if it is the same.
You can copy the console and post here if you want, but try to mark where you press different buttons.

dwoosnam, 2021-04-04:
This is a difficult task. Being bombarded every second with information in that console window, along with some delays on button presses from the web interface side, and confusion when pressing web interface buttons in anger.

### control panel:

**lock:**
LastButton: "10"

**timer:**
LastButton: "30"

**unit:**
LastButton: "89"

**heat (turns on heat and pump):**
LastButton: "1012"

**bubbles:**
LastButton: "1212"

**pump:**
LastButton: "1112"

**down:**
LastButton: "1312"

**up:**
LastButton: "20"

**power:**
LastButton: "00"

### web interface
**unit:**
LastButton: "00" ----> doesn't change at all, though pressing in anger resulted in a "1112"?

**pump:**
LastButton: "1112"

**bubbles:**
LastButton: "1012"
x2 LastButton: "30"

**heater:**
LastButton: "1212"

There was some weird goings on with the temperature slider, different outputs that seemed to be where i put the slider along the bar
**Set target temperature down:**
LastButton: "1312"

**set target temperature up:**
LastButton: "89"

visualapproach, 2021-04-04:
LOL, I wonder how the controller can tell the difference between an ordinary press and a press in anger!
It appears they (Bestway) has changed some button codes on this model, but it's not 100% consistent. Maybe has to do with it beeing a difficult task to read the info scrolling by. I think I'll do another version for you to run. Call you back but maybe not today.

dwoosnam, 2021-04-04:
A press in anger is more me just pressing the same button multiple times just to get it to do what it's meant to rather than something else. Haha.

I wish I had the skillset to check this all myself. I do appreciate your time and effort, as I'm sure will the others that get this pump that are looking into making it smart. Cheers.

visualapproach, 2021-04-05:
Hi again @dwoosnam replace your files with this version. Connect it to the pump and go to webinterface. There is a new row at the top that reports a special code for every button you press on the display control panel. Note the corresponding code for all the following:

NOBTN
LOCK
TIMER
BUBBLES
UNIT
HEAT
PUMP
DOWN
UP
POWER

[dwoosnam2.zip](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/6257121/dwoosnam2.zip)
(Use the mqtt with gpio example)

dwoosnam, 2021-04-05:
Nothing obvious of note shows in the GUI, but in Inspect/Console I can see the _PressedButton:_ codes, which differ to the _LastButton:_ codes I previously sent, I hope this is what you want.

NOBTN - This seems to remember the last button press from before a reboot. But after a while has changed to "_ffff_"
LOCK 0100
TIMER 0300
BUBBLES 1212
UNIT 0809
HEAT 1012
PUMP 1112
DOWN 1312
UP 0200
POWER 0000

visualapproach, 2021-04-05:
If you also upload LittleFS files, the webinterface will be updated :-)
Or you can go to YOUR_IP/upload.html and browse for the new index.html-file in the data-folder and upload to the ESP.
Not important, but it makes it easier. When testing buttons you will have to keep them pressed e few seconds so you catch the button when the page is updating.
There should be a code like 1B1B when no button is pressed. 
[image]

dwoosnam, 2021-04-05:
Updated and now see the button press option in the GUI. They match all the previous codes I got from the console.

There isn't a 1B1B as such. The no-button FFFF only seems to show when turning on for the first time. Each command is repeated contstantly in the console until the next button is pressed as per screenshot. (excuse the touchpad writing!)
[image]

NOBTN ffff
LOCK 0100
TIMER 0300
BUBBLES 1212
UNIT 0809
HEAT 1012
PUMP 1112
DOWN 1312
UP 0200
POWER 0000

visualapproach, 2021-04-05:
hm, No button is not shown here either. I'll see if I can get it to display correct

visualapproach, 2021-04-05:
Stupid cut&paste error 
Here, replace with this file, and recompile+upload. No need to update LittleFS data files this time.
[BWC_8266.zip](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/6258037/BWC_8266.zip)

visualapproach, 2021-04-05:
When we establish NOBTN code, I'll update another file for you to test, and hopefully everything will work fine

dwoosnam, 2021-04-05:
After recompiling and uploading, it's still not giving me a direct no button code. (I hope i'm right in assuming NOBTN translates as "no button")

On a fresh power on from the mains, it's showing me code 0809, even without pressing anything and the console keeps giving the same output every 2 seconds.

visualapproach, 2021-04-05:
You are assuming correct. Maybe I should explain exactly what happens, so we talk about the same thing.

Every 2 seconds or so the ESP sends a request to the control panel. The control panel answers with  16 bits (two bytes, or one word if you like). Depending on what buttons are pressed that exact moment, the answer will be some particular value. So in my case, when I don't press any button the answer will always be 1B1B (Hexadecimal representation of the bits 0001 1011 1011). The FFFF at startup can be disregarded since that is likely because all pins are high (1) before data communication has started properly. 16 bits of 1's is hex FFFF. 
So, after startup sequence is finished (give it a minute or two, to establish wifi, time etc) you SHOULD se a constant value like 1b1b or whatever they decided in your model. And pressing a button while the request is being done, should change the value to 1212 or whatever. And after releasing the button you should see the same value again, like 1b1b.

When you press LOCK button on the control panel I guess it says code:0100. What happens when you release that button? Still 0100 or switching to 0809 or anything else?

dwoosnam, 2021-04-05:
Thank you for your explaination.

It seems that the answer to your last question is that on the press of a control panel button, the command changes. But it continues to return that same code of the last button press until the next button press. The following example shows a long pres on UNLOCK, UNIT, HEATER and then UP.
[image]

For example. I turned bubbles on and then bubbles off, then disconnectecd the power, and reconnected the power. This shows that on boot up (after the websocket disconnection) the first command is 0809.
[image]

When the unit was set to C, it seemed that on a power cycle, it reverted to F. But when on F, it doesn't revert back to C. despite the 0809 code appearing to be the first button press after the reboot.

a reboot of the ESP from the web interface seems to keep the last button press in memory.
[image]

And then another power cycle at the mains thows 0000 into the story to mix it up....
[image]

visualapproach, 2021-04-05:
You make a great job logging your testing! Appreciate that. It appears that the new BWC_8266.zip is not in play. I can be wrong, but to be absolutely sure I give you another file with a unique text so we can tell what version is actually running. I think you know what you are doing but sometimes the IDE is playing tricks on you, an sometimes other things go wrong.
Recompile and upload with this file in the library:
[BWC_8266.zip](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/6260460/BWC_8266.zip)
Then in console there should be a (v3) suffix to the codes

dwoosnam, 2021-04-05:
Thank you for your kind words, I'm glad it looks like I know what I'm doing! :D It's my first ESP project. It's in my interest to give you as much information as possible. I don't want to waste your time, which I am very thankful for!

Just to be clear;

- I re-downloaded the dwoosnam2.zip and extracted it to a dwoosnam3 folder
- I downloaded the last v3 BWC_8266.zip and extracted that.
- I copied the new BWC_8266.cpp to the dwoosnam3 folder
- I moved the dwoosnam3 folder to Arduino/libraries
- I restarted Arduino IDE
- Open > Libraries > dwoosnam3 > Basic_MQTT_with_GPIO
- Updated MQTT credentials
- Verify > Upload

So I _should_ be ready to test again tomorrow, too dark now. I hope I'm not going to wear the screws out on this new pump!

visualapproach, 2021-04-05:
Oh, yes - Make sure you only have one copy of the files in library folder or arduino IDE will choose the wrong version! Move every copy of BWC... out of library except the one you want to use. Annoying as f but that's the way it is.

dwoosnam, 2021-04-06:
Got rid of all the other versions in my Arduino/library folder and re-uploaded last night.
I am back in work this week, but have managed to get the ESP all connected into the pump this morning and it seems to be happily spitting out 1b1b(v3) in the console. I hope that this is a good thing.

visualapproach, 2021-04-06:
Great! That's good. I'm also at work this week but I'll adjust the program to the new codes. Call you back :)

visualapproach, 2021-04-06:
@dwoosnam -> Updated with 2021 codes. Crossing fingers.
[BWC_8266.zip](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/6266119/BWC_8266.zip)
Check if this works as expected. Replace old files and compile, upload and upload LittleFS! Did you know you can upload over the air so you don't have to use the screwdriver every time?

dwoosnam, 2021-04-06:
I've seen there's an OTA update available.... But I can't find the instructions. Why did I wait this long to check about it?! 🤣

visualapproach, 2021-04-06:
Have the ESP on and then start arduino IDE and you should see a port "BWC at 192.168.xx.yy" along with the COM1 etc. If you're on a PC that is. Sometimes if it's not showing it helps to restart the pc.

dwoosnam, 2021-04-06:
OTA has just opened a new world for me, especially as it's started snowing here today for some reason! I have an issue in that LittleFS doesn't seem to like updating OTA, I get an authentication error - though it doesn't prompt for user or password.

The lid currently isn't screwed on properly anyway, so I'll drag it back in for a final flash and a bit of glue gun love on the bread board because I'm not going to change a thing now!

But... That aside, your genius shows no bounds. My web interface controls the things it's meant to, as does HomeAssistant/MQTT (found in the discussions).

I feel like I've taken a massive amount of your time and you and your patience should be rewarded! Thank you so much!

Out of curiosity now, does this get linked into the existing code, for any future updates, or is this strictly a 2021 pump code, just for me it seems!

visualapproach, 2021-04-06:
> OTA has just opened a new world for me, especially as it's started snowing here today for some reason! I have an issue in that LittleFS doesn't seem to like updating OTA, I get an authentication error - though it doesn't prompt for user or password.

Yes it's an annoying bug in the IDE I think, but if you first upload the sketch and give it the password, and directly after upload LittleFS it usually works. If it doesn't just visit the /upload.html page and upload the new files. In this case only index.html and websockets.js

> But... That aside, your genius shows no bounds. My web interface controls the things it's meant to, as does HomeAssistant/MQTT (found in the discussions).

That made my day, and I forced the kids to read that, haha. Thank you!

> I feel like I've taken a massive amount of your time and you and your patience should be rewarded! Thank you so much!

No problem, I want it to work on newer models as well, and now it does thanks to your testing and time.

> Out of curiosity now, does this get linked into the existing code, for any future updates, or is this strictly a 2021 pump code, just for me it seems!

I am going to update existing code. Just have to decide how one should choose the right pump. Probably a checkbox or something in the webinterface and an MQTT command.

Cheers, and thanks for contributing!

dwoosnam, 2021-04-06:
Happy to help. My kids will just be happy that they can use the hot tub again because I wont have the pump half open!

I want to look into the audio/beep/sound feature, but I'll give it a break for that! 

Thank you again for all your help. Take a relax in your tub!

visualapproach, 2021-04-11:
Updated the code to support 2021 models. You need to edit a line in a file to enable, see readme on main page.

sjeffrey101, 2021-04-12:
wow..... well done guys.  this is great.  I almost brought a new unit 2 weeks ago as my heater element has failed.  But didnt want to loose this functionality so i managed to locate a used heater, just need to swap it over, and hopefully it will work.  IF not i'll be getting a new one for sure.

dwoosnam, 2021-04-16:
Just to update this I have managed an OTA update to the latest version - with a manual update of each of the web files through the web interface - and all is working ok. The all controls still function over MQTT in HomeAssistant.
[image]

visualapproach, 2021-04-16:
I really like you people in this community! Smart, friendly and bother to report success!

rucati, 2021-04-18:
Hi
As I am waiting for the level shifters and cables to come in, would it be possible to add:
1) extra pictures on how to safely disassemble the 2021 shell (adding pictures or video on how to disassemble the unit past the touch panel all over to the main body shell, not breaking the plastic bits so I can reassemble too! I am a bit lost after unscrewing and taking off the lid from the touch panel).
2) add MQTT and Lovelace settings/files to get HA setup up and running? (or where to find them elsewhere)
Thanks for all the effort so far! As I have the 2021 too, I was hoping for this post to come! Great project and community!

visualapproach, 2021-04-18:
> Hi
>
> As I am waiting for the level shifters and cables to come in, would it be possible to add:
>
> 1) extra pictures on how to safely disassemble the 2021 shell (adding pictures or video on how to [...]

Hi,

1. I don't have one so someone else maybe can tell.
2. There are some examples in this forum. When I get my tub up again I might share the setup. I also have HA.
Thanks and have a nice day

dwoosnam, 2021-04-18:
@rucati Taking the lid off is easy it's just 6 screws underneath the ridge around the middle of the pump.
Gently lift the top half as it is only connected by the display cable. (Which our esp8266 sits in-between)
Check my photo post near the top of this issue.

As for the Lovelace / HA setup, I just copied all the config from #52 and used that and set it up as above.

dwoosnam, 2021-05-10:
Love this idea @TheRocketSurgeon! I feel my solar setup is pretty locked down, though it's possible that I just down know how to access it.

visualapproach, 2021-05-15:
I don't have solar power and know little about it, but just to point out an option - I have installed a Shelly 3EM module (energy monitor on incoming power to the house) on which I can see current (no pun intended) energy consumption and also returned energy. Zero in my case, but that could be used to trigger the hot tub on or off. I have it set up in Home Assistant such that heating turns off before my breakers pop, and then on again when the load is lower. The Shelly can even use MQTT but I haven't needed that since there is an integration already in HA.

GSzabados, 2021-05-16:
> Exactly what I thought when I woke up this morning! Mine's Solaredge if that happens to match.

SolarEdge has an official integration for Home Assistant (cloud) and a local custom component through modbus. (If you do use Home Assistant.) It makes easy to match the solar with MQTT and add some smart control.

There is a topic of the Wifi enabled hot tubs ([On the HA Community Forum.](https://community.home-assistant.io/t/lay-z-spa-hot-tub-wi-fi-pump-automation/229006)), where people shared some ideas how do they use solar power with the hot tub.

I am just playing with the idea to purchase a Maldives, but what really stopping me at this point is the missing wifi component. I saw that some work has been done already and it looks promising. #64 
The possibility of the integration to HA makes it so tempting, disregarding that the solar investment is still on the way for the house.

@dwoosnam, what solar model do you have? Most of them has some sort of API solution if they are connected to any monitoring. (Growatt, Enphase-Envoy, Huawei, Fronius, Tesla, etc. just to mention a few.)

dwoosnam, 2021-05-16:
@GSzabados without going in the attic to check, I think it's ABP.  The usage stats do get sent back to company that installed it, but I was under the impression that it was sent by SMS. It was installed about 7 years ago.

GSzabados, 2021-05-16:
> I think it's ABP

That is a tough one. To be honest, I have never heard of it. But I have seen a Growatt documentation recently dated from 2015 where they already had an API to access data. But if you are not aware what method it using for transfer, then most likely you are right and it would be something simple, like SMS or GPRS/EDGE.
IoT evolved a lot during the past 7 years. But that doesn't mean your inverter has no way to communicate any other way. ModBus is one of the good examples.
I would ask the installer that is there any option to get real time data. Maybe installing a smart meter, or having a smart meter with some output would put you at the same position, as @visualapproach suggested with the Shelly 3EM.
Don't you have a smart meter installed by any chance from the electricity supplier?
(Many companies has additional dongle options for their inverters as well, for wifi or 3G/LTE connection, but with a 7 years old model that is a good question.)

dwoosnam, 2021-05-16:
@GSzabados we had a smart meter installed the other day though from a different electricity supplier to the solar panels.
At the moment, as I understand, the setup is as follows:
Panels > attic (inverter > solar electricity meter (with sim card?)) > Ground floor( Main distribution board and smart meter)
I suppose an email to the solar company would be a good starting point. The worst they can say is no! Thanks!

GSzabados, 2021-05-16:
@dwoosnam, generally the inverter will provide production data. The smart meter will provide production and consumption data as well. (Basically what you used from the grid and what you produced to the grid.)

I would try to get some information about the smart meter first. The state of the actual production what you will really need, as you would like to know when you have excess power to spare.

I don't know which country are you from, but in Poland there is a kind of gentleman's agreement where your produced excess power is used by the grid, and you can get it back from the grid later at 80%. It makes sense as the electricity provider works as a storage and the yearly contracts uses those during the winter period. Anyhow...

If you are from the UK, there are most providers have smart meters and some has public access to the data, or the meter can be queried locally. I came across recently a custom made SmartThings integration for Geo connected smart meters, and someone posted that it was ported to OpenHab as well.

https://community.smartthings.com/t/smart-meter-uk-build-an-app-device-handler-for-api/214649

I would try to get some information what the smart meter can he used for and how you can access real time consumption data.

cyberfly79, 2021-06-03:
Thank you for your work!

I have a 2021 Miami.
I haven’t disassembled yet, I just put the controller together.
I have never worked at platformio before. I got an error compiling the code.

I had to write to line 13 of the file BWC_8266.h: #define ARDUINOJSON_USE_LONG_LONG 1

Then I was able to upload it and it works on the test bench.

visualapproach, 2021-06-06:
> Thank you for your work!
> 
> I have a 2021 Miami.
> I haven’t disassembled yet, I just put the controller together.
> I have never worked at platformio before. I got an error compiling the code.
> 
> I had to write to line 13 of the file BWC_8266.h: #define ARDUINOJSON_USE_LONG_LONG 1
> 
> Then I was able to upload it and it works on the test bench.
> 
> .pio\libdeps\nodemcuv2\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp:43:5: error: static assertion failed: To use 64-bit integers with ArduinoJson, you must set ARDUINOJSON_USE_LONG_LONG to 1. See https://arduinojson.org/v6/api/config/use_long_long/
> 43 | ARDUINOJSON_ASSERT_INTEGER_TYPE_IS_SUPPORTED(T);

Hi,
Strange, I don't get that error. Did you download from master branch or development branch?

cyberfly79, 2021-06-11:
My txs0108 LLC has finally arrived today (does not work with txb0108).
Its control works perfectly, I get the temperatures.
Unfortunately, the display does not work, the button lights flash (disco).
The device is a 2021 Miami. I put it together on a breadboard, I have already ordered the PCB.

Can you give me some advice to turn on the display as well?

Thanks!

cyberfly79, 2021-06-11:
Looks like I found the solution. A 560 Ohm resistor was needed per pin between the LLC and the display. I think we can say that it also works with 2021 Miami.
Thanks again!

visualapproach, 2021-06-11:
Okay, that's strange. I mean it's teriffic that it works, but I wonder why the resistors is needed. Maybe attenuates noise? Good to know though. Thanks for confirming the model compatibility. 
@cyberfly79 and @dwoosnam could you be so kind to check and report the model number on your pump please. So I can update the compatibility table.

cyberfly79, 2021-06-13:
I took a photo of the Miami datasheet.
I noticed an interesting thing, that doesn’t affect the operation. I couldn't figure out what could be causing it.
I made short videos of it.
I uploaded everything to dropbox.

https://www.dropbox.com/sh/w3rtz3kutip1op8/AAC_FSpbCnHjJcV46CnmbCKia

What do you think could cause it?

visualapproach, 2021-06-13:
> I took a photo of the Miami datasheet.
> I noticed an interesting thing, that doesn’t affect the operation. I couldn't figure out what could be causing it.
> I made short videos of it.
> I uploaded [...]

I think that's communication errors. Most likely between CIO and ESP. How did you decide you needed the resistors between DSP and ESP? Trial and error or any thoughts behind it? Maybe it's needed to the CIO also.

dwoosnam, 2021-06-14:
> Okay, that's strange. I mean it's teriffic that it works, but I wonder why the resistors is needed. Maybe attenuates noise? Good to know though. Thanks for confirming the model compatibility. [...]

My 2021 model is S100102.

cyberfly79, 2021-06-20:
Yes, there may be communication errors.
I tried the resistors but it didn't work out. Needed between DSP and ESP because otherwise it just flashes.
I have noticed that sometimes an mqtt message arrives, as if the pump or bubble is on. But usually the off also comes in the same second.
I think for a moment the DSP shows that in a flash.
Sometimes the pump will start for a moment.

The frequency is very variable. Interestingly, there is no bad data coming for heating.  At TMP, 3 came instead of 33, when 1 came incorrectly at the pump.
{"CONTENT":"STATES","LCK":0,"PWR":0,"UNT":0,"AIR":0,"GRN":0,"RED":0,"FLT":1,"TGT":35,"TMP":3,"CH1":32,"CH2":51,"CH3":32}

Has anyone else had this kind of problem? I'm thinking of some kind of error handling (comparison to the previous value), but that would slow down the operation. Could there be some overflow?

No idea what to do with it? Not so confusing, but still something is not good.
I use it for Home assistant with mqtt switch.

visualapproach, 2021-06-20:
> Yes, there may be communication errors.
> I tried the resistors but it didn't work out. Needed between DSP and ESP because otherwise it just flashes.
> I have noticed that sometimes an mqtt message [...]

Hard to tell when I don't have anything to test on, but this is my thought so far:
- One more user experienced the flashing DSP and tried the resistors, but had other problems.
- Some users report success with 2021 Miami models. Is it only Maldives that is problematic?
- I'm not sure why the resistors help. Maybe the CIO is using a higher transfer speed, that ESP has problems keeping up with. And in that case the resistors would attenuate the sharper edges that may cause reflections in the wires. Possible work around is to overclock ESP to 160 MHz. (insert this
[env:myenv]
; set frequency to 160MHz
board_build.f_cpu = 160000000L 
in platformio.ini)

visualapproach, 2021-06-25:
Hi all, Can someone test the new code (development branch) with the 2021 Maldives (the one with hydrojets) and film the display during startup? I would like to see if the LEDs are matched correctly, so the status will be correct in the web page.
Thanks in advance / visualapproach

[24 more replies]

closed 2023-12-04 by github-actions[bot]