visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #56: issue with version 1.28

starsx2016, 2021-03-14:
Hi, I made a pcb with an wemos d1 mini , when I plug my spa (miami 6pins) I got the melody but nothing on the display, I've check voltage on the HV side of the LLC and every line of the CIO and DSP I got 4.96v to 4.99v but the control don't light up and on the web interface no button working. before someone ask I try version 2 as well no luck no chime , no display and web interface not controling anything. I have no idea what's wrong here...  if it can help I'll try to upload pictures.

thanks

starsx2016, 2021-03-14:
[image]
[image]
[image]

visualapproach, 2021-03-14:
Hi there!
First thing I noticed was a lot of cold soldering. Use flux to get better joints. There might be open circuits even if it looks like there is contact.
Next, I see you have a similar LLC as someone else was having problems with in this "forum". Switching polarity on the OE did the trick but I can't remember if it was supposed to be tied to gnd or vcc. But it's in here somewhere if you look around. Maybe closed issues.

starsx2016, 2021-03-14:
So I swap the wemos d1 mini for an nodemcu clone and I got something on the screen 'have a nice spa' follow by the IP address , then after few second everything goes dim and touch button not work or either on the web interface , so I think my CIO is not able to send command to the esp via the LLC I have so I'll wait for the same as you to be deliver and will see.

thanks for you help and your good work with this project :)

visualapproach, 2021-03-14:
Thank you and good luck!

starsx2016, 2021-03-19:
With a new LLC it's working like a treat! :)

closed 2021-03-19 by starsx2016

visualapproach, 2021-03-19:
Super! Great work and thx for sharing!