visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #60: 404 error

TheSimTek, 2021-04-22:
As discussed in "No Connection #53" I have connected to my local network but cannot access anything after setting up the Auto Portal.

Apologies for my lack of understanding in coding.  I am reading a lot but cannot get over the line on this one.

Refer attached images.  Appreciate the fine work you have done and any help you can offer.  

[image]
[image]
[image]

visualapproach, 2021-04-22:
Are you using spiffs? Should use littlefs.

TheSimTek, 2021-04-22:
I not sure?  Coding isn't my strong point.  Hardware no issue, however haven't touched software in over 20 years.

I thought I followed your install instructions but perhaps I am missing the finer details.  How do I change the setting to LittleFS rather than the out dated spiffs.

Thanks again for your amazing work on this project.

visualapproach, 2021-04-22:
You are welcome!

Link is on the main page, but easy to miss:
LittleFS install:
https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html#uploading-files-to-file-system
Link to LittleFS upload tool: https://github.com/esp8266/arduino-esp8266fs-plugin/releases/

Then you will have a new option in Arduino IDE to upload LittleFS. Now, Arduino IDE is not always asking for a password when uploading files, so you would have to first upload the sketch again (with password "esp8266") and then upload files.
Good luck!

TheSimTek, 2021-04-22:
Thanks.  I didn't miss that information.  I downloaded the ESP8266FS-0.5.0.zip upload tool and using the instructions setup the Arduino IDE so that it appears in the tools menu.  It appears in the second image above, but as spiffs and not littlefs.  Nowhere has it asked for a password.  I will attempt a fresh load of sketch and files.

TheSimTek, 2021-04-23:
My apologies once again.  It looks like I may have put the tool in the wrong spot.  In the Arduino IDE under tools there is ESP8266 Sketch Data Upload but no LittleFS Data Upload.  I did not realise the difference.

I am running Windows 10 on a laptop with the Arduino application running off a second drive.  I have put the ESP8266FS zip file in these locations without success -

C:\Users\Win-10\AppData\Local\Arduino15\tools\ESP8266FS\tool
D:\Arduino IDE\arduino-1.8.13\tools\ESP8266FS\tool

Neither location obviously recognised by the program.  Any thoughts?  I will continue to investigate where it should go.  Thanks again for your support.

visualapproach, 2021-04-23:
You need to extract the files and end up with the file "esp8266littlefs.jar" in the folder Arduino/tools/ESP8266LittleFS/tool/
In the Arduino/tools/ folder there should be two subfolders. One for ESP8266FS (SPIFFS) and one ESP8266LittleFS.

visualapproach, 2021-04-23:
Oh, I see now the link points to the wrong place. this is littlefs tool:
https://github.com/earlephilhower/arduino-esp8266littlefs-plugin/releases
v2.6.0

visualapproach, 2021-04-23:
Great! The table with models is mostly gathered from internet sources. This community is welcome to report more accurate info to add to it.

TheSimTek, 2021-05-03:
I can confirm the project works with the Palm Springs (6 wire) which I have.  

A few things I have noticed -

1- the audio is attenuated (definitely sounds softer with the hack installed)
2- if attempting too many things too quickly (changing heat, turning heater on/off, bubble on, changing heat) it occasionally locks up and the top panel fades and becomes unresponsive until powered off and back on
3- creating a command que to turn on the heater at 10am set to 38 degrees, then a new command at 4pm to turn on heater set to 28 degrees actually turns off the heater.  The idea here was to utilise the solar to heat the spa during the day.

The code from 6 months ago did not work with the spa.  The code uploaded on the 21 April works.  Tested a couple of times to confirm.  No idea why.

Initial build included the 7 resisters but as it was not working I jumpered them out.  Issue believed to be the older code as it now works.  Jumpers left in.

Minor issues, but some fantastic coding.  Will continue testing.  Thank you for an amazing project.

[image]
[image]

visualapproach, 2021-05-03:
Hi, I'm happy you like the project!
1- I've heard someone else mentioning this. Don't know why that is. If you have an oscilloscope you could compare the signal levels. I don't have one.
2- okay, I don't experience that but maybe I'm too slow :) Probably the ESP hangs or reboots. It's got limited resources.
3- If you tried to 'set heater' to value 38, it will confuse the device. That command takes a Boolean value. 0/1. Use 'set target temp' instead with value 38. There is no error checking for this unfortunately.

TheSimTek, 2021-05-03:
Thanks for the response.  
1- if I get time but not a big deal
2- was pushing it hard, haven't repeated this exercise and it has been stable for 2 days
3- understood.  command to change heat value and will set pump on as required.

Now will read up on MQTT and how to trigger via solar power.

closed 2021-05-03 by TheSimTek