visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #61: No beeping / music on 2021 pump

scfede, 2021-04-30:
On the 2021 pump there is no beeping while touching a button on the display. And even if activated in the settings page. there is no music on startup either :-( Not a big bang, but i want to let you know.

Please don't hang me up right away, it's my first ESP project/hack ;-) 
I have installed the latest BWC_v2.0.0 code which works like a charm (2021 pump mode active). All buttons working well and the webinterface is acting good. So from my point of view there should not be any issue with my hardware/pcb/llc etc. I have used Eric's PCB with two 4ch llc. The beeper works well, when the display is directly connected to the pump.

Any ideas to figure out the problem / troubleshoot it?

 BTW: Fantastic work! Thank you very much fo this nice project.

visualapproach, 2021-04-30:
Hi @scfede , I don't hang up on anyone as long as they are polite :). The beep (on the pre2021) is just the line switching between high and low at the tone frequency. Check connection between pin 6 on DSP and the corresponding pin on the LLC. And likewise on the 3.3V side. Anyone else with 2021 having this issue? Thanks for reporting!

dwoosnam, 2021-05-03:
Have the 2021 and have noticed that when it's enabled from the config page, the beep is very quiet. It's a lot quieter than standard. It is there for me, though it really could be missed with a slight breeze or the hose running.

scfede, 2021-05-08:
Dammit, it found the reason. I missed R7 entirely as it sits a bit apart the other resistors on the 2x4chan PCB. Shame on me! And just to confirm @dwoosnam, yes it's beeps are much more quiet than before. But that's no problem for me. Sorry for bothering ;-)

closed 2021-06-12 by scfede