visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #64: Lay-Z-Spa Maldives (2021)

mhl66, 2021-05-04:
What a great project!!! But: no luck with my 2021 [Lay-Z-Spa Maldives HydroJet Pro](https://support.bestwayaftersales.co.uk/products/lay-z-spa-maldives-hydrojet-pro-bw54173/). Either on self-wired perfboard as well as on the Naesstrom PCB - both devices do not work at all, yet I carefully checked all wirings, and the I/O basics seems to be working: the temperature readout and C/F status are OK, but nothing else happens, no keys are processed, and no commands are recognized by the pump controller:
[image]
(it's the MQTT-enabled version, in black color scheme)
Deeply dedicated to scanning logs and reengineering the model - does anyone have encountered the same problem? What would @visualapproach  recommend?

visualapproach, 2021-05-04:
Thanks, I'm confused. I reckon that's a 4 wire pump right? And there is no "mqtt enabled" example for that unless you manually edit the main.cpp file to remove commented lines regarding mqtt.
Please give details about your setup.
Type of cable (4/6)
Which firmware you are using (BWC 2.0.0 or BW4W)
How did you connect it to the PCB etc
/visualapproach

mhl66, 2021-05-04:
I´ll do my best to reduce your confusion:

- It´s a **six-wire** model (seems to be the new standard)
- Firmware is **BWC_2.0.0** (mqtt_enabled example in arduino library) from the latest git master
- Connection is via **male (display) and female (pump) six-pin pinhead terminals**, strictly following the instructions here with ESP8266 NodeMCU using a) perfboard according to your schematic, and b) Eric´s PCB, ordered last week, both with resistors, and 8-channel LLCs circuits
- plus: pre2021 line in source code is commented, of course... :-)

visualapproach, 2021-05-04:
Oh, I see . First I'd try without resistors. They have been problems for some people. Replace them with wires (like 0 Ohms resistors) and see if that helps. You can also try the special version I posted for dwoosnan which shows the button codes when pressing on the display buttons. Or both :)
Good luck

mhl66, 2021-05-07:
Sorry, no luck, exactly the same with and without (=with 0 Ohm) resistors.
[image]
Nothing happens at all.

mhl66, 2021-05-07:
Worse yet, your recommended "dwoosnam" library fails to compile (in contrast to the current version):
```
C:\Program Files (x86)\Arduino\libraries\dwoosnam\BWC_8266.cpp: In member function 'void BWC::_startNTP()':
C:\Program Files (x86)\Arduino\libraries\dwoosnam\BWC_8266.cpp:729:33: error: invalid conversion from 'int32_t {aka int}' to 'const char*' [-fpermissive]
   DateTime.setTimeZone(_timezone);
                                 ^
In file included from C:\Users\mh\Documents\Arduino\libraries\ESPDateTime\src/ESPDateTime.h:11:0,

```

visualapproach, 2021-05-07:
> Worse yet, your recommended "dwoosnam" library fails to compile (in contrast to the current version):
>
> ```
>
> C:\Program Files (x86)\Arduino\libraries\dwoosnam\BWC_8266.cpp: In member function 'void [...]

They made a breaking change in the ntp library. Just remove that line and it should be fine. There may be more lines referencing 'settimezone'. Delete them also. And make sure you only have one version of BWC library at a time in the libraries folder.

mhl66, 2021-05-07:
Great!!! Sketch compiled, but could not upload LittleFS data, because "authentication failed" (OTA Password "eps8266" enabled)!?

visualapproach, 2021-05-07:
You may want to enter mqtt credentials and have a broker running. Esp spends a lot of cpu looking for it when not found

mhl66, 2021-05-07:
Just realized that I hat to use the "Basic_MQTT_with_GPIO" version, not the "Basic" version... stay tuned...
I´ll tolerate the MQTT-delay, just want to read out the f... key codes
(and many thanks for your ultra-quick replies)

mhl66, 2021-05-07:
Got it (don´t ask what went wrong in the meantime). Result: Incredible - some of the key codes changed:
NOBTN 1b1b
LOCK 0100
TIMER 0300
BUBBLES 1212
HEAT 1012
PUMP 1312 (NOT 1112)
DOWN 0809 (NOT 1312)
UP 0200
POWER 0000
HYDROJET 1112 (NEW, old=pump)

BUT STILL NO COMMAND SUCCESS AT ALL. Status updates well, but no commands are executed (console shows them, alas) and no key codes are sent! So close... Great @visualapproach - please help!

visualapproach, 2021-05-07:
Cool, I'll add those numbers in next revision. Do you have a logic analyzer (hope I spelled that word right..)?
I take it that when you press buttons on the pump display, nothing happens. And when toggling stuff from web interface nothing happens at the pump nor display?
If you could record data on the CIO port it would be useful.

mhl66, 2021-05-07:
How can I "record data on the CIO port"? Does the ESP need a special log mode? Sorry for the noob question - glad to follow your advice...!?

visualapproach, 2021-05-07:
If you had a logic analyzer you would hook it up to the data, clock and enable lines and record the activity with microsecond resolution. Then you could see if there were any mistakes in your wiring, or if they raised the clock speed beyond the esp capabilities etc.

mhl66, 2021-05-07:
OK, I understand. Sorry, won´t work. Let´s hope that any other Maldives owner can fill the gap of knowledge.
BTW: It´s a great spa, the HydroJet-nozzles produce a powerful massage, which adds greatly to the wellness feeling from the bubbles.

visualapproach, 2021-05-07:
A longshot but you could try setting 160 MHz in the Arduino and reupload.
And I bet it is nice with the jets :) have a nice one

visualapproach, 2021-05-08:
And now that we know the codes you should upload the ordinary program again.

mhl66, 2021-05-09:
...shoud read "**extraordinary** program"... [I´ll do my very best!](https://www.youtube.com/watch?v=szMisIXJzYc)

GSzabados, 2021-05-17:
Hello,

I've just ordered my Maldives this morning. So I will be the next one who will try to set this up on a Maldives 2021 model (hopefully).

I would love to get some help how is the easiest to source the board and pcb elements in Europe? I am in Poland, but I am a foreigner and have no knowledge who sells online these stuff (disregarding Botland and Elty, but I haven't found the lvc over there).

I had no luck finding the Molex connectors neither.

By any chance is there anybody who sells these boards soldered or in a kit?

Regarding @mhl66's problem, do I understand correctly that the module registers the button presses and everything what comes from the DSP, but not the CIO part?

I would think there might be something not connected correctly or the level of the signal is not met what the CIO requires. 

What logic analyser can you advise to get to debug it? (I will might have better luck to get that.)

Best Regards,

Gábor

GSzabados, 2021-05-17:
@visualapproach, could you confirm that these would be the right logic level converters?

https://botland.store/raspberry-pi-gpio-extensions/2468-txb0108-two-way-8-channel-logic-converter-adafruit-395.html

Or this is the 4 channel:

https://botland.store/voltage-converters/2259-logic-level-bi-directional-converter-4-channel-.html

visualapproach, 2021-05-17:
> @visualapproach, could you confirm that these would be the right logic level converters?
> 
> 
> 
> ~~https://botland.store/raspberry-pi-gpio-extensions/2468-txb0108-two-way-8-channel-logic-converter-adafruit-395.html~~
> 
> 
> 
> I see that it is not the right one, maybe this,
> 
> 
> 
> https://allegro.pl/oferta/konwerter-poziomow-logicznych-txs0108e-8-kanalowy-10174091354
> 
> 
> 
> Or this is the 4 channel:
> 
> 
> 
> https://botland.store/voltage-converters/2259-logic-level-bi-directional-converter-4-channel-.html
> 
> 
> 
> 
> 
> 

Hi, it depends on which PCB you choose to order. First one (blue) for my PCB. The other for Naestrom's.

visualapproach, 2021-05-17:
> Hello,
>
>
>
> I've just ordered my Maldives this morning. So I will be the next one who will try to set this up on a Maldives 2021 model (hopefully).
>
>
>
> I would love to get some help how is the easiest [...]

Cheapo Chinese one is good enough. 5V tolerant and they usually have high enough sample rate. (2MS/s)

xmillies, 2021-06-15:
> Cool, I'll add those numbers in next revision. Do you have a logic analyzer (hope I spelled that word right..)?
> I take it that when you press buttons on the pump display, nothing happens. And when [...]

Hi, I have a Maldvives 2021.

Do you know when 'll add 

- SETHYDRO or SETHYDROJETS
- HYDROSTATE or HYDROJETSSTATE 

in lib/BWC/src/BWC_8266_globals.h ?

Or I make a PR ?

visualapproach, 2021-06-15:
Hi!
It's on my todo list. I will work on it shortly. No need for PR :-)

visualapproach, 2021-06-16:
@xmillies you can test the development version now. Hydrojets *should* work.

xmillies, 2021-06-16:
I have made quicks tests.
It seems fine.

I'll check a bit deeper sometimes I see a divergence between SPA Control and Display.
Maybe / probaly issue with wifi. Refresh ...

I'll make more checks if I can this evening to confirm nominal functionalities.

Thank you.

visualapproach, 2021-06-16:
Yes, in particular I want you to check whether hydrojets state is working on the web page. I.e does it become green when you turn it on?

xmillies, 2021-06-16:
Finally except eveything, on little thing.

If you use Hydrojets and then
choose (in SPA Control) : Heater + Pump or Pump alone, the state of the Hydrojets in SPA Control is still on.

But when you choose it in SPA in SC( Spa Control),  Hydrojets goes on in SPA on and in SC.

Is it what you mention (bit 5, 6 or 7). I'm not sure, I think it's an issue on state decoding when pump (and heater) is on.

Pump and hydrojets can't be on together.

visualapproach, 2021-06-19:
Sorry for not answering in a while. Could you try to edit the file
"lib/BWC/BWC_8266_globals.h"
In line 48 change the value 5 to 6. Upload and test. Does the status on web page reflect the pump display now? If not change to 7 and test again.
Thanks

wanoo, 2021-06-28:
Hello, I just brought the maldive 2021, and found this repo. 

I have a logic analyser, how I can help you with this? 

Thanks for this project, you open the field of possibilities, I already imagine to check pH and Ec with some add on from atlas scientific 😊

visualapproach, 2021-06-28:
Hi, yeah there's a couple of things that's interesting to know;
if you can sample clk and data (SPI, Least significant bit first) then check data rate (bits per sec, doesn't have to be precise)
And also post the data in hex with jets on then off, no other changes. We want to know what bit is showing jets status on the display.
Thanks, ask if you need help

wanoo, 2021-06-28:
OK I will tryto get all data in the coming weeks 

Regards

closed 2023-03-13 by visualapproach