visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #75: mqtt connect error

cyberfly79, 2021-06-04:
Hi,
I configured all the data on the mqtt config page but did not connect to the server (local mosquitto in docker).
I use many other tools on mqtt, everything is fine there.

Can I do something wrong?

Thanks!

visualapproach, 2021-06-04:
> Hi,
> 
> I configured all the data on the mqtt config page but did not connect to the server (local mosquitto in docker).
> 
> I use many other tools on mqtt, everything is fine there.
> 
> 
> 
> Can I do something wrong?
> 
> 
> 
> Thanks!
> 
> 
> 
> 

Hi,
Assuming you're using the mqtt version, having the right credentials saved, with the device connected to your WiFi. What's the mqtt error message (code in earlier versions) on the web interface?

cyberfly79, 2021-06-04:
Yes the wifi connection is established.

The error message was "connect failed" (if I remember correctly)
I only use mqtt locally, everything is connected without a username and password. Isn't that a problem??

I overwrote the default settings in the globals.h file. But for some reason I couldn't upload the new code after the re-build.
I uploaded it a few times before.  (This was an error message while uploading: write timeout in the platformio)
(and now unfortunately K.O.)

I'll get a new nodemcu on Monday, then I'll try again.

cyberfly79, 2021-06-07:
It looks like the problem has been solved with a new nodemcu. With the old one, the web ui was pretty slow, it goes nicely.

closed 2021-06-07 by cyberfly79

visualapproach, 2021-06-07:
> It looks like the problem has been solved with a new nodemcu. With the old one, the web ui was pretty slow, it goes nicely.

I'm glad to hear that! But sorry for the poor NodeMCU that retired ;-)