visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #76: 404 Error using VSCode

DJWilz, 2021-06-04:
Hi
Forgive me for being useless with code.
I have downloaded VSCode and setup platformIO, downloaded the code, plugged in and setup a esp8266 with correct driver.
Clicked on the tick bottom right corner of vscode to build then clicked on arrow to upload to esp.
Opened wifi on my mobile and setup the AutoPortal, so far so good I thought.
Used a net analyzer app on phone to see the ip address allocated by esp then typed that into address bar and I get a 404 error.

visualapproach, 2021-06-04:
Hi,
You are forgiven. Just build and upload file system. If using Over The Air upload, first edit platformio.ini. Then use "upload file system". Not "upload file system OTA".

DJWilz, 2021-06-04:
Thanks for reply.
I see the section in platformio.ini referring to OTA.
Do I simply remove the semicolon on the 4 lines, or is there some info I need to edit in.

visualapproach, 2021-06-06:
> Thanks for reply.
> I see the section in platformio.ini referring to OTA.
> Do I simply remove the semicolon on the 4 lines, or is there some info I need to edit in.

Yes remove the semicolons and edit the IP address to whatever the device has.

DJWilz, 2021-06-10:
Still having problems, clearly too technical for me but I want to succeed. 
Trying again, opted to not go with OTA.
So so far logged into Auto Portal and added my WiFi details. 
With ESP powered up now using a portable battery pack, the ESP light is flashing away as if its broadcasting the web interface.
Question is how do I access the web interface ?
Looking at the Microsoft Sway Guide which is based around Arduino it says "The device’s IP is shown
in the serial monitor window."
Im using VSCode so scratching my head as to where the serial monitor is.
Sorry to be a pain.

visualapproach, 2021-06-11:
Serial monitor is one of many ways to find the IP. If you hover mouse over the icons in the bottom blue bar in VS it will show you tool tips. The one that looks like a contact is serial monitor.
Device must be connected via USB to use serial, and after reboot it will state it's IP. Or you can check your router. Or use other tools. Or modify the program to print IP on the pump display :-)

DJWilz, 2021-06-20:
[image]

This is all I was missing, I didn't know I had to do this. Your new write up for VSCode got me sorted, I was just clicking the bottom right arrow in VSCode to upload thinking that's all I needed to do, and then got a 404 error.
Thanks for all your efforts.
I now have a stand alone ESP8266 broadcasting a webpage.
Now to build the circuit and install.
I will be installing into a 6 wire St Moritz hot tub.
I will leave this open until its fitted so I can reply with info on if it works so that you can update your compatibility chart.

rucati, 2021-11-05:
In case others have this issue too:
step 1
curl -fsSL https://raw.githubusercontent.com/platformio/platformio-core/master/scripts/99-platformio-udev.rules | sudo tee /etc/udev/rules.d/99-platformio-udev.rules
step 2
sudo service udev restart
step 3
sudo usermod -a -G dialout $USER && sudo usermod -a -G plugdev $USER
logout to take effect
final steps: upload LittleFS Data as described above in PlatformIO

closed 2023-12-02 by visualapproach