visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #77: ArduinoJson library error

mhl66, 2021-06-06:
Using the new Visual Studio Code + Platform.io tool, the compilation fails:
```
.pio\libdeps\nodemcuv2\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp:43:5: error: static assertion failed: To use 64-bit integers with ArduinoJson, you must set ARDUINOJSON_USE_LONG_LONG to 1. See https://arduinojson.org/v6/api/config/use_long_long/    
   43 |     ARDUINOJSON_ASSERT_INTEGER_TYPE_IS_SUPPORTED(T);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Compiling .pio\build\nodemcuv2\FrameworkArduino\core_esp8266_non32xfer.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\core_esp8266_noniso.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\core_esp8266_phy.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\core_esp8266_postmortem.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\core_esp8266_si2c.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\core_esp8266_sigma_delta.cpp.o
*** [.pio\build\nodemcuv2\lib590\BWC\BWC_8266.cpp.o] Error 1
```

visualapproach, 2021-06-06:
I will look into it. Master or development or both?

mhl66, 2021-06-06:
Just compiled both (Master and Development branches), both the same error in: ".pio\libdeps\nodemcuv2\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp:43:5"
The advice from the development platform to check https://arduinojson.org/v6/api/config/use_long_long/ isn´t helpful at all, since I´ve not the slightest idea where to "set ARDUINOJSON_USE_LONG_LONG to 1".
Your help would be greatly appreciated...

cyberfly79, 2021-06-07:
for me it solved

https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/54#issuecomment-853637840

visualapproach, 2021-06-07:
They made breaking changes to the 8266 SDK. Temporary fix as stated below. I'll investigate it further at the end of the week.

Release Notes
3.0.0 released 6 days ago
New board Lifely Agrumino Lemon v4
Updated Arduino Core to the latest v3.0.0 (release notes)
Updated toolchain packages to the latest v10.2.0
Minor improvements
Breaking changes
The latest Arduino Core v3.0.0 is a major release that brings several breaking changes including removed and significantly changed APIs. Many popular libraries are not synchronized with these changes which may lead to non-obvious compilation and linker errors. Please thoroughly examine the summary of the changes in the release notes before updating the platform. In case your projects are not compilable after the update, you can always roll back to the previous version of the platform, for example:

[env:espresso_lite_v2]
platform = espressif8266@^2
framework = arduino
board = espresso_lite_v2

visualapproach, 2021-06-07:
released v2.1.1 to solve this issue. #closing this

closed 2021-06-07 by visualapproach