visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #79: Hide and block non relevant temperature format

faboaic, 2021-06-07:
Hi,
Currently I am running the latest version happily.
Thanks so much.

One improvement would be to remove the non relevant temperature format (in my case remove fahrenheit).
And block the c/f button on the local control unit.

My wife wanted to turn the pool temperature up... She pressed the c/f button which resulted in a massive peak in my temperature logging curve  ;-)

So an improvement could be to add a setting in the settings page which could have the name "only Celsius mode" or "only fahrenheit mode" which could switch the device to the desired temperature format and prevent changes on the buttons and hide the button in the control Website.

I think no one would ever change the temperature format after the initial setup.

Thanks and best regards,
Fabian

faboaic, 2021-06-07:
No, the water was not suddenly boiling  ;-)
[image]

visualapproach, 2021-06-07:
> No, the water was not suddenly boiling  ;-)
> 
> [image]
> 
> 

Hot wife entering the tub at 16? ;)

faboaic, 2021-06-08:
Found a hardware workaround, should we close the issue?
:-D

Fun aside.

After removing the c/f functionality from the button we could use it as a multi purpose input.
For example send an mqtt command and execute something within node red House Automation.
Notify wife to remove children from the spa, get more cold drinks, ...
Could be very useful

visualapproach, 2021-06-08:
LOL 😸 !
And in combination with the useless timer button we could be specific about when we would like the next drink. Without bubbles unless that button is pressed. Good idea, it's been saved.

visualapproach, 2021-07-19:
@faboaic I implemented this. You can disable buttons on the pump display by editing the code. Also the ESP is sending an MQTT message when pressing a button. Topic = BW_2.0.0/button, with a string like "TIMER" as payload. I can now control things in the house by pressing buttons on the pump 😄 
I didn't bother removing stuff from the web interface. You just have to make a custom copy of the index.html with the elements removed.

closed 2023-03-13 by visualapproach