visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #80: Any tutorial

williams407, 2021-06-11:
Hello !
I'm new here, for first thanks for this wonderful project!
So i have ordered the custom pcb,esp8266 and evertyhing is ready to be flashed but i never use any esp i have absolutly no skill at all in programing, is that possible to having like a tutorial for "noob frendly" cause i trying to read but i'm completly lost.

Thanks!

visualapproach, 2021-06-11:
Hi there! 
Thanks! Have you read the build instructions on the main page? I tried to cover most of the project specifics, but left out general things like how to install platformio etc. There is many good tuts on those things so I didn't bother to write down everything. I mean where should I draw the line? I can't cover everything like how to solder, how to edit files and so on. Is there a specific area I can help you with, or maybe point you in the right direction?

faboaic, 2021-06-13:
> , but left out general things like how to install platformio etc. 

Didn't you add a compiled binary some days ago?
Looks like you already prepared the project for technically not so affected users?

Or did the binary get there by accident?  ;)

I don't think there is a need for a complete noob to compile everything in an IDE , etc.

Just take a binary and flash. Then do some configuration in the web UI, done.

Update could be done via OTA Update from a web storage...

faboaic, 2021-06-13:
This one... 
[image]

visualapproach, 2021-06-13:
Hi @faboaic , and thanks. @engrstephens requested that a while ago, and I played around with it very briefly and realized there is a bit more to it than just uploading the bin file. To make it easy to just flash firmware I need to include not just the sketch but also include the filesystem. And maybe some other bits that I'm unaware of. And I haven't tested it with an uploader/flasher yet. So more work has to be done there. Unfortunately those files got along when I pulled the dev-branch to fix the 8266 Arduino framework breaking updates.

faboaic, 2021-06-13:
@visualapproach we all have to thank **you**

ESP8266 can be easy if you have all the needed binary files, know the offset where to put the "partitions" and have the flashing tool.

Or maybe clean a "golden master" esp8266, flash it with IDE and the fetch a binary directly from its flash.

This could look like this ( @ all: do not use ! This is just an example from other project! )

[image]
[image]

visualapproach, 2021-06-13:
Cool, I'll check it out in the upcoming weeks!

visualapproach, 2021-06-13:
@williams407 I updated the build instructions to show platformio instead of arduino. I hope this is what you wanted. For now, you have to go to the Development branch to find it. Thanks

williams407, 2021-06-15:
Hello,thanks you all for your time!
Sorry for the delay I'm really busy at this time,i will check the instruction again,try and give you some feedback when is done 🙂