visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #83: Issue setting up Helsinki (6 pin)

jondunning, 2021-06-19:
Hi, 

I ordered some PCBs and assembled them, looks awesome, thanks @visualapproach! Pics below!

Sadly im having issues browsing to http://192.168.4.1 after connecting to AutoPortal. Page just doesn't respond. Im using PlatformIO from within VS. Console output below, not sure if connecting........... is normal? Any suggestions?

[image]

[image]

visualapproach, 2021-06-19:
Nice! There is a problem with one of the libraries. Try development branch for now.

jondunning, 2021-06-19:
Awesome that worked, going to connect it all up now. Thanks for the quick response

closed 2021-06-19 by jondunning

visualapproach, 2021-06-19:
No problem. There might be bugs in dev version. I experience a problem here but I will sort it out in a couple of days I hope

jondunning, 2021-06-19:
Understand it’s the dev branch and there will be some issues but can I run these past you just to make sure they are bugs and not a wiring/soldering issue. 

1. Can’t control hot tub from the display panel - this might be desired behaviour but just checking
2. Can’t set temperature. Currently locked at 40
3. Actual and target temperature showing 0

No worries if it’s bugs but just wanted to check it wasn’t an issue my end before screwing everything back together. Thanks

visualapproach, 2021-06-19:
Yeah those are bugs. Same problem here. But I updated master now. I think it will be good

jondunning, 2021-06-19:
Ok I’ll give it a go. Can this be done OTA?

jondunning, 2021-06-19:
Ive done it OTA but the web browser issue has returned

visualapproach, 2021-06-19:
The credentials should survive OTA update. Meaning there is no access point if it finds your WiFi. Did you go back to master branch? That's what I updated

jondunning, 2021-06-20:
So, I think the mistake I made was to set the following as I want to use MQTT back to HomeAssistant;

```
[platformio]
src_dir = src-for-mqtt
data_dir = data-for-mqtt
```

This breaks web server. 

I've changed it back to;

```
[platformio]
src_dir = src-without-mqtt
data_dir = data-without-mqtt
```

and this works! Thanks

I notice that MQTT is also included in the UI on this version, without crawling through the code will MQTT work in this configuration?

Thanks again for your time

visualapproach, 2021-06-20:
MQTT should work either by editing the credentials.h or by entering the MQTT web page. I pulled the dev branch into master this morning and it was a mess, but I think I got everything right in the end. There is no need to choose src_dir and so on anymore. You can see the main changes in the release notes.

jondunning, 2021-06-20:
I actually messed about early this morning and got it working with MQTT. I have re-flashed again just now with your new master branch. Working perfectly! 

Master branch is looking nice and tidy. Again, thanks for your efforts.

visualapproach, 2021-06-20:
Very pleased to hear that. Thank you!