visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #92: St. Lucia (2021) SPA // ESP receives inputs but no control or SPA display

chrstnmr, 2021-07-12:
Hello,

I just received my St. Lucia Spa. It is from 2021 and I cannot get it running. The ESP works fine and the local page is also rendering fine. However, when I connect everything to the SPA:

- Boot-up "Welcome" mode shows "hello"+ IP address.
- also every button is lighting up shortly, however the "upwards" und "downwards" buttons do not light up
- console receives button pushes, but a button push does not fire any action on the pump. Nothing.
- after the initial boot-up welcome screen the display does not show anything
- Power button and pump buttons are shining very very dark (almost not visible)
- I tried it with different resistors and also no resistors at all
- I have an 2x4 PCB
- Soldering seems to be fine
- code has been adjusted for the 2021 SPAs
- tried it with current master and dev branch (the one that was published yesterday)
-
- There must be a problem on the connection from the display/ESP back to the machine
-- webpage does not render any specific state information (temperature is blank)
-- clicking on a button on the webpage does trigger an action on the console however the pump does not react to any button
-- pressing a button on the SPA does trigger a line in the console with the correct button id, however no reaction from the pump

I am currently out of ideas, does anyone else have an idea or had the same issues?

visualapproach, 2021-07-12:
Excellent description, detailed, assuming I know nothing about your setup. Which I don't.
If you
Socketed your llc:s try swapping them to see if one is bad.

chrstnmr, 2021-07-13:
> Excellent description, detailed, assuming I know nothing about your setup. Which I don't.
> If you
> Socketed your llc:s try swapping them to see if one is bad.

Technically you must be correct. The system receives energy from the pump but somehow the other pins do not seem to work. I asked the seller of the JST SM cable for a replacement (as it already had a weird defect at the hinge-thingy where it clicks in). Gonna update this in 1-2 weeks to see if that works.

Or could it be that the cable order is completely different on the 2021 models? Pump model S100101 (220-240V – 50Hz)

visualapproach, 2021-07-13:
S100101 should be pin compatible I think.

chrstnmr, 2021-08-03:
Alright, after a intense weeks of debugging, re-soldering a complete board and a holiday in between I found the reason. Or should I say reason**s**
First of all, I can confirm:

**St. Lucia (2021) is compatible**

If you are unsure what the problem is, dont hard-solder everything to the board and use male pins and female sockets to swap stuff out quickly.
There were 2 problems:

- **- Main Issue:** JST pins were too poorly isolated on the board by myself
- **- Bonus Issue:** one LCC overheated while soldering and broke

**Main Issue**
The main problem was that the pins of my JST were too close even though they did not had any contact to each other. I just tape to isolate each of the pin from input JST and output JST. 
My board had female sockets instead of soldering the JST directly onto the board (mainly for debugging), the male pins were horribly soldered onto the JST cable by myself, however it was good enough to find the mentioned issue above.

**Bonus Issue**
On top of that, one LCC got too hot and simply did not work any longer. I found that I by doing what @visualapproach suggested above. That brought me on the right track at first. Thanks a lot. 

**Solution and Implications**
(1) If you also run into this issue or maybe a similar issue, I recommend to buy some clips with which you can attach male pins to each of the JST cables to the board. Make sure that the JST is able to connect to the board in a very clean manner. 

(2) And second buy a soldering iron that is small enough to handle the size of the board. Otherwise you will maybe _**overheat**_ parts as your iron to solder is not so precise and you will need longer to solder a point. This will bring heat into the whole board and in worst case, as with me, it overheats and parts of them melt (for example semiconductors). (3) As a starter it is maybe helpful to buy a pack of 2-3 LLCs at once (since they cost almost nothing), so if you are unsure if you broke one, you can directly swap it.

**Additional Information**
I used a 2x4 board with two LLCs and placed seven resistors (560Ohm each) from R1 to R7 (I think its the 2x4 schematic linked at the GitHub main page of this project that is labelled as "Eric's PCB").

I am going to tackle understanding MQTT next. Does anyone have a helpful donkey-proved beginner's guide?

visualapproach, 2021-08-03:
Hi @chrstnmr 
Thank you for the great description! I think many people will benefit from reading it and I appreciate your dedication and patience for solving everything. And for reporting back with a happy story! Cheers!

skyluker, 2022-04-03:
> Alright, after intense weeks of debugging, re-soldering a complete board and a holiday in between I found the reason. Or should I say reason**s** First of all, I can confirm:
> 
> ### **St. Lucia (2021) is compatible**

Hey, good to hear. What model did you declare in model.h

visualapproach, 2022-04-07:
> I am going to tackle understanding MQTT next. Does anyone have a helpful donkey-proved beginner's guide?

No but if you dig deep enough you may find the threads were @dev877dev and others taught me about it.