visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #98: Timer vs Actual Temperature - on Web and MQTT

denxhun, 2021-07-19:
I really like this project and happily contribute in it.
I regularly use the timer function of the spa - to be sure that not forget to turn off after hours...
Unfortunately the actual temperature data (on web interface and on MQTT too) sends the remaining hours and the temperature periodically. I understand that this information is coming from the CIO but the ESP could distinguish between these (e.g. based on the last digit/character on the display) and not sending fake temperature to the world.

BTW is there any reason why the letter "H" is not recognised/displayed on web/MQTT during the timing? As I can see in MQTT messages the last character is 0x32 when time showed.

visualapproach, 2021-07-20:
> I really like this project and happily contribute in it.
> I regularly use the timer function of the spa - to be sure that not forget to turn off after hours...
> Unfortunately the actual temperature [...]

Hi and thanks for the feedback!
'H' is not recognized because 
1) It's upper case and I coded lower case. Unknown characters is replaced by 'space' (0x32) 
2) I never bothered about the timer button since this project is making that function redundant. Or so I thought at least. 

Just replace CHARCODES[] for h with 0xED instead of 0xE9 to recognize 'H' and I guess it will show up on the webpage/MQTT.

To mimic the timer you can disable it in the array EnabledButtons[]. It will send an MQTT message about which button is pressed and your home automation can then start a timer and send an MQTT command to shut down the pump. (and toggle some lights and roll up the awnings and so on). Actually, you don't even have to do that, you can let you're home automation turn it off after X hours of filter ON. (Assuming you have some kind of HA since you were looking at MQTT)

denxhun, 2021-07-21:
Thanks for the detailed answer. the "H" recognition and display works well as you described - thanks for that too!
Do you remember where the off-timer and on-timer led is encoded in the cio-dio communication? I'd like to enhance the system with timer handling and you are definitely know better these low level details than anyone else.

closed 2022-03-09 by visualapproach