visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #104: MQTT: "Unable to connect (Lookup error.)." // "CONNECT_FAILED"

chrstnmr, 2021-08-03:
I have been trying to get MQTT working via homebridge.
The homebridge is running on a raspberry pi. It also runs Mosquitto as a MQTT broker. Both services interact and I see a regular pinging in the MQTT messages (of course both services are interacting within a localhost frame).
When trying to integrate the SPA Wifi remote, I ended up clueless. I have been debugging for 10 hours straight now and I cannot find the issue.

The problem is that the Lay Z UI page gives me the feedback "MQTT: CONNECT_FAILED". When trying to connect to the broker via a different MacBook, it says "Unable to connect (Lookup error.)." eventhough the settings in the conf are correct (connections without credentials etc.). I also tried opening the ports on my router but it did not change anything.

Did someone have the same issue? Can someone provide me how their Mosquittos conf.d files look like (the main conf.d and the conf.d in the subfolder called "conf.d")

chrstnmr, 2021-08-04:
Was able to solve it by simply restarting the pump and the ESP by turning off the power at the cable.

visualapproach, 2021-08-20:
@chrstnmr 🥇