visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #109: MQTT Homebridge Integration

PierreBier, 2021-08-13:
Hi, everyone,
I successfully managed to connect the hack to my Lay Z Spa Miami 2021. The resistors were actually necessary.
Everything works, I can control both via the browser and the display.
Now I wanted to connect the whole thing to Hombridge. I already run Mosquitto on the RaspberryPi and installed the Homebridge Mqttthing plugin.
Unfortunately, I don't know much about MQTT yet and I just can't get any further. I created a temperature sensor and entered the path "BW_2.0.0/message" for "Get Current Temperature". Unfortunately it always shows me 0°C. What am I doing wrong?
Does anyone have a very simple example to read out the current temperature?
Regards

visualapproach, 2021-08-13:
Hi, if you have node red in homebridge you can check out my example in the node red folder

PierreBier, 2021-08-14:
unfortunately I don't know anything about NodeRed, so i would prefer it if it somehow worked via a homebridge plugin.

visualapproach, 2021-08-15:
I don't know anything about homebridge so can't help directly unfortunately.

PierreBier, 2021-08-17:
Maybe it will help someone. By trying it out, I managed to integrate it in Hombridge. I use the Homebridge-mqttthing plugin.

```
{
            "type": "temperatureSensor",
            "name": "Pooltemperatur",
            "username": "heizung",
            "password": "12345",
            "logMqtt": true,
            "topics": {
                "getCurrentTemperature": {
                    "topic": "dkspamqtt/message",
                    "apply": "return JSON.parse(message).TMP;"
                }
            },
            "accessory": "mqttthing"
        },
        {
            "type": "switch",
            "name": "Filter",
            "username": "heizung",
            "password": "12345",
            "logMqtt": true,
            "integerValue": true,
            "offValue": "return JSON.stringify({CMD:4,VALUE:0,XTIME:0,INTERVAL:0})",
            "topics": {
                "getOnline": {
                    "topic": "dkspamqtt/message",
                    "apply": "return JSON.parse(message).PWR;"
                },
                "getOn": {
                    "topic": "dkspamqtt/message",
                    "apply": "return JSON.parse(message).FLT;"
                },
                "setOn": {
                    "topic": "dkspamqtt/command",
                    "apply": "return JSON.stringify({CMD:4,VALUE:message,XTIME:0,INTERVAL:0})"
                }
            },
            "accessory": "mqttthing"
        },
        {
            "type": "switch",
            "name": "Blubber",
            "username": "heizung",
            "password": "12345",
            "logMqtt": true,
            "integerValue": true,
            "topics": {
                "getOnline": {
                    "topic": "dkspamqtt/message",
                    "apply": "return JSON.parse(message).PWR;"
                },
                "getOn": {
                    "topic": "dkspamqtt/message",
                    "apply": "return JSON.parse(message).AIR;"
                },
                "setOn": {
                    "topic": "dkspamqtt/command",
                    "apply": "return JSON.stringify({CMD:2,VALUE:message,XTIME:0,INTERVAL:0})"
                }
            },
            "accessory": "mqttthing"
        },
        {
            "type": "thermostat",
            "name": "Heizung",
            "username": "heizung",
            "password": "12345",
            "logMqtt": true,
            "integerValue": true,
            "topics": {
                "getCurrentHeatingCoolingState": {
                    "topic": "dkspamqtt/message",
                    "apply": "return JSON.parse(message).RED == 0 ? '0' : '1';"
                },
                "getCurrentTemperature": {
                    "topic": "dkspamqtt/message",
                    "apply": "return JSON.parse(message).TMP;"
                },
                "getTargetTemperature": {
                    "topic": "dkspamqtt/message",
                    "apply": "return JSON.parse(message).TGT;"
                },
                "setTargetTemperature": {
                    "topic": "dkspamqtt/command",
                    "apply": "return JSON.stringify({CMD:0,VALUE:message,XTIME:0,INTERVAL:0})"
                },
                "setTargetHeatingCoolingState": {
                    "topic": "dkspamqtt/command",
                    "apply": "return JSON.stringify({CMD:3,VALUE:message,XTIME:0,INTERVAL:0})"
                }
            },
            "minTemperature": 20,
            "maxTemperature": 40,
            "heatingCoolingStateValues": [
                "0",
                "1"
            ],
            "restrictHeatingCoolingState": [
                0,
                1
            ],
            "accessory": "mqttthing"
        }
```

closed 2021-08-17 by PierreBier

visualapproach, 2021-08-20:
@PierreBier thank you! I linked to this in the FAQ discussion.

chrstnmr, 2021-08-20:
It's cool to see that we both did our projects almost parallel with the goal to homebridge it.

you used homebridge-mqttthing and I used homebridge-mqtt 
I was not able to do it with mqttthing but you are proving that it works with mqttthing
Good job!

@visualapproach maybe you can add to the FAQ that PierreBier's is for "homebridge-mqttthing" and mine is for "homebridge-mqtt" (since both work a bit differently when config it in homebridge)

Heres my suggestion to copy-paste

> **Homebridge**
> - homebridge-mqttthing: See this example: #109 (Thank you @PierreBier)
> - homebridge-mqtt: and also this nice write-up by @chrstnmr #106

visualapproach, 2021-08-20:
@chrstnmr checked

MartinDdorf, 2022-07-21:
> Maybe it will help someone. By trying it out, I managed to integrate it in Hombridge. I use the Homebridge-mqttthing plugin.
>
> ```
> {
>             "type": "temperatureSensor",
>             "name": [...]

@PierreBier thank you for your solution 😀 I realized it your way but without node red.. and it works! For which purpose is node red necessary?
I own the hydrojet version of the lay-z spa.. could you provide me the code snippet for the hydrojet functionality so that i can also use it? 😀 Would be so kind from you 😀

filippocobelli, 2022-09-15:
@PierreBier thanks for your config file. Thanks to that i’was able to connect the spa to homebridge. I saw the temperature (always at 20 degree)  and the two switches but they do nothing.
What I’m doing wrong? I used homebridge Andes for Mqtt broker in conjunction with Homebridge-mqttthing. The spa show MQTT connected but the temperature reading and the switches doesn’t work 🤨