visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #116: 2021 Ibiza - No Display

Bonser41, 2021-09-13:
HI and thank you for this brilliant project. 

I have built the circuit and installed it in my 2021 Ibiza model. This one has airjets only (no hydrojets) and the pump model number is S100102. I programmed it as a Miami 2021 model which was the closest option I could see. The web interface is working fine and I can update the software OTA. When I turn the unit on, it plays the usual tune and then beeps every now and then. BUT I have absolutely nothing on the screen. No lights or display whatsoever. I've checked all connnections as best I can and everything looks fine. I suspect a faulty voltage converter and have ordered another to test. But in the meantime, does anyone have any suggestions on what I can do to fault find the problem? Many thanks in advance. 

[image]

visualapproach, 2021-09-14:
Thank you @Bonser41 
Firat check hardwares for faults. Then maybe try 560 ohms resistors btw LLC and dsp. Some ppl reported that to solve a flicker problem

Bonser41, 2021-09-14:
Thanks. I did see the thread about inserting resistors but it always seems to be to correct a flickering screen rather than a dead one. I'll try different leads and converter first then add the resistors. Fingers crossed. I did notice that when my device is connected the Web interface reports the correct water temperature but that's about the sum of the communication at this point.

Bonser41, 2021-09-14:
Well I've gone over it again in detail again and tried the following combinations:

- With and without 3 x 560 ohm resistors between LLC and DSP
- With and without the normal 10k resistor on the board
- Repaced the LLC with a brand new one  - same one used by others on this forum https://smile.amazon.co.uk/gp/product/B07DK6QGHG/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1
- Even built a completely new board and fitted new cables just to be absolutely sure.

But still no sign of life on the pump display, which remains completely dead. Not a flash of lights, nothing.
When I turn the pump on, I get the melody and lots of "clicking" sounds for about 30 seconds interspersed with occasional beeps. The web page is then available, with the temperature showing. No other controls on the web interface work. 

My pump is the same model number (S100102) as this one I found on a previous thread: https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/54

So I don't understand why I'm getting no output to the display - my pump is brand new this month, so I guess it may be possible Bestway have modified the display or software in some way. Not sure what else I can do, unfortunately. I'm good with a soldering iron and fine with uploading software, etc, but useless at fault finding electronic issues and don't want to risk bricking my brand new pump ;-(

Any suggestions are most welcome.

Thanks.

visualapproach, 2021-09-15:
I assume the display is working when ESP device is removed?
Strange. Bestway has changed stuff but so far it has only been other codes for the buttons and similar. It should still light up some segments. The clicking sound is a bit suspicious. You could visit the webpage 192.168.your.ip/debug.html
I don't know what we'd expect there, but maybe there is a clue.

[image]

Other than that, I'm not sure. Is your BWC_8266_globals.h file looking like this? If you get it running and the buttons don't do what they are supposed to do, you can try the Maldives and change HASJETS to FALSE further down in the code.

```
...
//uncomment your model and comment out the rest
//#define PRE2021			//the older one, no hydrojets
#define MIAMI2021		//no hydrojets
//#define MALDIVES2021	//hydrojets
```

Bonser41, 2021-09-15:
Thanks for your suggestions. I'll run the debug tomorrow and post the results. The display works perfectly as soon as the ESP is removed and my BEC_8266_globals.h file is exactly as your sample above.

As far as I can see the 5v and GND are passed straight through on your circuit from CIO to DSP so I'd at least expect the display to come to life even if the rest of it has been updated by Bestway. I wonder if they have done something to deliberately cripple this hack?

It would be nice to prove my circuit works with an older model, so if I continue to have problems I'm happy to send it free of charge if someone can fit and prove it. They can keep it as a reward for doing so. At lest we get some feedback and confirmation that Bestway have indeed updated their device. I have spare boards to build another if and when we get to the bottom of it.

Bonser41, 2021-09-16:
Good afternoon. So, I plugged the unit back in and fired it up. I got the usual melody playing but no screen activity. I then videoed the "ticking and beeping" sounds so you can hear what I mean - maybe this gives a clue? You have to turn up the volume quite high to hear it sorry. 

https://user-images.githubusercontent.com/78241508/133652376-a30bb665-7a80-48ec-a0ff-2ab242a70988.mp4

I then tried to access the web page which works:

[image]

Then tried the debug page but it errors - see below:

[image]

I'll try setting is as a Maldeves without jets, but I would assume this is very unlikely to have any effect. 

Thanks again.

visualapproach, 2021-09-16:
Debug is no error. It is an instruction.

visualapproach, 2021-09-17:
Sorry for beeing short, I was in a hurry. In the debug page you have to show the developer's console. There you will see a flow of messages which you could post here. 
Did you try running the ESP @160 MHz as suggested in #54 ?

SRowley90, 2021-09-17:
I have the same thing happening with me but without any of the beeps happening on startup. Same device model (S100102), tried the change to put it to 160 MHz with no change. I also get the odd clicking happening like a relay is being switched. I've not had chance to look at the console while attached to the pump yet.

Bonser41, 2021-09-20:
> Sorry for beeing short, I was in a hurry. In the debug page you have to show the developer's console. There you will see a flow of messages which you could post here.
> Did you try running the ESP @160 MHz as suggested in #54 ?

Sorry, I missed your reply. I'll check the console later today and report back. Thanks.

Bonser41, 2021-09-20:
> I have the same thing happening with me but without any of the beeps happening on startup. Same device model (S100102), tried the change to put it to 160 MHz with no change. I also get the odd [...]

OK that's interesting to know. Is your pump new in the last few months like mine?

SRowley90, 2021-09-20:
> OK that's interesting to know. Is your pump new in the last few months like mine?

Yes, start of summer this year. Got the Fiji model in the UK. Would hazard a guess we have the same pump.

Bonser41, 2021-09-20:
OK here is the debug info in the hope it helps with some clues?

debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632154932,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":574,"PUMPTIME":4294968,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":512,"COST":0,"FINT":4294967295,"CLINT":4294967295}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"f8ff","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632154934,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":576,"PUMPTIME":4294968,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":514,"COST":0,"FINT":4294967295,"CLINT":4294967295}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"fcff","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632154936,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":578,"PUMPTIME":4294968,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":516,"COST":0,"FINT":4294967295,"CLINT":4294967295}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"f8ff","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632154938,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":580,"PUMPTIME":4294968,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":518,"COST":0,"FINT":4294967295,"CLINT":4294967295}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"feff","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632154940,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":582,"PUMPTIME":4294968,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":520,"COST":0,"FINT":4294967295,"CLINT":4294967295}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"feff","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632154942,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":584,"PUMPTIME":4294968,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":522,"COST":0,"FINT":4294967295,"CLINT":4294967295}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"feff","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632154944,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":586,"PUMPTIME":4294968,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":524,"COST":0,"FINT":4294967295,"CLINT":4294967295}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"f7ff","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632154946,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":588,"PUMPTIME":4294968,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":526,"COST":0,"FINT":4294967295,"CLINT":4294967295}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"00e0","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632154947,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":589,"PUMPTIME":4294969,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":527,"COST":0,"FINT":4294967295,"CLINT":4294967295}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"ffff","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632154948,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":590,"PUMPTIME":4294969,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":527,"COST":0,"FINT":4294967295,"CLINT":4294967295}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"ffff","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632154948,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":590,"PUMPTIME":4294969,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":528,"COST":0,"FINT":4294967295,"CLINT":4294967295}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"ffff","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632154950,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":592,"PUMPTIME":4294969,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":530,"COST":0,"FINT":4294967295,"CLINT":4294967295}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"00ff","HASJETS":0}
[... cut ...]
​ {"CONTENT":"TIMES","TIME":1632155014,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":656,"PUMPTIME":4294969,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":594,"COST":0,"FINT":4294967295,"CLINT":4294967295}
​ {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"80ff","HASJETS":0}
​ {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
​ {"CONTENT":"TIMES","TIME":1632155016,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":658,"PUMPTIME":4294969,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":596,"COST":0,"FINT":4294967295,"CLINT":4294967295}

Bonser41, 2021-09-20:
Also tried 160MHz setting and tried setting it to Maldives model and disabling the hydrojets. No luck I'm afraid. The screen stays completely off ;-(

visualapproach, 2021-09-20:
Okay, did you press any buttons on the pump display whilst recording the debug sequence?

SRowley90, 2021-09-20:
I noticed the button press events in the log too, I didn’t have it connected up to the pump and still got them.

visualapproach, 2021-09-20:
@SRowley90 If the dsp is not connected there will be random bits on the input. With the dsp connected the button code should be 1B1B when no buttons are pressed. I wonder if those lines needs pull-up/down resistors. Just thinking out loud. I guess it would be harmless to connect the three data lines to either GND or 5V through a 10KOhms resistor (per line).
LLC----DSP
    |10K
   Gnd or 5V

?

Bonser41, 2021-09-20:
> Okay, did you press any buttons on the pump display whilst recording the debug sequence?

No, I didn't touch anything whilst recording the sequence.

visualapproach, 2021-09-20:
> > Okay, did you press any buttons on the pump display whilst recording the debug sequence?
> 
> 
> 
> No, I didn't touch anything whilst recording the sequence. 

@Bonser41 Okay see my thoughts above.

SRowley90, 2021-09-20:
I might need a little bit of hand holding, which are the data lines?

Bonser41, 2021-09-21:
OK, I tried 5v pull-up 10K resistors on the three data lines to the DSP and posted the log below: I'll try pull down when I get a moment. Not sure if this gives us any more clues ?

debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632251508,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":589,"PUMPTIME":4294968,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":528,"COST":0,"FINT":4294967295,"CLINT":4294967295}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"feff","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632251508,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":589,"PUMPTIME":4294968,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":528,"COST":0,"FINT":4294967295,"CLINT":4294967295}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"ffff","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632251510,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":591,"PUMPTIME":4294968,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":530,"COST":0,"FINT":4294967295,"CLINT":4294967295}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"feff","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632251512,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":593,"PUMPTIME":4294968,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":532,"COST":0,"FINT":4294967295,"CLINT":4294967295}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"feff","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632251514,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":595,"PUMPTIME":4294968,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":534,"COST":0,"FINT":4294967295,"CLINT":4294967295}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"ffff","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632251516,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":597,"PUMPTIME":4294968,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":536,"COST":0,"FINT":4294967295,"CLINT":4294967295}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"ffff","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632251518,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":599,"PUMPTIME":4294968,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":538,"COST":0,"FINT":4294967295,"CLINT":4294967295}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"ffff","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632251520,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":601,"PUMPTIME":4294968,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":540,"COST":0,"FINT":4294967295,"CLINT":4294967295}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"ffff","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632251522,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":603,"PUMPTIME":4294968,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":542,"COST":0,"FINT":4294967295,"CLINT":4294967295}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"ffff","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632251524,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":605,"PUMPTIME":4294968,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":544,"COST":0,"FINT":4294967295,"CLINT":4294967295}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"feff","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632251526,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":607,"PUMPTIME":4294968,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":546,"COST":0,"FINT":4294967295,"CLINT":4294967295}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"ffff","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632251528,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":609,"PUMPTIME":4294968,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":548,"COST":0,"FINT":4294967295,"CLINT":4294967295}
[... cut ...]
debug.html:20 {"CONTENT":"TIMES","TIME":1632251552,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":633,"PUMPTIME":4294968,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":572,"COST":0,"FINT":4294967295,"CLINT":4294967295}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"fcff","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":36,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1632251554,"CLTIME":4294967295,"FTIME":4294967295,"UPTIME":635,"PUMPTIME":4294968,"HEATINGTIME":4294966,"AIRTIME":4294967,"JETTIME":574,"COST":0,"FINT":4294967295,"CLINT":4294967295}

visualapproach, 2021-09-21:
Ok thanks for trying. When I get some time I'll see if I can arrange something to find the problem.

Bonser41, 2021-09-21:
OK thank you for your continued assistance. I love the project and really want to do what I can to help solve the mystery. I'm happy to post you my complete board so you can double-check it is all correct ?- You don't need to send it back - I have another. Otherwise, I'll sit tight and monitor the forum until someone more technical than me can ID the issue. Thanks again.

visualapproach, 2021-09-24:
> OK thank you for your continued assistance. I love the project and really want to do what I can to help solve the mystery. I'm happy to post you my complete board so you can double-check it is all [...]

@Bonser41 yes, but you can hold on to your device a while. I think we can test something by software first, but I need some free time to tinker with it...

AlbertDijksma, 2021-10-11:
Hi, Thank you for this fantastic project and the work you put into it.
I bought a spa at the Action https://shop.action.com/nl-nl/p/8711252249490/dreamstream-oplaagbaar- Jacuzzi
This includes a model 2021 pump.
Built your project as a 6 wire version with the 10k resistors between the ESP and LLC.
The software that I have loaded into it is the development brance. And set as MIAMI2021.
Unfortunately the display doesn't work for me either.

[image]
[image]

[image]

[image]

[image]

Debuglog:
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":37,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1633954344,"CLTIME":1633941876,"FTIME":1633701078,"UPTIME":561,"PUMPTIME":6,"HEATINGTIME":0,"AIRTIME":1,"JETTIME":171,"COST":0.006468,"FINT":10,"CLINT":2}
debug.html:20 {"CONTENT":"OTHER","MQTT":0,"PressedButton":"abeb","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":37,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1633954345,"CLTIME":1633941876,"FTIME":1633701078,"UPTIME":562,"PUMPTIME":6,"HEATINGTIME":0,"AIRTIME":1,"JETTIME":172,"COST":0.006505,"FINT":10,"CLINT":2}
debug.html:20 {"CONTENT":"OTHER","MQTT":0,"PressedButton":"ff67","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":37,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1633954347,"CLTIME":1633941876,"FTIME":1633701078,"UPTIME":564,"PUMPTIME":6,"HEATINGTIME":0,"AIRTIME":1,"JETTIME":174,"COST":0.006579,"FINT":10,"CLINT":2}
debug.html:20 {"CONTENT":"OTHER","MQTT":0,"PressedButton":"ff6b","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":37,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1633954349,"CLTIME":1633941876,"FTIME":1633701078,"UPTIME":566,"PUMPTIME":6,"HEATINGTIME":0,"AIRTIME":1,"JETTIME":176,"COST":0.006652,"FINT":10,"CLINT":2}
debug.html:20 {"CONTENT":"OTHER","MQTT":0,"PressedButton":"fadf","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":37,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1633954351,"CLTIME":1633941876,"FTIME":1633701078,"UPTIME":568,"PUMPTIME":6,"HEATINGTIME":0,"AIRTIME":1,"JETTIME":178,"COST":0.006726,"FINT":10,"CLINT":2}
debug.html:20 {"CONTENT":"OTHER","MQTT":0,"PressedButton":"9f7e","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":37,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1633954353,"CLTIME":1633941876,"FTIME":1633701078,"UPTIME":570,"PUMPTIME":6,"HEATINGTIME":0,"AIRTIME":1,"JETTIME":180,"COST":0.0068,"FINT":10,"CLINT":2}
debug.html:20 {"CONTENT":"OTHER","MQTT":0,"PressedButton":"adab","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":37,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1633954355,"CLTIME":1633941876,"FTIME":1633701078,"UPTIME":572,"PUMPTIME":6,"HEATINGTIME":0,"AIRTIME":1,"JETTIME":182,"COST":0.006874,"FINT":10,"CLINT":2}
debug.html:20 {"CONTENT":"OTHER","MQTT":0,"PressedButton":"bfee","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":37,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1633954357,"CLTIME":1633941876,"FTIME":1633701078,"UPTIME":574,"PUMPTIME":6,"HEATINGTIME":0,"AIRTIME":1,"JETTIME":184,"COST":0.006947,"FINT":10,"CLINT":2}
debug.html:20 {"CONTENT":"OTHER","MQTT":0,"PressedButton":"e68f","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":37,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1633954359,"CLTIME":1633941876,"FTIME":1633701078,"UPTIME":576,"PUMPTIME":6,"HEATINGTIME":0,"AIRTIME":1,"JETTIME":186,"COST":0.007021,"FINT":10,"CLINT":2}
debug.html:20 {"CONTENT":"OTHER","MQTT":0,"PressedButton":"eb0f","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":37,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1633954361,"CLTIME":1633941876,"FTIME":1633701078,"UPTIME":578,"PUMPTIME":6,"HEATINGTIME":0,"AIRTIME":1,"JETTIME":188,"COST":0.007095,"FINT":10,"CLINT":2}
debug.html:20 {"CONTENT":"OTHER","MQTT":0,"PressedButton":"ffdb","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":37,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1633954363,"CLTIME":1633941876,"FTIME":1633701078,"UPTIME":580,"PUMPTIME":6,"HEATINGTIME":0,"AIRTIME":1,"JETTIME":190,"COST":0.007168,"FINT":10,"CLINT":2}
debug.html:20 {"CONTENT":"OTHER","MQTT":0,"PressedButton":"be47","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":37,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1633954365,"CLTIME":1633941876,"FTIME":1633701078,"UPTIME":582,"PUMPTIME":6,"HEATINGTIME":0,"AIRTIME":1,"JETTIME":192,"COST":0.007242,"FINT":10,"CLINT":2}
debug.html:20 {"CONTENT":"OTHER","MQTT":0,"PressedButton":"ef9f","HASJETS":0}
[... cut ...]
debug.html:20 {"CONTENT":"TIMES","TIME":1633954403,"CLTIME":1633941876,"FTIME":1633701078,"UPTIME":620,"PUMPTIME":6,"HEATINGTIME":0,"AIRTIME":1,"JETTIME":230,"COST":0.008642,"FINT":10,"CLINT":2}
debug.html:20 {"CONTENT":"OTHER","MQTT":0,"PressedButton":"dbf9","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":37,"TMP":36,"CH1":32,"CH2":51,"CH3":54,"HJT":1}
debug.html:20 {"CONTENT":"TIMES","TIME":1633954405,"CLTIME":1633941876,"FTIME":1633701078,"UPTIME":622,"PUMPTIME":6,"HEATINGTIME":0,"AIRTIME":1,"JETTIME":232,"COST":0.008716,"FINT":10,"CLINT":2}
debug.html:20 {"CONTENT":"OTHER","MQTT":0,"PressedButton":"0310","HASJETS":0}

AlbertDijksma, 2021-10-11:
Hi,
I have removed the 3 resistors between lcc and the esp for the display.
I can now operate the keys and read the temp on the display.
The display and buttons still flash, but a lot less.
Tomorrow I will put resistors between the lcc and the display module.
I will also open up a display to find out the components.

eried, 2022-04-15:
> Good afternoon. So, I plugged the unit back in and fired it up. I got the usual melody playing but no screen activity. I then videoed the "ticking and beeping" sounds so you can hear what I mean - maybe this gives a clue? You have to turn up the volume quite high to hear it sorry.
> 
>  20210916.171204.1-1.mp4 
> I then tried to access the web page which works:
> 
> [image]
> 
> Then tried the debug page but it errors - see below:
> 
> [image]
> 
> I'll try setting is as a Maldeves without jets, but I would assume this is very unlikely to have any effect.
> 
> Thanks again.

@Bonser41 Oh, I have this same sound... I see the diagram of your solution, but I do not understand if you actually added

eried, 2022-04-15:
@Bonser41 @SRowley90 did you find a solution?

Bonser41, 2022-04-15:
No, never did, unfortunately.

eried, 2022-04-15:
:/ oh, did you test adding more resistors to some of the lines? can you share a photo of your board? I really want to make it work on mine and I get exactly the same sound.

segapus, 2022-07-02:
I see this post is now quite old but maybe I can shine light in someway. I too have this weird sound, model S100101 (Helsinki 2021) bought June 2022 in Sweden. My finding this morning when first trying to connect the module is that the sound that comes from the display definitely is in sync with the small LED that flashes showing wifi connectivity (the LED to the right of the antenna on the esp8266).
Hope this can help in some way. For me the web interface works and the controls as well, just the display and physical buttons that is not working.

wouter-jacobs, 2022-07-02:
good afternoon, I also bought a S100101 (Helsinki 2021) in May 2022. exactly the same problem. The web interface works and so do the control buttons. The display on the pump and the buttons do nothing. Put the new release on the ESP yesterday. (29-06 dev version). This didn't solve anything. There is a problem with both the stable version of 24-05 and the development version of 29-06. Anyone tips?

visualapproach, 2022-07-02:
Since other people reports success on that model I'd say it's not software related.

closed 2023-03-13 by visualapproach