visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #117: Spa miami 4 wire wird control

Klooon, 2021-09-16:
Hello from France
Thanks a lot for all that work , verry interesting update 🙂 . 
I have miami laz-y  with 4 wire display . 
My wiring is 
5v  ( cable 1 from display ) 
Grd ( cable 4 from display ) 
Rx  (  cable 2 from display)  
Tx   (cable 3 from display ) 
I have access to the 8266 . 
The temp writte on the display is the same on my phone . Doesn't  that mean  my wiring is Good ? 
  When I select take control .  If I click  heater that start bubbles ... 
When I click on pump I hear a relay triger ( I click again fastly to not burn something :)  )  
When I click on jets there is also a relay triger  ( there is no jets on my spa ).
If I click bubbles I don't hear anything .
There is no watter in my spa yet  ( maybe with watter I can know what the relay triger 🙂 ) . The most important for my is bypass the time limite . I'm not sur that can work like that if bouton not match ...  .  
Maybe I have done someting  wrong or I have wierd modèle,  if sombody have clue what is going on , I apeciate to know :) . 

Level : beginer , I don't know what I'm doing 😁 . 

[image]

appletrapz, 2021-09-16:
it sounds like you have the 54123 version. try changing model.h to this 
> //#define NO54138 //this is the only option for now. Maybe there will be other pumps supported in the future.
//WARNING: THIS DEVICE HAS DIFFERENT PINOUTS!!! CHECK BEFORE USING
//NOT TESTED BY ME!!
#define NO54123 //not tested. ref https://github.com/mrQ000/layz-rc link to presentation slides

Klooon, 2021-09-16:
Thanks for  that clue  :)  . 
But I really don't know how To use it 😁  . How To change model.h   .  I have to edit the files BWC_8266_4w_globals.h ???

appletrapz, 2021-09-16:
So inside the **model.h** file change the code to what i put above and then reupload the files to the ESP8266

Klooon, 2021-09-16:
Thanks , then I have done the change , and now nothing :)  , display powering   but touch pad don't work  and I don't have any sign of link on my phone,  temp is 00 ... no bouton work on it .   With that change my wiring can be wrong?  I mean swap data wire 3 and 4 ?  I don't know how that  work  ... technically  if I succeed  read temp  and trigger wiring like that , that mean it's ok ?  Thanks    .

appletrapz, 2021-09-16:
This is my Miami Model #54123 4 wire setup 
if you have a 4 wire white cable like mine each wire will have a pattern on them and I soldered them to the PCB in this order

From CIO:
5V 
GND (-- -- -- --)
CIO RX (x x x x) to D3
CIO TX (—— ——) to D2

From DSP:
5V 
GND (-- -- -- --)
DSP TX (x x x x) to D6
DSP RX (—— ——) to D7

appletrapz, 2021-09-16:
This is my Miami Model #54123 4 wire set up 
if you have a white cable like mine each wire will have a pattern on them and i soldered them to the PCB in this order

From CIO:
5V 
GND (-- -- -- --)
CIO RX (x x x x) to D3
CIO TX (—— ——) to D2

From DSP:
5V 
GND (-- -- -- --)
DSP TX (x x x x) to D6
DSP RX (—— ——) to D7

visualapproach, 2021-09-17:
Looks like you had it more right the first time, except the buttons are not defined correctly for this pump. Can you find the model number on the pump? Like 54123 or S100102

Klooon, 2021-09-17:
Your Wright It's a 54123 
With my previous wiring I have now the temp   I can't triger with my phone even If I take control  , the spa  work with display touch , and i see bouton moving on the phone when I click on display , but the bubble  and pump touch display triger   the same  ( the bouton  heater ) on the phone ....  
[image]

Klooon, 2021-09-17:
I restart with the new files I download , make the change model .h files ,  upload it , and the bouton   are ok    pump is pump , bubble is bubble  ...   I my rookie level probably add problem where there is no problem  . 
If I change  temp expected  on spa , that don't change in phone , and reverse . Is that normale  ? ( to be sure )  .  
If i selected take control  .   I set the  temp i want  on phone   and  the value temp set in the display isn't important  until the 8266 don't have control  ? Thanks for the help  .

visualapproach, 2021-09-17:
Congrats! And you are absolutely right; when you take control the display is out of the loop. It's ESP or DSP. Not both. Cheers

Klooon, 2021-11-04:
I finaly found time to fill it with water and test it  . The  temp is not attempt for now 🤞 but that looks OK  👍 .   I have a question I don't know if I have to open an other subject   . Is there a way to add a password , even basic  security  even if that the same for evrybody who have  your board / firmware install In is spa ?   :) thanks in adavance

visualapproach, 2021-11-05:
I guess it is possible, but:
I'm guessing that you want to control the device from outside your home network, and wants to open ports to the world? I highly recommend using a VPN instead, to connect to your home network. Like Wireguard.
Or do you share your home network with people that shouldn't be able to control the Spa?

Klooon, 2021-11-07:
I don't need acces form internet.  That for my home network ,  wifi key hack look easy ,   everything on my home network have pasword , I found that safer  .

visualapproach, 2021-11-08:
Okay I see your point. Added to my todo-list.

visualapproach, 2021-11-09:
@Klooon Try the development branch. user: admin password: esp8266

closed 2022-03-09 by visualapproach