visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #118: SaluSpa Honolulu - Model S100104

simracingstudio, 2021-09-18:
Hello guys,

Just finished to solder pcb and uploaded the code.
[image]

I cannot control the spa via web interface, neither display.

If I have the PCB connected when I plug the unit to the wall the Display does not beeps.
It shows red on the C and that's all, none of the buttons work.

I have tried Dev and Stable branches, also tried to use the 160 Frequency and no luck.

I think the issue is the communication of the ESP32 with the mainboard, because I commands come 0 in the debug.html
[image]

This is the display
[image]

Model
[image]

I believe the next step is to try the resistor, but I'm not sure where to solder and each one. If someone can provide a light, I will really appreciate.

visualapproach, 2021-09-18:
Forgot to solder the LLC pins?

simracingstudio, 2021-09-18:
You are right I was going to turn off everything and spotted that.

Any concern if I use the dev branch?

Don’t close it yet let me do more tests, just finished to solder

visualapproach, 2021-09-18:
Dev branch should be stable also at the moment but with a few updates.
Good luck

simracingstudio, 2021-09-18:
It works!!! The only thing missing are the beeps.

visualapproach, 2021-09-18:
Great 👍 
Some newer models appear to have lower volume when using this device. I don't know why that is. Probably a different circuit. Can't do anything until someone figure out what the problem is. (I have an older pump)

closed 2021-09-18 by simracingstudio