visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #119: Mains control

gabrielke, 2021-09-22:
As I understand this project it implies the spa is basically turned on (so power is green).
This doesn't change the behavior after a power outage (whole thing remains turned off as factory default, right?)

Or it can "press" the power button too somehow?

gabrielke, 2021-09-22:
changed to a question in discussions instead

closed 2021-09-22 by gabrielke

visualapproach, 2021-09-22:
@gabrielke This only applies to the 6 pin version. 
As far as I can tell, the pump is always "on", even when you press POWER OFF on the original pump. It only dims the display and nothing else is running. Mains is never turned off unless you pull the plug or hit the protection breaker. The device can turn OFF or ON the display as you would when pressing the button yourself. What happens after power outage is more or less configurable from the web interface - restore states or everything is off.