visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #124: Current draw - Maldives 54173 (2019) - 4 pin

Lukey1979, 2021-09-25:
Hey @visualapproach,

Want to get some clarification around some stuff and also share a somewhat success story.

We have. managed to get this solution working on the model Maldives 54173 (2019) - 4 pin
Bubbles, pump, take control, jets all seem to be working from the interface and MQTT (which is so epic thank you)

MQTT turning on and off cmd 11 (jets) doesn't reflect on the web interface, however does operate the physical jets. (not something I care about just sharing a possible bug) every other option is reflected accurately during MQTT command

The web interface also has the area with the red square brackets, not sure what we are meant to be seeing there but since we are using a model that is 4 pin and we need to turn on "Take Control" maybe we aren't reading the data here like you would see on other models. (also not something I care about, just sharing)

My question really is in the way the spa is heating whilst in take control mode, the target temp is at 40 degrees however the actual temp is at 37, I have a smart power monitoring plug on the spa and have been monitoring the spa power usage long before this project and now looking at it during the project I can see the power consumption during its heating cycle is significantly less than it would have been before installing the controller. It's at about half its power usage, I'm also seeing it take a really long time to get the temperature up a single degree. Reading your other comments I read that the set temperature is handed to the spa controller by means of temp up and down buttons, and lets the spa regulate the heat as normal. I am unsure if that is the case during "take over" mode, and if so, what is it doing?

Another quick share, I had the spa running overnight and compared its power usage, I will share the graph here but notice before the controller install the spa would spike up high and then drop off whilst the controller sees the heater just stay at a lower power usage. The daily power consumption resets every night at midnight.

[image]

[image]

Lukey1979, 2021-09-25:
We have just done another experiment, we have turned off "Take Control" and set the target temp to 40 on the spa panel (actual temp 37), we are now seeing an increase in Live Consumption (w) up to 1400w, however still short of the regular 1640w before the controller was installed.

So for some reason the heater isn't able to operate at its full potential whilst the added controller is installed.

Lukey1979, 2021-09-25:
Edit: It seems to have boosted to a higher power consumption nearly exactly 1 hour after turning off "Take Control"

[image]

visualapproach, 2021-09-25:
Hi @Lukey1979 

> We have. managed to get this solution working on the model Maldives 54173 (2019) - 4 pin
> Bubbles, pump, take control, jets all seem to be working from the interface and MQTT (which is so epic thank you)

Very good. No communication errors? Did you do anything special or did it work as is? What PCB version do you use?

> 
> MQTT turning on and off cmd 11 (jets) doesn't reflect on the web interface, however does operate the physical jets. (not something I care about just sharing a possible bug) every other option is reflected accurately during MQTT command
> 

Noted thank you.

> The web interface also has the area with the red square brackets, not sure what we are meant to be seeing there but since we are using a model that is 4 pin and we need to turn on "Take Control" [...]

It's a legacy from the 6 wire pumps, but I left it there for displaying errors. If it doesn't show anything you're good.

> My question really is in the way the spa is heating whilst in take control mode, the target temp is at 40 degrees however the actual temp is at 37, I have a smart power monitoring plug on the spa [...]

As I was developing this on the bench only (the floor rather), I probably missed the "stage 2" bit. We'll see what we can do.
Or it's a different pump design.

Lukey1979, 2021-09-25:
> Very good. No communication errors?

Does E13 count as communication errors? lol, we had the wifi reboot and that caused it once, but it also just seems to happen randomly. This does not effect web interface control, only spa panel control... and the beeping is only slightly annoying haha. power spa off and back on is the only fix? Does an overload of web interface reactions effect this, or having MQTT messages pushed or pulled too frequently?

> Did you do anything special or did it work as is? 

Special?? Um.. fluffed around with attempts to flash the board, windows issues. multimeter on the wires to ensure +5v and ground were what the guide said they could be... (yes black and red are swapped on this model here in Australia also)

> What PCB version do you use?

We built both your version and Eric's version and have supplies to spare. (we get excited about projects like this).

> It's a legacy from the 6 wire pumps, but I left it there for displaying errors. If it doesn't show anything you're good.

should we be seeing e13 here? or an answer as to what causes it? Next to it, is "MQTT" and ours shows 0 even though we have something subscribed to its messages

> As I was developing this on the bench only (the floor rather), I probably missed the "stage 2" bit. We'll see what we can do.
> Or it's a different pump design.

Awesome, let us know if we can help at all. Might not be as efficient as yourself but willing to have a crack at it

Lukey1979, 2021-09-25:
Curious to know if this is expected behaviour also. Maybe there isn't a temperature tolerance like I'm thinking.

https://user-images.githubusercontent.com/63522396/134764666-87d7d9a6-0372-4edc-876c-8b48945e9c31.mov

visualapproach, 2021-09-25:
Haha I thought I had prevented that. Apparently there is more to check...

Lukey1979, 2021-09-26:
Hi @visualapproach Sorry to bombard you with requests / issues. Just wondering how I would go about changing the code to allow the Bubbles to operate without turning off the pump and heater? With this model spa, the bubbles CAN run at the same time as the heater and pump whereas the Jets do not allow this (Jets turning off the Heater / Pump is correct)

visualapproach, 2021-09-26:
If there is stupid questions This isn't one of them. So no worries! The logic is in the library which is found under libs folder. If I remember correctly-  I'm on my phone now but will check it later

visualapproach, 2021-09-26:
What model is defined in your "model.h" file? 
#define NO54138 or NO54123?

visualapproach, 2021-09-26:
@Lukey1979 (Assuming NO54138) Try development branch:
 JET-button on webpage should now update
Heating should not flipflop as in your movie, but - repeat [heat to target+1 then cool to target-1]

To activate all your heating elements, you could try editing line 69 in "BWC_8266_4w_globals.h" end setting HEATBITMASK = 112.
That will probably not work because it's just a guess that setting that unused bit will magically turn on the heater :-)

To keep filter pump and heating on together with bubbles you can comment out lines 229-230 in "BWC_8266_4w.cpp" like this:

```
case SETBUBBLES:
	_cio.states[BUBBLESSTATE] = _commandQ[0][1];
//        _cio.states[PUMPSTATE] = false;
//        _cio.states[HEATSTATE] = false;
break;
```

I can't test 4 wire program since I don't have that pump, but it compiles alright.

visualapproach, 2021-09-26:
remember to upload the LittleFS files also. Or just the WebSockets.js file if you don't want to erase the saved settings.
To upload only one file you visit device-IP/upload.html

visualapproach, 2021-09-26:
> Does E13 count as communication errors? lol, we had the wifi reboot and that caused it once, but it also just seems to happen randomly. This does not effect web interface control, only spa panel [...]

Yes E13 is com err. Don't know what causes it unfortunately.

> should we be seeing e13 here? or an answer as to what causes it? 

Yes, the error number should be displayed here. Not the cause though.

> Next to it, is "MQTT" and ours shows 0 even though we have something subscribed to its messages

      MQTTclient.state return code meanings...
      -4 : MQTT_CONNECTION_TIMEOUT - the server didn't respond within the keepalive time
      -3 : MQTT_CONNECTION_LOST - the network connection was broken
      -2 : MQTT_CONNECT_FAILED - the network connection failed
      -1 : MQTT_DISCONNECTED - the client is disconnected cleanly
      0 : MQTT_CONNECTED - the client is connected
      1 : MQTT_CONNECT_BAD_PROTOCOL - the server doesn't support the requested version of MQTT
      2 : MQTT_CONNECT_BAD_CLIENT_ID - the server rejected the client identifier
      3 : MQTT_CONNECT_UNAVAILABLE - the server was unable to accept the connection
      4 : MQTT_CONNECT_BAD_CREDENTIALS - the username/password were rejected
      5 : MQTT_CONNECT_UNAUTHORIZED - the client was not authorized to connect *

> Awesome, let us know if we can help at all. Might not be as efficient as yourself but willing to have a crack at it

Thx

Lukey1979, 2021-09-27:
Hi @visualapproach Thank you for all of your help. For some reason, using the Dev Version, both versions of my PCB's will not connect to wifi when connected to the spa but will connect to wifi if I have them at my computer and plugged into USB. This doesn't make any sense. Has anything changed with the Dev version that could affect the device connection?

I haven't had a chance yet but I will put the 'Master' version back onto the devices to see if I am still having issues. I just find it strange that both devices were connecting to wifi fine when connected to the spa and now they are not. 

Unfortunately Ive been unable to test your changes in the Dev version. I will give it a go again tomorrow and let you know how I go. 

On a side note: The changes you made regarding the JET button on the web page updating and the Heating Flip Flop, where are they located so I can possible modify my downloaded copy of your Master. (This is to eliminate any issues from the Dev version causing my WiFi connection issues)

visualapproach, 2021-09-28:
Hi again, nice to hear that my changes works! I was thinking about the power supply as a potential problem also. Maybe a capacitor will help remedy short spikes  but if the average power consumption is too high it won't help.
The label on your pump says ~1800W with blower+heater which is lower than just heater (~2000W). With the new fix all heater elements will perhaps be on, so I think I want to change that.

Lukey1979, 2021-09-28:
hi @visualapproach So to give you some figures for the heating elements. 
Prior to adding the Spa Wifi Card my power consumption with heater on was approx 1650W
After adding the wifi card, this reduced to anywhere between 800W & 1000W

Changing the `const uint8_t HEATBITMASK = 48;` to const uint8_t HEATBITMASK = 112; has increased my power consumption to anywhere between 1650W and 1950W (Clearly has reinstated the second heating element assuming there is one)

Operating the bubbles at the same time as the heating element increases my output from approximately 1650W-1900W to approximately 2340W (I would assume this is still running both heating elements, I have not tested the output with heating and bubbles with the Wifi Controller removed)

I wonder if there is a way to write into the code, for example: If bubbles are turned on while heating is turned on reduce the HEATBITMASK = 48 when bubbles are turned off, increase the HEATBITMASK = 112;

Just I thought, Im no expert at coding whatsoever.

visualapproach, 2021-09-28:
You are thinking right regarding the HEATBITMASK. I must consider compatibility with all models though so I can't hard code for this particular model. I'm working on it and will update soon.

visualapproach, 2021-09-28:
Updated the 4-wire code now. You need to edit "model.h" and choose 54173 model. You should now see lower wattage when running bubbles and heater. When turning off bubbles the power should increase again. No jets allowed during bubbling. Feel free to test the new code and report any bugs I may have introduced.

Lukey1979, 2021-09-29:
Hi @visualapproach I have uploaded the updated Dev version to a new board and will test tomorrow after work and report back :-) p.s.  I love your work on this and hope more people get onboard with the testing of different model spas :-)

I also have a family member that is an electrical engineer so I will ask his advice on the possible "Brownouts" and what a solution may be. I think your right in that a capacitor may help. Just so you know, last night I had 2 E13 errors, when I went to inspect on both occasions the Web Interface had rebooted or disconnected at some point and the Take Control had been switch off automatically.... This was all with the Pump running and Heater switch on and the spa maintaining a temp of 39. Now my suspicion is that when the heater elements would turn on to heat, this would cause the system to error and reboot / disconnect from wifi, hense the E13.. So... to test this further, I reset the system, and this time set the Target temp of the spa to 29 (10c lower than its current temp) and left the pump running over night... It still has not dropped down to the Target Temp of 29 and therefor the heater elements have not had to power-up and almost 24hours later, still NO E13 errors.. Making it look more like a power / brownout issue.. Just a guess though. We'll see what happens when the spa hits its Target Temp of 29 and tries to maintain that..

visualapproach, 2021-09-29:
Hey @Lukey1979 that's a very good observation!! If you look at the config.html page, at the bottom it says the reason for rebooting. If it shows "power-on-reset" it suggests that you are right.
Presently when turning on the heater it is full blast at once. Maybe if we perform a two-stage turn on we will not have the power resets. Let me update the code again...

visualapproach, 2021-09-29:
On the other hand, you reported E13 errors before we even used the 2nd heater element. But maybe we will reduce the number of E13 with a code change, and eliminate it completely with a capacitor. (by sticking a 47 or 100 uF cap between 5V and GND. Check polarity on cap.)

visualapproach, 2021-09-29:
uploaded. Now the heater elements *should* turn on in sequence, 10 seconds apart.

Lukey1979, 2021-09-30:
Hi @visualapproach Ill have to test all the changes tomorrow.. Its pouring down with rain outside and its probably not a good idea to play with electronics in the rain :-) I will report back tomorrow.. Keeping my fingers crossed..

Lukey1979, 2021-10-01:
@visualapproach Ok so far its been a success.. uploaded the new Dev onto the ESP8266, and the power stepping appears to work well. I am also not getting the big Wattage jump when I turn on the bubbles with the heater already on. In fact the Watts go from about (1610W-1800W range) to (approx 1560 Watts) so a nice slight power drop. There are some little programming logic niggles though, for example, if I have the Jets on and then turn on the bubbles the jets turn off (This should not happen) Also, if I have the heater off and the bubbles on and then turn the heater on with the bubbles, the bubbles turn off. (This should not happen) Its working the way you have programmed it though, as in, "heater ON" ----> Turn on bubbles and heater stays on. We are getting there :-) I should say you are getting there... Your doing a brilliant job mate..

visualapproach, 2021-10-01:
Splendid! And thank you, it is easier to fix things when I get clear info as you provided 😃 
I'm away a couple of days so I'll look into the quirks when I get home.

visualapproach, 2021-10-01:
> Hi @visualapproach Ill have to test all the changes tomorrow.. Its pouring down with rain outside and its probably not a good idea to play with electronics in the rain :-) I will report back tomorrow.. Keeping my fingers crossed..
> 
> 
> 
> Just out of curiosity, how did you come to the 47uF or 100uF figure.. 
> 
> I was watching this YouTube video ---->  https://www.youtube.com/watch?v=J66_8P043ko
> 
> and he was using 1000uF for an ESP8266 (I have almost zero understanding of this, so am asking out of curiosity and so I can learn :-) Most of my research so far has suggested that I should use a slightly higher voltage capacitor than the devices input / output voltage). 

Well, in this case the cap is acting like a 'battery' that provides some juice while the 5V line is stalling so the more the marrier one would think. But the larger cap the more inrush current when powering up the pump. (The power supply sees the cap as a short circuit until it's charged which will take longer on a big one) 
Anything less than a uF would would power the esp for, say a couple of  useconds. And a 1000uF may work but would result in high inrush. Maybe the CIO will survive a short circuit but I don't want to gamble. So I suggested something relatively modest. No science behind it really.

visualapproach, 2021-10-01:
And btw, the voltage reading on the caps is what they can take before break, sort of. So you'd want something well above 5V. 16V is common and would be ok.

visualapproach, 2021-10-03:
@Lukey1979 you can try the new code now. Again - remember to choose model (another model than yours is default)
My policy when coding was "the user is always right". That is not always the case in reality, but it works like this: if the user has bubbles on and presses HEAT ON then the bubbles will stop so full heating power will be possible. If you then turn on bubbles, the bubbles will start and the heat will be on at a lower power.

Lukey1979, 2021-10-04:
Thank you @visualapproach Ill give the new code a try once it stops raining again :-) FYI, No E13 errors for a few days now and its been running good (I haven't installed any capacitors as yet either). I did however notice a higher daily kWh usage with the new heating pattern but I would like to test this and chart it more accurately over a few weeks so we can compare the before and after charts.. 

So just to clarify, with the latest changes to the Code, what did you modify for me?

visualapproach, 2021-10-04:
Pretty much what you asked for. Take a look in the ...globals.h and you will see a matrix that governs which functions are allowed in combination with other functions. So turning on jets won't turn off bubbles and so on.

Lukey1979, 2021-10-09:
Hi @visualapproach I finally got around to testing the latest Dev version. I think something went really wrong with the programming logic. I can confirm this is now what happens:

New Issues:

**(Scenario / State)** - Bubbles ON
**(Whats Happening)** - Turn Jets On and then bubbles turn off automatically 
**(Change to)** - Bubbles should stay on but heater and pump off if they were previously on with the bubbles

**(Scenario / State)** - Jets ON 
**(Whats Happening)** - Turn Bubbles On and then Jets turn off automatically 
**(Change to)** - Jets should stay on

**(Scenario / State)** - Pump ON & Heater ON
**(Whats Happening)** - Turn Bubbles On and then the Pump turns off automatically however the heater appears to stay on (_This could mean its heating water in the heating element without pumping any new water through it_) - 
**(Change to)** - Heater & Pump should stay on when Bubbles are switched on however only heating at 50% like previously programmed

**(Scenario / State)** - Pump ON & Heating ON
**(Whats Happening)** - Turn Jets On and the Pump & Heater stay ON
**(Change to)** -  Pump & Heater should turn off when Jets are turned on

Most of the above Scenario's  were working prior to this latest change in the Dev version. I believe the only changes should have been the below:

1.
(Scenario / State) - Bubbles ON but heater and pump are off  
(Whats Happening) - Turn on Pump & Heater and bubbles turn off automatically 
(Changed / Change to) - Bubbles Stay On

2.
(Scenario / State) - Jets ON - Bubbles, Pump & Heater Off 
(Whats Happening) - Turn on Bubbles and Jets turn off automatically 
(Changed / Change to) - Jets stay On
(The above scenario is with heater & pump off as it should be)

Hopefully Im not being a pain but there is at least some good news. I have had ZERO E13 errors in just over a week :-)

visualapproach, 2021-10-09:
I had a brief look at it and found one thing that is wrong. Will look at it again after breakfast.

Lukey1979, 2021-10-09:
@visualapproach I found 1 more thing that's wrong LOL -  I cannot find your "Buy me a coffee" button so I can compensate you for all of your hard work :-)

Lukey1979, 2021-10-09:
@visualapproach and no rush mate, its 6:30pm here in Australia so I won't be making any changes tonight :-)

visualapproach, 2021-10-09:
Haha thanks! You can also tinker after your next breakfast then..
I'll try to fix it asap anyway if someone else is using it. Great description you wrote btw!

visualapproach, 2021-10-09:
@Lukey1979 In order to make it work I had to scrap the "combination matrix" and introduce a new "jumptable". In short - one has to define all possible states the device is allowed to be in, and then also define what state/mode to change to for every button press. You don't have to understand what I'm talking about, it's just me saying I made quite a large change, prone to new errors ;-)
As always - change to your model in "model.h"

Lukey1979, 2021-10-10:
Hi @visualapproach I accidentally closed this and can't figure out how to reopen the issue.

Lukey1979, 2021-10-10:
Accidentally closed

visualapproach, 2021-10-10:
I corrected a few mistakes now. And also changed 'WebSocket.js' file. The Heater button should be to the right when heat is on. I saw it wasn't in your movie. After uploading the new code, visit IP/upload.html and upload websockets.js file.

visualapproach, 2021-10-10:
You can also compile and upload LittleFS data as you did the first time, but that will wipe any save files. I hope it will work fine now so you don't have to test all the time.

Lukey1979, 2021-10-11:
@visualapproach So Far So Good.. All is working as it should.. Thanks a million.. Now when its pay day this week ill shout you a coffee :-) I will close off this thread now with a huge thanks to all of your hard work.

closed 2021-10-11 by Lukey1979

visualapproach, 2021-10-11:
> @visualapproach So Far So Good.. All is working as it should.. Thanks a million.. Now when its pay day this week ill shout you a coffee :-) I will close off this thread now with a huge thanks to all of your hard work.

Man that's good to hear! By making detailed reports and requests You paved the road for other 4 wire users so you deserve a thank you yourself! Have a nice spring down there and don't hesitate to post again. /Thomas

Lukey1979, 2021-10-11:
@Inazone i built both boards to test but ended up having better luck with the original "Red LLC" board. This is the board with the 7 x 560ohm Resistors. Not the "Blue LLC" board.. By the way, I have spare blank boards and spare LLC's if you need. I am having almost no e13 errors with this setup. I was going to try and use a capacitor on the board to help stop the sudden power drops on the ESP32 but with the new coding it all seems to be working well.

Inazone, 2021-10-11:
> @Inazone i built both boards to test but ended up having better luck with the original "Red LLC" board. This is the board with the 7 x 560ohm Resistors. Not the "Blue LLC" board.. By the way, I have [...]

Hi @Lukey1979 thanks for the reply. If you have some spare boards, that would be very useful. Let me know what the costs are, I can PayPal you what you want. Can you DM on github.com? Or can give you my email address... Thanks again for your efforts.

Lukey1979, 2021-10-11:
Hi @Inazone you can send me an email with your postal address and I'll send off some items to you :-) my email address is smartdumb.iothings@gmail.com