visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #135: Integration directly to homeassistant (config)

strozl, 2021-10-21:
I don't use nodered in combination with home-assistant, so i integrated mqtt messages directly to home-assistant

configuration.yaml
```
sensor:
  - platform: mqtt
    name: spa button
    state_topic: "BW_2.0.0/button"
    availability_topic: "BW_2.0.0/Status"
    payload_available: "Alive"
    payload_not_available: "Dead"
  - platform: mqtt
    name: spa connect count
    state_topic: "BW_2.0.0/MQTT_Connect_Count"

  - platform: mqtt
    name: spa temperature
    state_topic: "BW_2.0.0/message"
    unit_of_measurement: '°C'
    value_template: "{{ value_json.TMP }}"
    expire_after: 600
    availability_topic: "BW_2.0.0/Status"
    payload_available: "Alive"
    payload_not_available: "Dead"
  - platform: mqtt
    name: spa target temperature
    unit_of_measurement: '°C'
    state_topic: "BW_2.0.0/message"
    value_template: "{{ value_json.TGT }}"
    expire_after: 600

switch:
  - platform: mqtt
    command_topic: "BW_2.0.0/command"
    state_topic: "BW_2.0.0/message"
    value_template: |- #"{{ value_json.GRN }}"
      {% if value_json.RED == 1 %}
        1
      {% elif value_json.GRN == 1 %}
        1
      {% else %}
        0
      {% endif %}
    name: spa heater
    icon: mdi:radiator
    state_on: 1
    state_off: 0
    payload_on: '{CMD:3,VALUE:true,XTIME:0,INTERVAL:0}'
    payload_off: '{CMD:3,VALUE:false,XTIME:0,INTERVAL:0}'
    availability_topic: "BW_2.0.0/Status"
    payload_available: "Alive"
    payload_not_available: "Dead"
  - platform: mqtt
    command_topic: "BW_2.0.0/command"
    state_topic: "BW_2.0.0/message"
    value_template: "{{ value_json.AIR }}"
    name: spa airbubbles
    icon: mdi:chart-bubble
    payload_on: '{CMD:2,VALUE:true,XTIME:0,INTERVAL:0}'
    payload_off: '{CMD:2,VALUE:false,XTIME:0,INTERVAL:0}'
    state_on: 1
    state_off: 0
    availability_topic: "BW_2.0.0/Status"
    payload_available: "Alive"
    payload_not_available: "Dead"
  - platform: mqtt
    command_topic: "BW_2.0.0/command"
    state_topic: "BW_2.0.0/message"
    value_template: "{{ value_json.FLT }}"
    name: spa pump
    icon: mdi:pump
    payload_on: '{CMD:4,VALUE:true,XTIME:0,INTERVAL:0}'
    payload_off: '{CMD:4,VALUE:false,XTIME:0,INTERVAL:0}'
    state_on: 1
    state_off: 0
    availability_topic: "BW_2.0.0/Status"
    payload_available: "Alive"
    payload_not_available: "Dead"

  - platform: mqtt
    command_topic: "BW_2.0.0/command"
    state_topic: "BW_2.0.0/message"
    value_template: "{{ value_json.UNT }}"
    name: spa temperature unit
    icon: mdi:temperature-celsius
    payload_on: '{CMD:1,VALUE:true,XTIME:0,INTERVAL:0}'
    payload_off: '{CMD:1,VALUE:false,XTIME:0,INTERVAL:0}'
    state_on: 1
    state_off: 0
    availability_topic: "BW_2.0.0/Status"
    payload_available: "Alive"
    payload_not_available: "Dead"
binary_sensor:
  - platform: mqtt
    name: spa lock
    state_topic: "BW_2.0.0/message"
    value_template: |- #"{{ value_json.LCK }}"
      {% if value_json.LCK == 1 %}
        OFF
      {% else %}
        ON
      {% endif %}
    expire_after: 600
    icon: mdi:lock
    availability_topic: "BW_2.0.0/Status"
    payload_available: "Alive"
    payload_not_available: "Dead"
    device_class: lock

  - platform: mqtt
    name: spa heating
    state_topic: "BW_2.0.0/message"
    value_template: |- #"{{ value_json.RED }}"
      {% if value_json.RED == 1 %}
        ON
      {% else %}
        OFF
      {% endif %}
    icon: mdi:fire
    availability_topic: "BW_2.0.0/Status"
    payload_available: "Alive"
    payload_not_available: "Dead"
    expire_after: 600
  - platform: mqtt
    name: spa ready
    state_topic: "BW_2.0.0/message"
    value_template: |- #"{{ value_json.RED }}"
      {% if value_json.TMP > 30 %}
        {% if value_json.TMP >= value_json.TGT-1 %}
          ON
        {% else %}
          OFF
        {% endif %}

      {% else %}
        OFF
      {% endif %}
    icon: mdi:hot-tub
    availability_topic: "BW_2.0.0/Status"
    payload_available: "Alive"
    payload_not_available: "Dead"
    expire_after: 600
  - platform: mqtt
    name: spa power
    device_class: power
    state_topic: "BW_2.0.0/message"
    value_template: |- #"{{ value_json.RED }}"
      {% if value_json.PWR == 1 %}
        ON
      {% else %}
        OFF
      {% endif %}
    availability_topic: "BW_2.0.0/Status"
    payload_available: "Alive"
    payload_not_available: "Dead"
    expire_after: 600
```

automations.yaml
```
- id: '1634672047071'
  alias: Spa target (from spa)
  description: ''
  trigger:
  - platform: state
    entity_id: sensor.spa_target_temperature
  condition: []
  action:
  - service: input_number.set_value
    data:
      entity_id: input_number.spa_target_temp
      value: '{{states(''sensor.spa_target_temperature'')}}'
  mode: single
- id: '1634672260932'
  alias: Spa target (to spa)
  description: ''
  trigger:
  - platform: state
    entity_id: input_number.spa_target_temp
  condition: []
  action:
  - delay:
      hours: 0
      minutes: 0
      seconds: 0
      milliseconds: 0
  - service: mqtt.publish
    data:
      topic: BW_2.0.0/command
      payload_template: '{CMD:0,VALUE:{{states(''input_number.spa_target_temp'')|int}},XTIME:0,INTERVAL:0}'
  mode: single
``
`

you need to add helper input_number with name spa_target_temp 
```
input_number:
  spa_target_temp:
    name: SPA Target Temperature
    min: 20
    max: 40
    step: 1
```

If you don't send messages to the default topic prefix, adjust it.

visualapproach, 2021-10-31:
Thank you @strozl !!! Very good, I will put this in a HA folder if you don't mind. With minor changes.

sjeffrey101, 2021-11-21:
I added a discussion a while ago with how i had it in HA.  [Here](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/discussions/52)

Any way i am now able to do it without an automation to control temperature, direct in a MQTT climate.  Working well for me, let me know if its any help

`Climate:
 - platform: mqtt
    name: SPA Temperature
    max_temp: 40
    min_temp: 20
    precision: 1.0
    temperature_unit: "C"
    modes:
      - "off"
      - "heat"
    mode_state_topic: "BW_2.0.0/message"
    mode_state_template: >-
      {% if value_json.RED == 1 %}
        heat
      {% elif value_json.GRN == 1 %}
        heat
      {% else %}
        off
      {% endif %}  
    action_topic: "BW_2.0.0/message"
    action_template: >-
      {% if value_json.RED == 1 %}
        heating
      {% elif value_json.GRN == 1 %}
        idle
      {% else %}
        off
      {% endif %}          
    temperature_state_topic: "BW_2.0.0/message"
    temperature_state_template: "{{ value_json.TGT }}"
    current_temperature_topic: "BW_2.0.0/message"
    current_temperature_template: "{{ value_json.TMP }}"
    temperature_command_topic : "BW_2.0.0/command"
    temperature_command_template: '{"CMD":0,"VALUE":{{ value|int }},"XTIME":0,"INTERVAL":0}'
    power_command_topic: "BW_2.0.0/command"
    payload_on: '{"CMD":3,"VALUE":1,"XTIME":0,"INTERVAL":0}'
    payload_off: '{"CMD":3,"VALUE":0,"XTIME":0,"INTERVAL":0}'
    availability_topic: "BW_2.0.0/Status"
    payload_available: "Alive"
    payload_not_available: "Dead"
`

NitramH, 2022-01-24:
> Thank you @strozl !!! Very good, I will put this in a HA folder if you don't mind. With minor changes.

Hi,

I implemented your version into homeassistant and it works quite well, thanks!

One point I found out:
After heating up and reaching the target temp, the heating as well as the pump is switched off.
Unfortunately the turned off pump causes the part in the pump box oooling down very quickly
while the spa itself keeps the temperature much longer.
The result is that the 2-level control operates more frequently than necessary (1h instead of 5h).

Do you have an idea to resolve this? I`m not so familiar with autpmations in Homeassistant, I wonder how
to enable the pump after reaching the target?

Additionally I have some more questions:
- Time to target is not updated (is this a feature for future development?)
- Energie consumption as well
- What is the spa connect count for?
- Does anyone has an idea for cycling the pump to save energy?
- Are the filter change and chlor fill up dates permanently stored in the ESP? 
  I want to turn off the system when it is not in use in summertime

visualapproach, 2022-01-24:
@NitramH 
> Do you have an idea to resolve this? I`m not so familiar with autpmations in Homeassistant, I wonder how
to enable the pump after reaching the target?

I'm not sure exactly what you mean. You're right about the pump cooling faster than the main body of water. If you want longer intervals between ON/OFF you can adjust the lower bound.
```
- id: 'start heater at target temp-3'
  alias: SPA-Starta värme vid gränsvärde
  description: ''
  trigger:
  - platform: template
    value_template: '{{ (states("sensor.spa_target_temperature")|int - states("sensor.spa_temperature")|int)
      > 2 }}'  // *******  EDIT HERE TO A LARGER TEMP DIFF
  condition: []
  action:
  - service: switch.turn_on
    target:
      entity_id: switch.spa_heat_regulation
  mode: single
```

To the other questions:

> Time to target is not updated (is this a feature for future development?)

It updates when it gets something to calculate - temp must change before recalc. You must wait for two updates before it gets somewhat accurate.

> Energie consumption as well

Yes, it's sent as "KWH" both in websockets and MQTT and there is a sensor template for HA.

> What is the spa connect count for?

Count how many times the MQTT has connected. >1 indicates a network/MQTT problem

> Does anyone has an idea for cycling the pump to save energy?

Don't understand

> Are the filter change and chlor fill up dates permanently stored in the ESP?
I want to turn off the system when it is not in use in summertime

It's saved until you erase the filesystem, but it saves the timestamp, so it will show how many days since you reset the timer. Not how many days it has been running.

NitramH, 2022-01-24:
With cycling I mean ON for 20min, OFF for 40min etc.

Regarding spa connect count it increases to 12. This correlates to the Wifi issue I posted separately.
So at first I have to fix the WiFi, then I will check the other topics later :-(

visualapproach, 2022-01-24:
Okay you can do it from the web gui by adding commands to the queue, or from an automation in HA. The latter requires a stable WiFi ofc.

sjeffrey101, 2022-01-24:
ive tried quite a few different ways of doing the heating.  i now think the best way of doing it, is just to leave the spa to do it.

closed 2023-03-13 by visualapproach