visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #140: Strange behavior

AlbertDijksma, 2021-10-30:
Hi,
I see something strange happening.
The SPA is on but the pumps and heating are off.
Still I see the pump and air times slowly increasing.
I see this happening in the debug log as well.
[image]
[image]

debug.html:20 {"CONTENT":"TIMES","TIME":1635616932,"CLTIME":1634405731,"FTIME":1634405733,"UPTIME":48529,"PUMPTIME":199,"HEATINGTIME":0,"AIRTIME":104,"JETTIME":0,"COST":0.013071,"FINT":30,"CLINT":14}
debug.html:20 {"CONTENT":"OTHER","MQTT":0,"PressedButton":"1b1b","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":1,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":38,"TMP":13,"CH1":32,"CH2":49,"CH3":51,"HJT":0}
debug.html:20 {"CONTENT":"TIMES","TIME":1635616934,"CLTIME":1634405731,"FTIME":1634405733,"UPTIME":48531,"PUMPTIME":199,"HEATINGTIME":0,"AIRTIME":104,"JETTIME":0,"COST":0.013071,"FINT":30,"CLINT":14}
debug.html:20 {"CONTENT":"OTHER","MQTT":0,"PressedButton":"1b1b","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":1,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"**FLT":0**,"TGT":38,"TMP":13,"CH1":32,"CH2":49,"CH3":51,"HJT":0}
debug.html:20 {"CONTENT":"TIMES","TIME":1635616936,"CLTIME":1634405731,"FTIME":1634405733,"UPTIME":48533,"PUMPTIME":199,"HEATINGTIME":0,"AIRTIME":104,"JETTIME":0,"COST":0.013071,"FINT":30,"CLINT":14}
debug.html:20 {"CONTENT":"OTHER","MQTT":0,"PressedButton":"1b1b","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":0,"UNT":0,"AIR":0,"GRN":0,"RED":0,"**FLT":1**,"TGT":38,"TMP":0,"CH1":32,"CH2":32,"CH3":32,"HJT":0}
debug.html:20 {"CONTENT":"TIMES","TIME":1635616937,"CLTIME":1634405731,"FTIME":1634405733,"UPTIME":48534,"**PUMPTIME":199**,"HEATINGTIME":0,"AIRTIME":104,"JETTIME":0,"COST":0.013071,"FINT":30,"CLINT":14}
debug.html:20 {"CONTENT":"OTHER","MQTT":0,"PressedButton":"1b1b","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":1,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"**FLT":0**,"TGT":38,"TMP":13,"CH1":32,"CH2":49,"CH3":51,"HJT":0}
debug.html:20 {"CONTENT":"TIMES","TIME":1635616938,"CLTIME":1634405731,"FTIME":1634405733,"UPTIME":48535,"**PUMPTIME":200**,"HEATINGTIME":0,"AIRTIME":104,"JETTIME":0,"COST":0.013074,"FINT":30,"CLINT":14}
debug.html:20 {"CONTENT":"OTHER","MQTT":0,"PressedButton":"1b1b","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":1,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":38,"TMP":13,"CH1":32,"CH2":49,"CH3":51,"HJT":0}
debug.html:20 {"CONTENT":"TIMES","TIME":1635616938,"CLTIME":1634405731,"FTIME":1634405733,"UPTIME":48535,"PUMPTIME":200,"HEATINGTIME":0,"AIRTIME":104,"JETTIME":0,"COST":0.013074,"FINT":30,"CLINT":14}
debug.html:20 {"CONTENT":"OTHER","MQTT":0,"PressedButton":"1b1b","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":1,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":38,"TMP":13,"CH1":32,"CH2":49,"CH3":51,"HJT":0}
debug.html:20 {"CONTENT":"TIMES","TIME":1635616940,"CLTIME":1634405731,"FTIME":1634405733,"UPTIME":48537,"PUMPTIME":200,"HEATINGTIME":0,"AIRTIME":104,"JETTIME":0,"COST":0.013075,"FINT":30,"CLINT":14}
debug.html:20 {"CONTENT":"OTHER","MQTT":0,"PressedButton":"1b1b","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":1,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":38,"TMP":13,"CH1":32,"CH2":49,"CH3":51,"HJT":0}
debug.html:20 {"CONTENT":"TIMES","TIME":1635616942,"CLTIME":1634405731,"FTIME":1634405733,"UPTIME":48539,"PUMPTIME":200,"HEATINGTIME":0,"AIRTIME":104,"JETTIME":0,"COST":0.013075,"FINT":30,"CLINT":14}

visualapproach, 2021-10-30:
Hmm. See if it does this in the latest dev version also. I suspect glitches in the communication that may add up over time. I made some small changes that hopefully improves this.

AlbertDijksma, 2021-10-31:
Hi,
Running now on dev and see the same glitches
[image]

visualapproach, 2021-10-31:
Okay I'll see if I can do something.

visualapproach, 2021-10-31:
I pushed a fix to what I believe was the problem. (Dev version)

AlbertDijksma, 2021-10-31:
Sorry still the same problem.
But found another strange problem.
I think it worked in the official release.
The Unit button on the spa has stopped working. (works remotely)
Is the debug log did I first press the unlock button 0100 ?
And then press the Unit button a few times. 0809 ?

debug.html:20 {"CONTENT":"STATES","LCK":1,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":1,"FLT":1,"TGT":38,"TMP":38,"CH1":32,"CH2":51,"CH3":56,"HJT":0,"BRT":2}
debug.html:20 {"CONTENT":"TIMES","TIME":1635707235,"CLTIME":1635687912,"FTIME":1635687910,"UPTIME":20368,"PUMPTIME":19437,"HEATINGTIME":365,"AIRTIME":123,"JETTIME":0,"COST":0.134176,"FINT":30,"CLINT":14,"KWH":0.447254,"TTTT":-1350}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"1b1b","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":1,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":1,"FLT":1,"TGT":38,"TMP":38,"CH1":32,"CH2":51,"CH3":56,"HJT":0,"BRT":2}
debug.html:20 {"CONTENT":"TIMES","TIME":1635707237,"CLTIME":1635687912,"FTIME":1635687910,"UPTIME":20370,"PUMPTIME":19439,"HEATINGTIME":367,"AIRTIME":123,"JETTIME":0,"COST":0.1345,"FINT":30,"CLINT":14,"KWH":0.448333,"TTTT":-1352}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"1b1b","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":1,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":1,"FLT":1,"TGT":38,"TMP":38,"CH1":32,"CH2":51,"CH3":56,"HJT":0,"BRT":2}
debug.html:20 {"CONTENT":"TIMES","TIME":1635707239,"CLTIME":1635687912,"FTIME":1635687910,"UPTIME":20372,"PUMPTIME":19441,"HEATINGTIME":369,"AIRTIME":123,"JETTIME":0,"COST":0.134824,"FINT":30,"CLINT":14,"KWH":0.449412,"TTTT":-1354}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"1b1b","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":1,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":1,"FLT":1,"TGT":38,"TMP":38,"CH1":32,"CH2":51,"CH3":56,"HJT":0,"BRT":2}
debug.html:20 {"CONTENT":"TIMES","TIME":1635707241,"CLTIME":1635687912,"FTIME":1635687910,"UPTIME":20374,"PUMPTIME":19443,"HEATINGTIME":371,"AIRTIME":123,"JETTIME":0,"COST":0.135147,"FINT":30,"CLINT":14,"KWH":0.450491,"TTTT":-1356}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"1b1b","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":1,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":1,"FLT":1,"TGT":38,"TMP":38,"CH1":32,"CH2":51,"CH3":56,"HJT":0,"BRT":2}
debug.html:20 {"CONTENT":"TIMES","TIME":1635707243,"CLTIME":1635687912,"FTIME":1635687910,"UPTIME":20376,"PUMPTIME":19445,"HEATINGTIME":373,"AIRTIME":123,"JETTIME":0,"COST":0.135471,"FINT":30,"CLINT":14,"KWH":0.45157,"TTTT":-1358}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"1b1b","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":1,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":1,"FLT":1,"TGT":38,"TMP":38,"CH1":32,"CH2":51,"CH3":56,"HJT":0,"BRT":2}
debug.html:20 {"CONTENT":"TIMES","TIME":1635707245,"CLTIME":1635687912,"FTIME":1635687910,"UPTIME":20378,"PUMPTIME":19447,"HEATINGTIME":375,"AIRTIME":123,"JETTIME":0,"COST":0.135795,"FINT":30,"CLINT":14,"KWH":0.452649,"TTTT":-1360}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"1b1b","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":1,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":1,"FLT":1,"TGT":38,"TMP":38,"CH1":32,"CH2":51,"CH3":56,"HJT":0,"BRT":2}
debug.html:20 {"CONTENT":"TIMES","TIME":1635707247,"CLTIME":1635687912,"FTIME":1635687910,"UPTIME":20380,"PUMPTIME":19449,"HEATINGTIME":377,"AIRTIME":123,"JETTIME":0,"COST":0.136118,"FINT":30,"CLINT":14,"KWH":0.453728,"TTTT":-1362}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"1b1b","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":1,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":1,"FLT":1,"TGT":38,"TMP":38,"CH1":32,"CH2":51,"CH3":56,"HJT":0,"BRT":2}
debug.html:20 {"CONTENT":"TIMES","TIME":1635707247,"CLTIME":1635687912,"FTIME":1635687910,"UPTIME":20380,"PUMPTIME":19449,"HEATINGTIME":377,"AIRTIME":123,"JETTIME":0,"COST":0.136118,"FINT":30,"CLINT":14,"KWH":0.453728,"TTTT":-1362}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"1b1b","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":1,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":1,"FLT":1,"TGT":38,"TMP":38,"CH1":32,"CH2":51,"CH3":56,"HJT":0,"BRT":2}
debug.html:20 {"CONTENT":"TIMES","TIME":1635707248,"CLTIME":1635687912,"FTIME":1635687910,"UPTIME":20381,"PUMPTIME":19450,"HEATINGTIME":377,"AIRTIME":123,"JETTIME":0,"COST":0.136122,"FINT":30,"CLINT":14,"KWH":0.453739,"TTTT":-1363}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"1b1b","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":1,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":1,"FLT":1,"TGT":38,"TMP":38,"CH1":32,"CH2":51,"CH3":56,"HJT":0,"BRT":2}
debug.html:20 {"CONTENT":"TIMES","TIME":1635707249,"CLTIME":1635687912,"FTIME":1635687910,"UPTIME":20382,"PUMPTIME":19451,"HEATINGTIME":378,"AIRTIME":123,"JETTIME":0,"COST":0.136284,"FINT":30,"CLINT":14,"KWH":0.454279,"TTTT":-1364}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"1b1b","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":1,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":1,"FLT":1,"TGT":38,"TMP":38,"CH1":32,"CH2":51,"CH3":56,"HJT":0,"BRT":2}
debug.html:20 {"CONTENT":"TIMES","TIME":1635707251,"CLTIME":1635687912,"FTIME":1635687910,"UPTIME":20384,"PUMPTIME":19453,"HEATINGTIME":380,"AIRTIME":123,"JETTIME":0,"COST":0.136607,"FINT":30,"CLINT":14,"KWH":0.455358,"TTTT":-1366}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"1b1b","HASJETS":0}
[... cut ...]
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"1b1b","HASJETS":0}
debug.html:20 {"CONTENT":"STATES","LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":1,"FLT":1,"TGT":38,"TMP":38,"CH1":32,"CH2":51,"CH3":56,"HJT":0,"BRT":2}
debug.html:20 {"CONTENT":"TIMES","TIME":1635707295,"CLTIME":1635687912,"FTIME":1635687910,"UPTIME":20428,"PUMPTIME":19497,"HEATINGTIME":424,"AIRTIME":123,"JETTIME":0,"COST":0.143728,"FINT":30,"CLINT":14,"KWH":0.479093,"TTTT":-1410}
debug.html:20 {"CONTENT":"OTHER","MQTT":-1,"PressedButton":"1b1b","HASJETS":0}

visualapproach, 2021-10-31:
Haha, it's actually a feature, not a bug! No I'm not kidding you. You can disable buttons on the display, to use them for better things. I have disabled my unit and timer buttons, and use them to open/close my awnings when in the spa :-). The device disables the original functions, but sends the button press over MQTT so your home automation can do whatever it wants.

Edit line 77 in BWC_8266_globals.h. Change the zeroes to ones
`const uint8_t EnabledButtons[] = {1,1,0,1,0,1,1,1,1,1,1};`

AlbertDijksma, 2021-10-31:
Awesome,
Now I can use those buttons for something else to call my wife for a glass of wine

visualapproach, 2021-10-31:
Yes, and when you press bubbles, she brings champagne 🍾

visualapproach, 2021-11-01:
@AlbertDijksma I can confirm I also see this behaviour. I logged it for a couple of hours and only gained one second and only on the pumptime. I think communication glitches from the CIO might accumulate over time. When I hacked the protocol I noticed the CIO was kind of sloppy with the timings, but maybe that was my setup. The glitches has to persistent over two messages in a row to affect the ESP so it's a little bit strange. It can ofc be rounding errors or similar. I have some ideas how to improve it when I get time.

AlbertDijksma, 2021-11-02:
What you should keep in mind is that the esp8266 only has 1 core.
The wifi part of the esp requires a lot of interrupts.
Could this be the problem?

visualapproach, 2021-11-03:
@AlbertDijksma Maybe, if it delays my own interrupts used to read data from CIO. But anyhow, I made some small changes if you would like to try. I haven't got any increased numbers in an hour of testing now.

AlbertDijksma, 2021-11-03:
You are great,
Now running for 2 hours and none glitches

visualapproach, 2021-11-03:
Lovely! Thanks for reporting and testing!