visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #144: E13 error after 5mins running

Inazone, 2021-11-08:
Hi there, I have the maldives pump early version same as @Lukey1979 and built the module in the same way. I have commented out the correct details in the software. When I connect my module it works fine for about 5mins but then the display starts beeping and displays the E13 code. I also noticed that the WiFi blue led would flash on and off, but when just plugged into the USB port it is permanently lit. I'm wondering whether the WiFi link is causing the issue? Also how can I get back to the WiFi portal to change the WiFi LAN as I have several to choose from?

Many thanks

Alex

visualapproach, 2021-11-08:
go to the page
YOUR-IP/resetwifi/
to forget the wifi credentials and activate the portal

Inazone, 2021-11-08:
Excellent thanks for that, I'll give that a go tomorrow. Do you think it's spotty Wi-Fi which is causing the E13 issue?

Inazone, 2021-11-09:
Hi, I've tried different WiFi sources and it still seems to drop off the network which then kicks in the E13 error. It's seems the loss of WiFi causes the E13 error. Not sure why it is dropping off the network as the signal is quite strong! Any ideas on where to look next?

visualapproach, 2021-11-09:
You can try hardcoding the wifi settings in main.cpp/startwifi()
Don't remember exactly how, but you can edit the autoportal there. Can look later but if you are impatient there is a description somewhere in the forum.

Inazone, 2021-11-09:
Ok thanks, I will take a look at that. Thanks for your quick response.

Inazone, 2021-11-09:
I'm afraid I didn't have much luck with your suggestion, couldn't find anything which remotely looked like a place to put my network details in. Unfortunately my coding skills are practically zero...

visualapproach, 2021-11-09:
To avoid autoportal to run and block the program, 
Change this
```
void startWiFi() { // Start a Wi-Fi access point, and try to connect to some given access points. Then wait for either an AP or STA connection
  WiFi.mode(WIFI_STA);
  //wm.setConfigPortalBlocking(false);
  wm.autoConnect("AutoPortal");
  while (WiFi.status() != WL_CONNECTED) {  // Wait for the Wi-Fi to connect
    delay(250);
    //Serial.print('.');
    //checkAP_request();
  }
```
to this
```
void startWiFi() { // Start a Wi-Fi access point, and try to connect to some given access points. Then wait for either an AP or STA connection
  WiFi.mode(WIFI_STA);
  //wm.setConfigPortalBlocking(false);
  //wm.autoConnect("AutoPortal");                                               COMMENT OUT THIS LINE
WiFi.begin("put your SSID here", "and your password here");    //<---------- enter credentials within the citation marks
  while (WiFi.status() != WL_CONNECTED) {  // Wait for the Wi-Fi to connect
    delay(250);
    //Serial.print('.');
    //checkAP_request();
  }
```

Inazone, 2021-11-09:
That's great, I will give that a go tomorrow and see if it works! I also read on one of the posts about using a capacitor to smooth the voltage spikes. Do you know if anyone has tried this?

visualapproach, 2021-11-10:
> That's great, I will give that a go tomorrow and see if it works! I also read on one of the posts about using a capacitor to smooth the voltage spikes. Do you know if anyone has tried this?

I'm not aware of anyone using capacitors. The problem with brown outs seemed to disappear after implementing the two-stage heater on procedure.

Inazone, 2021-11-10:
Ok, what voltage should I be getting to feed the module and to output to the display as I will take some measurements. I am assuming I should be getting 5V. What voltage would cause a brown out?

visualapproach, 2021-11-10:
Yes 5V from the pump powers the ESP and display. I don't know exactly when brown out will occur. I suspect 3.3V + voltage drop out for the 3.3V voltage regulator on the nodeMCU. So maybe 4V?

Inazone, 2021-11-10:
Here is a video of what happens when the E13 error comes up. I was wondering whether you could tell what might be happening when the blue led comes on and off and then starts flashing again...
https://photos.app.goo.gl/Qc26V7pV3KqcKCcMA

visualapproach, 2021-11-10:
The LED turns on/off every time CIO sends a message. I'd replace the resistors with wires instead (0 Ohms resistors).

visualapproach, 2021-11-10:
Does it fail while idling, or when turning on something?

Inazone, 2021-11-10:
Seems to fail after running for a few minutes

Inazone, 2021-11-10:
I'm wondering if it's loosing wifi connection

visualapproach, 2021-11-10:
Loosing wifi should not matter when autoportal is removed. Resistors are known to cause E13.

Inazone, 2021-11-10:
Ok I will remove the resistors and try again. Thanks for your help.

visualapproach, 2021-11-10:
no problem

Inazone, 2021-11-11:
What's your thoughts on the below? I have been wondering whether the WiFi drops out which causes the module to reset or throw the E13. Reading around the WiFi module on the nodemcu is an older version esp-12e where as the latest version is esp-12f which has an optimised antenna. See link below.
https://www.ebay.co.uk/itm/NodeMcu-Lua-V3-CH340G-ESP8266-Wireless-Internet-Develop-Board-WeMos-D1-NEW-/202066482994?mkcid=16&mkevt=1&_trksid=p2349624.m46890.l49286&mkrid=710-127635-2958-0

visualapproach, 2021-11-11:
Did you try https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/144#issuecomment-964403694 and https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/144#issuecomment-965062414 ?
The 12F would probably give a better signal (I don't *know* that but it's a fair guess) but I *think* it's unrelated to your E13 problem.

Inazone, 2021-11-15:
Hi there, OK finally got round to removing the resistors and replacing them with normal wires. Unfortunately I am getting the same error. It connects to WiFi, the blue led flashes for about 3 or 4 minutes until the led light stays on and then the pump stops and the E13 error flashes up and it starts beeping... Any ideas on what to try next?

Inazone, 2021-11-15:
Oh just to mention I checked the voltage from the spa and I am getting 4.97V and then on the other side of the module 4.96V. Is the voltage enough or should I be getting 5V or more?

visualapproach, 2021-11-15:
That's 5V if you ask me. I don't know what you mean by "other side". High voltage side on the LLC should be 5-ish and the Low Voltage side should sit at 3.3V

visualapproach, 2021-11-15:
Did you ever disable the autoportal?

Inazone, 2021-11-15:
Ok I wasn't sure how sensitive the electronics are with regards to the voltage. The otherside I am referring to the output of the module where it plugs into the display.
Yes I have disabled the auto portal.

Inazone, 2021-11-15:
Are there any logs that get generated?

visualapproach, 2021-11-15:
Not automatically. But now that I know what you've tried I'll think about it. You should test the pass through code and see if there is errors

Inazone, 2021-11-15:
Is the pass through code on the dev branch?

Lukey1979, 2021-11-15:
Hi @Inazone Can you try something for me. 
1. Turn off power to the spa (mains power)
2. Turn on power to the spa (do not press the power button on the spa control panel)

3. Wait about 10 or 15 seconds for the controller you made to connect to wifi.

If it starts beeping repeat process.

If no beeping, using the web interface, press take control and then start pump.

I find when I operate my spa this way it is way more successful. 

Let me know if this makes any difference.

If it works, try operating your jets and bubbles to make sure they all work without triggering the beeping.

If this all works, try now pressing the power button on the spa itself. If you want to control the spa from the actual spa control panel turn off take control from the web interface

Inazone, 2021-11-15:
> Hi @Inazone Can you try something for me.
>
> 1. Turn off power to the spa (mains power)
> 2. Turn on power to the spa (do not press the power button on the spa control panel)
> 3. Wait about 10 or 15 [...]

Thanks Luke, I will give your suggestion a try tomorrow. Although I think it's something to do with the esp8266 as it all works great for a few minutes but then something happens and the blue led light stays on and then the pump stops and the E13 error comes up.

Inazone, 2021-11-17:
@Lukey1979 I have tried your suggestion but no luck. The beeping starts very quickly approx 25 to 30 secs after mains power on. Before I even get the chance to refresh the spa webpage. Interestingly if I disconnect the display plug from the module, the module appears to function longer than with it connected, although the spa doesn't work the buttons do on the webpage. I'm wondering if I have some kit? When the esp is connected to the USB port the blue led is solid on and I can easily get on to the webpage ok.

villmatt, 2021-11-24:
Hey, so I’m with @Lukey1979 and a very interesting thing happened,

We overheated and broke the router by means of home automation loop haha. The loop had nothing to do with the spa controller. Was just a light on and off loop. 

But since then, and now with a new router. The spa controller keeps beeping with e13 error, we are about to attempt a re flash however this has made be confident that it has something to do with wifi connectivity. Frequently before the router broke, if we bombarded the network with stuff the spa controller would start beeping, requiring us to power down and back up to get it running again. 

We are stumped since this controller was working relatively seamlessly minus the random hiccups, before we fried the wireless router

villmatt, 2021-11-26:
We re-flashed, no different. But..... and maybe another finding.

Allowed the board to boot up without the spa screen plugged into it. Waited about 30seconds to a minute until the blue light started flashing, then plugged in the screen. 

Screen powered up without e13 error, will report back if anything changes.

Maybe it has to do with the single core board trying to do too many things upon boot and it doesnt pass all the messages needed across to the display quick enough. Not too sure, but everything seems to be pointing to a network upon boot issue

visualapproach, 2021-11-27:
> Maybe it has to do with the single core board trying to do too many things upon boot and it doesnt pass all the messages needed across to the display quick enough. Not too sure, but everything seems to be pointing to a network upon boot issue

I'll have a look. Maybe I can optimize start-up

visualapproach, 2021-11-27:
Check development branch. I think the startup will be faster now. Previously it tried to sync time NTP before wifi was started, so it waited 5 seconds in vain. It will now start wifi, then try to sync NTP for max 3 seconds. Also made a few other tweaks like energy reporting and some fixes. Hopefully I did not introduce new errors.

closed 2022-03-09 by visualapproach