visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #149: Is there no "HASJETS" option for 4-pin models to disable them?

chunkysteveo, 2021-11-22:
Just starting up my setup, got the PCB's and only just realised my NodeMCU boards I have are V2, not V3... going to get creative with a bodge wire on the left(!!) till my V3's show up.

Anyhow - I have a Paris (2018) 4 pin model, No: 54154. Heater, Pump, Bubbles - no jets.

Have just been playing with the firmware and the web gui and MQTT and could see that there was no option in the firmware setup to disable the jets, so the webgui doesn't snow the button etc. Can this be written in? Am guessing it's because of the lack of info on the 4 pin models?

Happy to help and inform for any work on the 4 pin models when I have all my bits!

Cheers,

Steve

visualapproach, 2021-11-22:
Hi @chunkysteveo it's not implemented yet. Meanwhile I think you can just set the jet button visible property to false in websockets.js.

chunkysteveo, 2021-11-22:
That'll do for me till it is. As long as I wasn't being stupid for not seeing it in the code! 👍🏻

closed 2021-11-22 by chunkysteveo